eutelescope/eutelescope #190: Seg fault due to wrong histogram information.

AlexanderMorton, 2015-02-15:
If you incorrectly set the histogram path to the wrong directory in the steering file for the converter step you produce a set fault. This should be improved to exit more appropriately. Moreover we should not seg fault for not producing histogram information.

tbisanz, 2015-02-16:
I'm sorry, I don't quite get it. What do I exactly have to do to replace it? Why is this labeled as "Converter", we don't histogram in the EUDAQ converter plugins. Could you paste a stacktrace?

tbisanz, 2015-02-16:
If I modify the path I don't get a segfault: 
´´´
11:29:58 jobsub.converter(ERROR): SysError in <TFile::TFile>: file ./output/histogramsr/run000097-converter.root can not be opened (No such file or directory)
´´´

but it keeps running