eutelescope/eutelescope #201: compiler error

yeda, 2015-02-21:
EUTelTrackAnalysis gives compiler error, see below

[  3%] Building CXX object CMakeFiles/Eutelescope.dir/src/EUTelTrackAnalysis.cpp.o
/nfs/dust/atlas/user/yeda/ilcsoft/v01-17-05/Eutelescope/trunk/src/EUTelTrackAnalysis.cpp: In member function ‘float eutelescope::EUTelTrackAnalysis::calculatePValueForChi2(eutelescope::EUTelTrack)’:
/nfs/dust/atlas/user/yeda/ilcsoft/v01-17-05/Eutelescope/trunk/src/EUTelTrackAnalysis.cpp:166: error: expected ‘,’ or ‘;’ before ‘return’
/nfs/dust/atlas/user/yeda/ilcsoft/v01-17-05/Eutelescope/trunk/src/EUTelTrackAnalysis.cpp:165: warning: unused variable ‘pValue’
/nfs/dust/atlas/user/yeda/ilcsoft/v01-17-05/Eutelescope/trunk/src/EUTelTrackAnalysis.cpp:167: warning: no return statement in function returning non-void
/nfs/dust/atlas/user/yeda/ilcsoft/v01-17-05/Eutelescope/trunk/src/EUTelTrackAnalysis.cpp: At global scope:
/nfs/dust/atlas/user/yeda/ilcsoft/v01-17-05/Eutelescope/trunk/src/EUTelTrackAnalysis.cpp:164: warning: unused parameter ‘track’
make[2]: *** [CMakeFiles/Eutelescope.dir/src/EUTelTrackAnalysis.cpp.o] Error 1
make[1]: *** [CMakeFiles/Eutelescope.dir/all] Error 2
make: *** [all] Error 2

yeda, 2015-02-21:
it is fixed, please merge my pull request

tbisanz, 2015-02-21:
PR merged, closing issue! Sorry about that and thanks for the fix!

closed 2015-02-21 by tbisanz