eutelescope/eutelescope #210: Remove Deprecated Functionality for v2.0

tbisanz, 2015-03-24:
Remove:

[] EUTelProcessorApplyAlignment
[] EUTelProcessorHotPixelFinder
[] EUTelReferenceHit
[] EUTelAlignmentConstant
[] EUTelMatrixDecoder
[] EUTelPreAlignment

This is the follow up issue from #5, aiming at the release after v1.0

jha92, 2019-01-25:
Still open to this topic are:

- [ ] EUTelProcessorApplyAlignment
- [ ] EUTelAlignmentConstant
- [ ] EUTelMatrixDecoder

The EUTelPreAlignment will be used further.

Maybe adding more for removal at some point.

edorossi, 2019-01-25:
The starting point of this issue is a bit old and now probably there are more processors which are obsolete. I close the issue with the idea in mind of opening a similar one starting from newer EUTelescope versions.

closed 2019-01-25 by edorossi