eutelescope/eutelescope #253: ApplyPreAlignment with SkipNEvents

TCThenna, 2015-08-18:
When skipping at least one event in the config file
>>>
[trackdaf]
# Maximum distance in um for for matches used in EUTelMille/findTrack:
MaxRecordNumber		= 10000 
SkipNEvents 		= 1
...
<<<

the alignment collection is "UNDEFINED".

[ ERROR5 "ApplyPreAlignment"] Alignment collections are UNDEFINED, the processor can not continue. EXIT 
[ ERROR "Marlin"] 
[ ERROR "Marlin"]  **********************************************************
[ ERROR "Marlin"]  *                                                        *
[ ERROR "Marlin"]  *   Stop of EventProcessiong requested by processor :    *
[ ERROR "Marlin"]  *                  ApplyPreAlignment
[ ERROR "Marlin"]  *     will call end() method of all processors !         *
[ ERROR "Marlin"]  *                                                        *
[ ERROR "Marlin"]  **********************************************************

tbisanz, 2015-08-31:
Sorry for the late response - vacation ... I guess you're using the EUTelApplyAlignmentProcessor? It will become obsolete very soon. Despite this, the problem should not be due to the skip event. The alignment collections seems not to be loaded correctly, but it's very difficult to tell from only this. A breakpoint on StopProcessingException and a backtrace from there would help, together with your steering files. But then again - I can think of more usefull things to do than debugging obsolete code.

Neztor, 2015-08-31:
The code you are using is very old.

When you do the prealginment, you get an lcio database file containing
exactly 1 event with the prealignment values (usually in
output/database/runXXXXXX-prealignment.slcio). If you skip that 1 event
(which you do if SkipNEvents is > 0), then there is no alignment to apply,
and it fails.

tbisanz, 2015-08-31:
@Neztor EUTelProcessorApplyAlignment works like this, I'm not sure about EUTelApplyAlignmentProcessor (quickly glancing at the code, it doesn't look like this) - but, you're right, this is how the old alignment database works. One of the reasons why it's deprecated.

TCThenna, 2016-12-07:
the skip event should only skip events in the data file, not in the alignment file (where there is only one event)

tbisanz, 2016-12-12:
The problem is that strictly speaking, these are Marlin features, not EUTelescope ones - and EUTelescope has this bad habit of storing things (alignment and noisy pixels) in the first events which renders the SkipNEvents option useless.

Given that we will use new alignment anyways (it's been there for probably around two years now) - I wont fix this in the old alignment processors. We can talk about the noisy pixel processors - there we might want to change this.

closed 2018-05-17 by tbisanz