eutelescope/eutelescope #270: Updated GEAR File

tbisanz, 2016-02-03:
Hey,

as we will migrate to a new GEAR file format for the next to the upcoming version of EUTelescope, I wanted to propose some ideas and hopefully get some feedback from you. These changes are motivated by the "newly" introduced telescope geometry description.

Just for reference, here is a GEAR file as it is currently used:

https://github.com/eutelescope/eutelescope/blob/master/jobsub/examples/datura-noDUT/gear_desy2012_150mm.xml

Some points which you should notice: there is this weird "Geometry" parameter in line 11, every layer has a sensitive and a ladder section which more or less (should) contain some values duplicated. Furthermore, there are four rotation (rotation1-4) as well as rotationXY/ZY/ZX. If you even dig further into some other recent applications, you will find weird ID's used for certain planes (magic numbers!) used in GBL.
For the upcoming changes, please keep in mind certain applications: it could be that one detector plane is composed multiple sensors (which need to be aligned independently); Mimosas have Kapton foil before and after the sensor; support structures for the sensors (e.g. Aluminium support plates); other material within the telescope, not attached to a sensor (cooling boxes, etc.).

My initial (maybe naive) proposal would be as following:
```XML
<materials>
        <material name="Kapton" A="10" Z="100" density="700" radLength="100" intLength="100"/>
		<material name="Silicon" A="10" Z="100" density="700" radLength="100" intLength="100"/>
</materials>

<layers>
	<layer ID="0" zPos="100" rotXY="0" rotZY="0" rotZX="0">
		<ladder relZPos="-5" material="Kapton" sizeX="50" sizeY="13"/>
		<ladder material="Silicon" sizeX="50" sizeY="13">
			<sensitive ID="0"
				npixelX="1152" npixelY="576"
				pitchX="18.4" pitchY="18.4"
				posX="-10.6" posY="0"
				rot1="1" rot2="0"
				rot3="0" rot4="1"
				geometry="Miomosa26"
			/>
			<sensitive ID="1"
				npixelX="1152" npixelY="576"
				pitchX="18.4" pitchY="18.4"
				posX="10.6" posY="0"
				rot1="1" rot2="0"
				rot3="0" rot4="1"
				geometry="Miomosa26"
			/>
		<ladder/>
		<ladder relZPos="5" material="Kapton" sizeX="50" sizeY="13"/>
	<layer/>
<layers/>
```
Ignore the actual numeric values (especially the material ones). This example would place two Mimosas as a double chip side by side, with Kapton foil in front and behind them. Every layer is composed of arbitrary ladders, these can be either just material (Kapton) foil or be sensitive, i.e. a detector. I would for simplicity just store rotations for the the whole layer system (as they are linked together). At first glance, this imposes the restriction, that the two Mimosa planes cannot be rotated against each other. This could be corrected by introducing correction terms within the sensitive element, e.g.:
```XML
			<sensitive ID="1"
				npixelX="1152" npixelY="576"
				pitchX="18.4" pitchY="18.4"
				posX="10.6" posY="0"
				rot1="1" rot2="0"
				rot3="0" rot4="1"
				geometry="Miomosa26"
				deltaRotXY="0.6" deltaRotZY="-0.2" deltaRotZX="0.9"
			/>
```
Obviously, such rotations would only be determined by the alignment (Millepede) and thus these values will be automatically filled by EUTelescope. The same goes for uncertainties on the alignment constants (which are all stored in the GEAR file, if you're not familiar with these changes yet).
Additional parameters to the layer element can be introduced, and this is where I need you input. What could we want to specify there? Especially for GBL related stuff!

simonspa, 2016-10-13:
the geometry parameter which you don't explain any further actually reads the TGeo classes, right?

simonspa, 2016-10-13:
Do we still need the confusion of two rotation matrices? one for the sensor orientation (flip) and the Euler angles? Can't we unify this?
I know that this lead to significant user confusion in the past...

tbisanz, 2016-10-13:
Concerning the first question: yes it does. 

The second one: it is renamed to Flip1,2,3,4 now, I'd still prefer to have it in there as it easily allows to flip x and y-axis and (partially, or fully) invert them at the same time. This is then done without the need to apply a rotation matrix.

closed 2019-01-25 by edorossi