eutelescope/eutelescope #280: Eutelescope installation : root donwload fails

andrewjblue, 2016-05-18:
Hi
Tried on 2 different systems and both fail to install due to root failing to download.
Has the link changed to the code? Cheers
Andy

********************************************************************************
*** Downloading sources for ROOT version 5.34.18...
********************************************************************************
--2016-05-18 16:52:05--  ftp://root.cern.ch/root/root_v5.34.18.source.tar.gz
           => â
Resolving root.cern.ch (root.cern.ch)... 137.138.13.143, 2001:1458:201:45::100:b
Connecting to root.cern.ch (root.cern.ch)|137.138.13.143|:21... failed: Connection timed out.
Connecting to root.cern.ch (root.cern.ch)|2001:1458:201:45::100:b|:21... failed: Network is unreachable.

*** ERROR in module [ ROOT ]: DEBUG INFO: {'PLATFORM': 'Linux - unknown', 'GCC_VERSION': 'gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-4)', 'LSB_RELEASE': '', 'UNAME': 'Linux atlasitk1.desy.de 3.10.0-327.13.1.el7.x86_64 #1 SMP Thu Mar 31 11:10:31 CDT 2016 x86_64 x86_64 x86_64 GNU/Linux'}

*** ERROR in module [ ROOT ]: Problems ocurred downloading sources!!

Logfile for failed module: /home/atlas/recon/v01-17-05/.ilcinstall-logs/release-standalone-2016-5-18_14-52.log

andrewjblue, 2016-05-18:
Think i fixed this.
It;s an issue with the root.py file from ilcinstall

It points to a link on ROOT that no longer exists

Should be updated to
self.download.url = 'https://root.cern.ch/download/root_v%s.source.tar.gz' % self.version

pheloniusfriar, 2016-06-02:
Thank you, I just ran into this same problem myself and this saved me a bunch of time!

lime86, 2016-07-14:
Just to mention in the svn version, the root svn also doesn't exist anymore.

andrewjblue, 2016-07-14:
To follow up I emailed the ilc soft guys
They will make the above changes to the trunk

However I don't know it this will also be changed in the eutel branch /tag

tbisanz, 2016-10-05:
Is done!

closed 2016-10-05 by tbisanz