eutelescope/eutelescope #318: Add new libraries in MARLIN_DLL

tbisanz, 2016-10-27:
See discussion in #316

simonspa, 2016-10-28:
To be more precise on this:
the file `ilcinstall/ilcsoft/eutelescope.py` has to be adapted to automatically add the new libraries to the generated `build_env.sh` files.

tbisanz, 2016-12-09:
This is resolved in our new development installer - `https://svnsrv.desy.de/public/ilctools/ilcinstall/branches/v01-17-10-eutel-dev` which should be used to derive new tags. However, I'm still testing it works before closing this issue

tbisanz, 2016-12-09:
Tested and seems to work!

closed 2016-12-09 by tbisanz

chenzhl2018, 2022-07-18:
I can't open https://svnsrv.desy.de/public/ilctools/ilcinstall/branches/v01-17-10-eutel-dev.How should I do?

simonspa, 2022-07-18:
Two steps:

a) Use `ilcinstall` from its new home https://github.com/iLCSoft/iLCInstall
b) Drop EUTelescope and move to Corryvrekckan https://gitlab.cern.ch/corryvreckan/corryvreckan/

:slightly_smiling_face:

jha92, 2022-07-18:
I fully second Simon with urging you to switch to Corry. Otherwise, are you using the "newer" install version of [https://github.com/eutelescope/iLCInstall](https://github.com/eutelescope/iLCInstall)?