eutelescope/eutelescope #324: Poosibility to remove events with corrupt data?

TCThenna, 2016-12-07:
Hi,

I have one event in a large data set, that has pixel 1365/1365 fired, where there are only 1152x512 pixels in the Mimosa26. During converter step, I get an error (which is good):
13:00:53 jobsub.converter(ERROR): [ ERROR5 "HotPixelMasker"]    Pixel: 1365|1365 on plane: 5 fired.

Is there  a way to mark this event as "to be discarded for later analysis"?

During filter step, this event causes problems when the processor tries to retrieve noise info of that pixel:
13:03:29 jobsub.filter(ERROR): *** Break *** segmentation violation

I could retrieve the seg fault to this line:
https://github.com/eutelescope/eutelescope/blob/v1.0-tag/src/EUTelClusterFilter.cc#L952
this is where it tries to access information of a pixel that does not exist. 

Any ideas?

Cheers,
Hendrik

tbisanz, 2016-12-07:
The seg-fault behaviour is expected and will not be "fixed" - with corrupted data we don't gurantee defined behaviour as this would impose a significant overhead. I'd suggest writing a processor which discards events which are corrupted (e.g. have pixels outside the allowed regime). Such processor does not exist right now.

Dirty hacks would be: Make the converter plugin discard those events or hits or make your pixel so large that it actually also contains this pixel. I don't think we should provide any of this by default as it will make users not aware that they are working with corrupted data.

TCThenna, 2016-12-07:
throwing the seg fault is fine, this was just how I noticed it. 

well, after conversion, before clustering, there could be a data integrity test, that manipulates events in a desired way, that are corrupt by certain standards. Could that be something useful for many users?

tbisanz, 2016-12-09:
To be honest - in my opinion, if you have corrupted data something more fundamental is wrong and I think it's dangerous to provide "defaults" to fix this. But this is just my opionon!

simonspa, 2016-12-09:
I would say in principle this is the fault of the EUDAQ NI converter plugin which should take care of only delivering sane data.

TCThenna, 2016-12-12:
It is exactly 1 event in 100 M. Unfortunately, it is the first event of a run, which (for reason unknown to me) seems to not be skip-able. 

So whatever the physical reason is to have this single corrupted event, it might be reasonable to have an option to skip certain events (or blocks of events) from analysis easily (via the config file) -- for all kinds of purposes. If one uses this option afterwards for skipping single upset events or whatever is up to the user of course.

TCThenna, 2016-12-13:
Reflecting your other comments in #253 and #325, being able to skip any combination of events would also be a Marlin feature, not an EUTelescope one. Thanks for clarifying. I might ask Frank about this option.