eutelescope/eutelescope #325: Under which conditions is processRunHeader() called from Marlin?

TCThenna, 2016-12-07:
Hi,

is it correct to say, that ::processRunHeader (LCRunHeader * runHeader) is only called, in case the first event, that is processed after the BORE, is an event with internal event number 0 ?

I noticed, that when the first data event (the one you get from 'dumpevent file 1') is not 0, processRunHeader () is not called. Is that a desired functionality?

TCThenna, 2016-12-07:
probably related with #253 

(and I should update my code ...)

tbisanz, 2016-12-12:
Please check with Marlin, not EUTelescope - according to their Doxygen it's 

> Called for every run, e.g. overwrite to initialize run dependent histograms.

TCThenna, 2016-12-13:
I will check with Frank Gaede, and report here.

TCThenna, 2016-12-15:
This is Frank's answer (sry, in German):

Hallo Hendrik,

processRunHeader() wird immer dann aufgerufen, wenn im LCIO file ein LCRunHeader 
gefunden wir. Da der typischerweise vor dem ersten Event kommt, wird processRunHeader()
nicht mehr aufgerufen, wenn du das erste Event ( oder die n ersten events ) skippst.
In Marlin werden LCIO files sequentiel gelesen.

So gesehen macht eine implementierung der processRun Header Methode nur Sinn, wenn man 
weiss, das es es RunHeader gibt und dies auch immer gelesen werden...

Das hat nichts mit dem SthepReader zu tun - der liest ja kein LCIO file, sondern stdhep
und ruft daher vorm dem ersten event procesRunHeader().

<<<

Carrying out an implementation of a certain functionality in the proceessRunHeader() might therefore be taken with caution, as this is usually not called if the first event got skipped at some point during the reconstruction. I'll check if one can move the code in processRunHeader() for the aligngbl processor to a different place.

TCThenna, 2016-12-15:
What is probably doable is:

void EUTelProcessorApplyAlign::init () {
  [...]
  _isFirstEvent = true;
}

void EUTelProcessorApplyAlign::processEvent (LCEvent * event) {
  if (_isFirstEvent) {
    ... do whatever you want to do for the first event ...
   _isFirstEvent = false;
  }
}

tbisanz, 2016-12-15:
I'd guess that init() is only called once when the processor is initialised, so it doesn't help if you have something like multiple runs. I would instead go for solutions which don't require action on the first event at all but only during initialisation.

TCThenna, 2016-12-15:
if I run two jobs, e.g.
> jobsub.py -c somconfig.cfg -csv somerunlists.csv --subdir trackgbltriplettest 2,3

I see the init() being called twice !

tbisanz, 2016-12-15:
Yes, this probly is two Marlin calls - but if you concatenate runs it might (should) be different. Although, I must admit I'd have to check

edorossi, 2019-01-25:
Hi @jha92 , this is actually a quite crucial topic for EUDAQ2. With the EUDAQ2 converter often times the event number 1 is loss (it is only TLU information) so the processHeader function is not called. The biggest problem is then that the noisy pixels are masked in such a function.
I would leave this issue open until is fixed.

TCThenna, 2019-01-28:
I'd also agree on leaving it open. though I have no solution to the problem. AS Frenk Gaede said, processRunHeader() is only called if there is a LCRunHeader in the LCIO file. Of we cannot guarantee this, than basically processRunHeader() shouldn't be used AT ALL. Which would mean, that the all the functionality that is at the moment implemented in a processRunHeader() method, should be moved to somewhere else ...