eutelescope/eutelescope #337: Lib files not present

LouisDeramo, 2017-01-11:
Hi,
I tried to install Eutelescope following the instruction given [here](http://eutelescope.web.cern.ch/content/installation). But it seems that the lib files from Gear (and Root) are not included in the git repository. 

I attach here the log file (sorry for the French included in the log)

Regards,
-- 
Louis D'Eramo
LPNHE Paris

[release-standalone-2017-1-9_9-3.txt](https://github.com/eutelescope/eutelescope/files/698405/release-standalone-2017-1-9_9-3.txt)

tbisanz, 2017-01-16:
Hey, sorry for the delayed reply - ROOT and Gear (in that order) should be installed by the ilcinstall-script and they are actually found under: /gpfs/t3/atlas/atlas/lderamo/ilcsoft/v01-17-10/root/6.06.06/lib/*

What doesn't work is that the two variables: ROOT_INCLUDE_DIRS PACKAGE_VERSION_COMPATIBLE are not set. Try to set them manually!

LouisDeramo, 2017-01-17:
Hello,
I've decided to reinstall everything from scratch following the instructions given above (avoiding so to import my local lib files for the installation). But it seems that some c++11 errors are raised by CMake (see log file attached).
I tried also to disable the option c++11 but some errors with Pyhtia8 appeared (see log attached). Have you already encounter such errors.

Many Thanks,
-- 
Louis
[release-standalone-c++11False.log.txt](https://github.com/eutelescope/eutelescope/files/710633/release-standalone-c.11False.log.txt)
[release-standalone-c++11True.log.txt](https://github.com/eutelescope/eutelescope/files/710634/release-standalone-c.11True.log.txt)

tbisanz, 2017-01-30:
Hey, the top line in both logs show following message: 

Using g++ (GCC) 4.4.7 20120313 (Red Hat 4.4.7-17) - you need at least gcc 4.8, have you tried it?

LouisDeramo, 2017-01-30:
Ok that pb is solved, but I've now an issue with python. It asks for version 2.7 or newer but I'm indeed running with python 2.7.3. I join the log for more informations.
[log_eutelescope.txt](https://github.com/eutelescope/eutelescope/files/738733/log_eutelescope.txt)

edorossi, 2017-02-20:
Hi,
Using root 6.06.06 I obtain the same error as you. A temporary way to make the installation work is to install root 5.34.30 instead. You can modify the version you install in release-versions.py.

tbisanz, 2017-08-18:
We have reworked all of the installers, see http://eutelescope.web.cern.ch/content/installation - also check out our awesome Dockerfiles if it doesn't work! Please open a new issue if still encountering errors!

closed 2017-08-18 by tbisanz