eutelescope/eutelescope #342: Compilation failure (at GEAR) with v01-17-10-eutel-dev

jackylie, 2017-03-30:
Hello, 
I am trying to install the reconstruction software for DESY testbeam data taken in March 2017, using the branch "v01-17-10-eutel-dev". However, the compilation fails at GEAR (TrackerPlanesParametersImpl.cc) - error log attached. I am installing on lxplus, using ROOT 5.34.30, gcc 4.8.1, python 2.7.4, use_cpp=true in this setup.

Many thanks for any help! 

[release-standalone-2017-3-28_15-29.txt](https://github.com/eutelescope/eutelescope/files/882551/release-standalone-2017-3-28_15-29.txt)

jackylie, 2017-04-13:
Hello,
I can now compile on lxplus by doing these modifications in the file TrackerPlanesLayerLayoutImpl.h (under $ILCSOFT/v01-17-10/gear/v01-06-eutel-pre/include/gear/gearimpl/):

line 614, change 
 		      , (*ml).getMaterial()
to 
 		      , (*ml).getMaterial().c_str()
and line 632, change 
		      , (*sl).getMaterial()
to
		      , (*sl).getMaterial().c_str()

They seem to be a bug before std::string and and char* when passing as arguments in a function. 

(Setup used: SLC6, ROOT 5.34.30, gcc 4.9.3)

tbisanz, 2017-08-18:
We only support gcc 5.x, if you're using SLC 6 or CernCentOS 7 please use devtoolset-4 or newer!

closed 2017-08-18 by tbisanz