eutelescope/eutelescope #343: Processing Radial Strips

andrewjblue, 2017-04-10:
Hi 
As discussed today we would like to be able reconstruct data taken with radial strip sensors at testbeam.

We successfully reconstructed axial strips data from last year using GBL alignment.

However are unsure what changes need to be made to the code to process radial strips

Thanks

Andy

tbisanz, 2017-04-12:
Can you give us some more info?

andrewjblue, 2017-04-12:
HI Tobias
Later today/tomorrow we should have on CDS a document on what we did last year with EUTelescope to process the "regular" Strips.  So for us the only difference will be the sensor geometry for radial strisp, which we also provide v soon

I can email the document just now if that helps

Andy

tbisanz, 2017-04-24:
Can you upload the documentation here?

andrewjblue, 2017-04-24:
[EUTelescope_Guide.pdf](https://github.com/eutelescope/eutelescope/files/952449/EUTelescope_Guide.pdf)

As attached

Thanks

Andy

dreyling, 2019-01-09:
That is individually solved for the ATLAS ITk user.

closed 2019-01-09 by dreyling