eutelescope/eutelescope #344: How to recompile eudaq against eutel after changes of a plugin

TCThenna, 2017-04-11:
Dear all,

I tried to recompile eudaq (v1.2.2) against eutelescope in order to update a data conversion plugin following these instructions:
http://eutelescope.web.cern.ch/content/updating (at the bottom)

After switching the USE_CXX11 option ON in the Platform.cmake, I could compile eudaq, but it does not link against eutel:

hjansen@nafhh-cms01:build]$ make install
[ 69%] Built target EUDAQ
[ 80%] Built target NativeReader
Linking CXX executable euLog.exe
../main/lib/libEUDAQ.so: undefined reference to `eutelescope::EUTelAPIXMCDetector::EUTelAPIXMCDetector(unsigned short)'
../main/lib/libEUDAQ.so: undefined reference to `eutelescope::EUTelMimosa26Detector::EUTelMimosa26Detector()'
../main/lib/libEUDAQ.so: undefined reference to `eutelescope::EUTelMimoTelDetector::EUTelMimoTelDetector()'
../main/lib/libEUDAQ.so: undefined reference to `eutelescope::EUTelFortisDetector::EUTelFortisDetector()'
../main/lib/libEUDAQ.so: undefined reference to `eutelescope::EUTelTakiDetector::EUTelTakiDetector()'
../main/lib/libEUDAQ.so: undefined reference to `eutelescope::EUTelMimosa18Detector::EUTelMimosa18Detector()'
../main/lib/libEUDAQ.so: undefined reference to `eutelescope::EUTelTimepixDetector::EUTelTimepixDetector()'
../main/lib/libEUDAQ.so: undefined reference to `eutelescope::EUTelDEPFETDetector::EUTelDEPFETDetector()'
../main/lib/libEUDAQ.so: undefined reference to `eutelescope::EUTelSetupDescription::EUTelSetupDescription(eutelescope::EUTelPixelDetector*)'
collect2: error: ld returned 1 exit status
make[2]: *** [gui/euLog.exe] Error 1
make[1]: *** [gui/CMakeFiles/euLog.exe.dir/all] Error 2
make: *** [all] Error 2
[hjansen@nafhh-cms01:build]$

any ideas?

Thanks for your help.

Cheers,
Hendrik

TCThenna, 2017-04-11:
I leave this here for reference:

I forgot to turn off the gui, so the solution is to
cmake USE_LCIO=1 USE_EUTELESCOPE=1 -DBUILD_main=OFF -DBUILD_nreader=ON -DBUILD_gui=OFF ..
make install -j

in any case, the http://eutelescope.web.cern.ch/content/updating require an update, I guess. Maybe we close the issue only after the update was done?

tbisanz, 2017-04-12:
That is rebuilding EUDAQ, EUTelescope doesn't depent on EUDAQ (build wise)

closed 2017-08-18 by TCThenna