eutelescope/eutelescope #348: EUTelescope Installation

tbisanz, 2017-05-08:
A new installer is temporarily available in my iLCInstall github fork - to install, make a directory for your installation, cd into it and run:

`
export ILCSOFT=$PWD
git clone https://github.com/tbisanz/iLCInstall
cd iLCInstall
git checkout dev
./ilcsoft-install -i examples/eutelescope/release-standalone.cfg
`

I'll be collecting issues in the comments below

tbisanz, 2017-05-08:
Issues with SLC6: you need devtoolset for gcc and python27. Even then, the $PATH variables don't get updated correctly (maybe because of the SCL shell for the developer packages?) - you need to manually adjust them during installation! Then the CMake files won't properly find the include dirs

j1990grossmann, 2017-06-12:
Do you use devtoolset 4 ? Do you have a full gcc installation, otherwise ROOT will not compile and work correctly. https://sft.its.cern.ch/jira/browse/ROOT-6207 . Currently I want to setup a new installation of eutelescope on slc6 wit optionally with ROOT 6, a new gcc and python >2.7. What other variables do have to be set apart from $PATH pointing to /opt/rh/devtoolset-4/root/usr/bin/ etc. I had problems with pyhton path as well and it did not work properly in the end.

arummler, 2017-07-16:
Ubuntu 16.04
requires GSL_path = "/usr/lib/x86_64-linux-gnu"
fails during LCIO compilation:
/home/rummler/Software/eutelescope/installed_master/v01-19-02/lcio/master/build/rootdict/UTIL.cxx: In function 'void {anonymous}::TriggerDictionaryInitialization_UTIL_Impl()':
/home/rummler/Software/eutelescope/installed_master/v01-19-02/lcio/master/build/rootdict/UTIL.cxx:8153:38: error: 'R' was not declared in this scope
     static const char* fwdDeclCode = R"DICTFWDDCLS(
                                      ^
UTIL dictionary forward declarations' payload:5:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:7:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:9:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:10:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:11:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:12:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:13:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:14:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:15:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:16:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:17:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:18:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:19:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:20:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:21:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:22:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:23:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:24:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:25:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:26:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:27:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:28:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:29:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:30:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:31:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:32:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:33:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:34:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:35:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:36:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:37:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:38:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:39:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:40:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:41:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:42:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:43:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:44:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:45:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:46:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:47:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:48:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:49:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:50:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:51:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:52:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:53:1: error: 'namespace' definition is not allowed here
UTIL dictionary forward declarations' payload:54:1: error: 'namespace' definition is not allowed here
[... cut ...]  'Linux vir 4.8.0-58-generic #63~16.04.1-Ubuntu SMP Mon Jun 26 18:08:51 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux'}

*** ERROR in module [ LCIO ]: At least one of these files: ['/home/rummler/Software/eutelescope/installed_master/v01-19-02/lcio/master/lib/liblcio.a', '/home/rummler/Software/eutelescope/installed_master/v01-19-02/lcio/master/lib/liblcio.so', '/home/rummler/Software/eutelescope/installed_master/v01-19-02/lcio/master/lib/liblcio.dylib']
is required for using this installation of LCIO

tbisanz, 2017-08-18:
Completely reworked installers, see http://eutelescope.web.cern.ch/content/installation

closed 2017-08-18 by tbisanz