eutelescope/eutelescope #352: argparse problem with "master" installation

kwraight, 2017-05-24:
Trying SL6 installation following docker file description.

Get the following output when trying to install EUTelescope svn version:
svn co https://svnsrv.desy.de/public/ilctools/ilcinstall/branches/v01-17-10-eutel-dev/ $ILCSOFT/ilcinstall_v01-17-10-eutel 

Problem with load in argparse from link...

--2017-05-23 17:11:44--  http://argparse.googlecode.com/files/argparse-1.2.1.tar.gz
Resolving argparse.googlecode.com... 66.102.1.82, 2a00:1450:400c:c06::52
Connecting to argparse.googlecode.com|66.102.1.82|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2017-05-23 17:11:45 ERROR 404: Not Found.

Extracting tar archive...
tar (child): /afs/phas.gla.ac.uk/user/k/kwraight/telescopeStuff/AprilVersion/v01-17-10/Eutelescope/master/external/packages.tmp/argparse-1.2.1.tar.gz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
tar: Error is not recoverable: exiting now
... done with argparse python module

kwraight, 2017-05-24:
replace with link:
https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/argparse/argparse-1.2.1.tar.gz

freidt, 2017-05-24:
@kwraight did you run the installation with the updated link? Does it change anything for you?

I normally simply ignored this error on SLC6 as the installation was working anyhow.

kwraight, 2017-05-24:
@freidt 
Yep, I think if you have a newer version of Python it's not a problem.
I updated $ILCSOFT/v01-17-10/Eutelescope/master/tools/install-externals/install-externals.sh L27:
#wget --no-check-certificate -P "$tempdir" 'http://argparse.googlecode.com/files/'$argparsefile
wget --no-check-certificate -P "$tempdir" 'https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/argparse/'$argparsefile

Now the tar file is found and extracted. However, installation still fails due to another issue #353

closed 2017-05-24 by kwraight