eutelescope/eutelescope #359: Compilation failure (against GEAR) with v01-19-02

j1990grossmann, 2017-07-08:
None of the gear versions provided in eutelescope/gear compiles against the current master branch of eutelescope. There should be a new version somewhere around. Currently trying a fix with v01-06-eutel

tbisanz, 2017-08-18:
GEAR compiles fine with gcc 5.x which is available from the devtoolset-4 and newer under SLC 6 and CernCentOS 7. However, there are some issues with incorrect string parsing and stuff like that.

I'll leave this issue open for now - since we SHOULD fix it nevertheless, but please note that we only support gcc 5 and newer! See instructions here: http://eutelescope.web.cern.ch/content/installation

julienbeyer, 2017-08-31:
Using gcc version 5.3.1:

[ 77%] Building CXX object src/CMakeFiles/testFTD.dir/test/testFTD.cc.o
/mnt/Data_storage/software/v01-19-02/gear/v01-06-eutel-pre/src/test/testFTD.cc:18:25: fatal error: TGeoManager.h: No such file or directory
compilation terminated.
src/CMakeFiles/testFTD.dir/build.make:62: recipe for target 'src/CMakeFiles/testFTD.dir/test/testFTD.cc.o' failed
make[2]: *** [src/CMakeFiles/testFTD.dir/test/testFTD.cc.o] Error 1
CMakeFiles/Makefile2:1738: recipe for target 'src/CMakeFiles/testFTD.dir/all' failed
make[1]: *** [src/CMakeFiles/testFTD.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

*** ERROR in module [ GEAR ]: DEBUG INFO: {'PLATFORM': 'Linux - unknown', 'GCC_VERSION': 'gcc (GCC) 5.3.1 20160406 (Red Hat 5.3.1-6)', 'LSB_RELEASE': '', 'UNAME': 'Linux  4.5.6-200.fc23.x86_64 #1 SMP Wed Jun 1 21:28:20 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux'}

*** ERROR in module [ GEAR ]: At least one of these files: ['/mnt/Data_storage/software/v01-19-02/gear/v01-06-eutel-pre/lib/libgear.a', '/mnt/Data_storage/software/v01-19-02/gear/v01-06-eutel-pre/lib/libgear.so', '/mnt/Data_storage/software/v01-19-02/gear/v01-06-eutel-pre/lib/libgear.dylib']
is required for using this installation of GEAR

Any advice?

j1990grossmann, 2017-08-31:
Is everything ok with your GEAR installation? 
Compiling the current master branch never worked for me out of the box with the gear versions provided in https://github.com/eutelescope/gear/ . There are self made bug fixes (see pull request eutelescope/gear) in the gear packages from https://github.com/eutelescope/gear/tree/v01-06-eutel . 
Note that there is some work going on for a docker installation of the software. There is also a new installer around. Maybe you should try this.

julienbeyer, 2017-08-31:
Maybe I should have explained it a bit more. First of all, thanks for the response and your hint. I was trying to install the newest version of the Eutelescope package tagged with v01-19-02 - so I am already trying to use the newest installer. However, the installation fails at this point which is the compilation of gear.

julienbeyer, 2017-08-31:
I have just seen that the first problem happens a bit early, still during the installation of gear:

[ 32%] Building CXX object src/geartgeo/CMakeFiles/geartgeo.dir/TGeoGearDistanceProperties.cc.o
Scanning dependencies of target gearxml
[ 33%] Building CXX object src/CMakeFiles/gearxml.dir/gearxml/TrackerPlanesParametersXML.cc.o
/mnt/Data_storage/software/v01-19-02/gear/v01-06-eutel-pre/src/geartgeo/TGeoGearDistanceProperties.cc:6:30: fatal error: TVirtualGeoTrack.h: No such file or directory
compilation terminated.
src/geartgeo/CMakeFiles/geartgeo.dir/build.make:110: recipe for target 'src/geartgeo/CMakeFiles/geartgeo.dir/TGeoGearDistanceProperties.cc.o' failed
make[2]: *** [src/geartgeo/CMakeFiles/geartgeo.dir/TGeoGearDistanceProperties.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/mnt/Data_storage/software/v01-19-02/gear/v01-06-eutel-pre/src/geartgeo/TGeoGeometryInitializer.cc:6:24: fatal error: TGDMLParse.h: No such file or directory
compilation terminated.
src/geartgeo/CMakeFiles/geartgeo.dir/build.make:158: recipe for target 'src/geartgeo/CMakeFiles/geartgeo.dir/TGeoGeometryInitializer.cc.o' failed
make[2]: *** [src/geartgeo/CMakeFiles/geartgeo.dir/TGeoGeometryInitializer.cc.o] Error 1
/mnt/Data_storage/software/v01-19-02/gear/v01-06-eutel-pre/src/geartgeo/TGeoGearPointProperties.cc:8:24: fatal error: TGeoVolume.h: No such file or directory
compilation terminated.
src/geartgeo/CMakeFiles/geartgeo.dir/build.make:182: recipe for target 'src/geartgeo/CMakeFiles/geartgeo.dir/TGeoGearPointProperties.cc.o' failed
make[2]: *** [src/geartgeo/CMakeFiles/geartgeo.dir/TGeoGearPointProperties.cc.o] Error 1
CMakeFiles/Makefile2:2201: recipe for target 'src/geartgeo/CMakeFiles/geartgeo.dir/all' failed
make[1]: *** [src/geartgeo/CMakeFiles/geartgeo.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 34%] Linking CXX shared library ../lib/libgearxml.so
[ 53%] Built target gearxml
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
[  7%] Built target gearsurf
[ 29%] Built target gear
[ 50%] Built target gearxml
[ 51%] Linking CXX executable ../bin/testMaterialBudget
[ 52%] Built target testMaterialBudget
[ 53%] Linking CXX executable ../bin/testInteractionLengths
[ 54%] Built target testInteractionLengths
[ 55%] Linking CXX executable ../bin/testtpcproto
[ 56%] Built target testtpcproto
[ 57%] Linking CXX executable ../bin/testGearMgr
[ 58%] Built target testGearMgr
[ 59%] Linking CXX executable ../bin/printMaterials
[ 60%] Built target printMaterials
[ 61%] Linking CXX executable ../bin/testSiPlanesgear
[ 62%] Built target testSiPlanesgear
[ 63%] Linking CXX executable ../bin/mergeXML
[ 65%] Built target mergeXML
[ 66%] Linking CXX executable ../bin/printParameters
[ 68%] Built target printParameters
[ 68%] Linking CXX executable ../bin/testVTXgear
[ 69%] Built target testVTXgear
[ 70%] Linking CXX executable ../bin/testXMLCopy
[ 71%] Built target testXMLCopy
[ 72%] Linking CXX executable ../bin/testMaterialBudgetNew
[ 72%] Built target testMaterialBudgetNew
[ 73%] Linking CXX executable ../bin/testMaterialMap
[ 74%] Built target testMaterialMap
[ 75%] Linking CXX executable ../bin/testIsInsideModule
[ 76%] Built target testIsInsideModule
[ 77%] Building CXX object src/CMakeFiles/testFTD.dir/test/testFTD.cc.o
/mnt/Data_storage/software/v01-19-02/gear/v01-06-eutel-pre/src/test/testFTD.cc:18:25: fatal error: TGeoManager.h: No such file or directory
compilation terminated.
src/CMakeFiles/testFTD.dir/build.make:62: recipe for target 'src/CMakeFiles/testFTD.dir/test/testFTD.cc.o' failed
make[2]: *** [src/CMakeFiles/testFTD.dir/test/testFTD.cc.o] Error 1
CMakeFiles/Makefile2:1738: recipe for target 'src/CMakeFiles/testFTD.dir/all' failed
make[1]: *** [src/CMakeFiles/testFTD.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

However, still it looks like some files are missing rather than there is really a compilation problem which would require the "Hotfix" mentioned before. Any other ideas?

j1990grossmann, 2017-08-31:
Which installer do you use? Where does your GEAR package come from? SVN /GIT ? GEAR does not have so many external dependencies. A missing  TGeoManager.h may come from incomplete ROOT Environment Variables or missing or incomplete ROOT installation. If you want you can compile GEAR only with the environment set by ILCSoft.cmake.env.sh and build_env.sh in the gear directory. Check if cmake is aware of ROOT and recompile. (By construction the installer tries to compile everything even if your external dependencies are not resolved or your environment is broken. It just compiles as far as it can. For the error messages you have to search the installation log file. )

julienbeyer, 2017-08-31:
What exactly do you mean with which installer I use?! The version is v01-19-02, the version you get if you would now install it as it is suggested on the eutelescope homepage.

To me, it looks like an incomplete installation of root since in the root/geo folder where the missing file should be located, it is not. Any idea why the root installation would be incomplete without giving any noticable error? How can I reinstall root, maybe with another version?

julienbeyer, 2017-08-31:
BTW: root should have been installed in version 6.10.04

j1990grossmann, 2017-08-31:
https://github.com/eutelescope/iLCInstall/
This is the repository for the current installer mainly the Tags are different in the correct set of versions for the subpackages (GEAR, GBL, ... ) from wherever they come from (git, svn, wget) .  If you do not want to recompile ROOT (or load other packages like Eigen3... ) during installation, I recommend to go through the installer configure the installer to symlink to ROOT etc. . ROOT you may get evtl. via CVMFS. Otherwise you can reinstall a different ROOT version by hand. (The installer think there is already something there so it does not remove and replace everything after the ROOT directory is created.) For me 6.08/02 works fine via CVMFS .

julienbeyer, 2017-09-01:
Ahh, I have tried to use the dev-base branch, if that is what you mean. Can you perhaps explain:

 I recommend to go through the installer configure the installer to symlink to ROOT etc. .

a bit more. How can I configure to install everything besides root and use the system wide root?

edorossi, 2019-01-21:
The problem has not been seen with the latest installation scripts. I close the issue and we can reopen it if the problem appears again.

closed 2019-01-21 by edorossi