eutelescope/eutelescope #361: TelescopeWithDUT

sqytau, 2017-07-09:
EUTelTestFitter.cc
Line 574
_planeThickness[iz]=_siPlanesLayerLayout->getLayerThickness(ipl) + _siPlanesLayerLayout->getSensitiveThickness(ipl) ;
Should not it be?:
_planeThickness[iz]=_planeThickness[iz]=_siPlanesLayerLayout->getDUTThickness() + _siPlanesLayerLayout->getDUTSensitiveThickness() ;

For TelescopeWithDUT, at some iteration ipl is equal to _iDUT and _siPlanesLayerLayout->getLayerThickness(ipl) cause segmentation fault.

Another question is about track fitting position in DUT. Here are parts of the code which might do something not consistently.

1889        pos[0]=fittedX[_nTelPlanes*ifit+ipl];
1890        pos[1]=fittedY[_nTelPlanes*ifit+ipl];
1891        pos[2]=_planePosition[ipl];
1892
1893        getFastTrackImpactPoint(pos[0], pos[1], pos[2], fittrack, event);
        this function (getFastTrackImpactPoint) calls: int sensorID = guessSensorID( pos );
        and then uses the position of sensorID for position calculation.
        That sensorID is never corresponds to DUT, so that pos[3] of DUT is changed to pos[3] of a telescope plane closest to DUT 
        and in the end pos[1] and pos[2] do not match pos[3] when ipl==_iDUT.
Later:        
1986        // Use position at DUT as a track reference point.
1987        // If no DUT present: use position in the first plane !
1988
1989        if(ipl==_iDUT || (_iDUT<0 && ipl==0)) 
1990        {
1991          for(int iref=0;iref<3;iref++) 
1992          {
1993            refpoint[iref]=pos[iref];
1994          }
1995        }
1996      }
1997      // Store track reference point.
1998      fittrack->setReferencePoint(refpoint);

It gives something strange for reference point with x, y corresponding to track fitting in DUT and z of the telescope plane closest to DUT.

TCThenna, 2017-07-12:
There is a new datura-noDUT example in the master, which works and gives nice results using the EUTelTripletGBL processor. 

Tobias should correct me, but the EUTelTestFitter might not be maintained in the future. 

Hendrik

TCThenna, 2017-08-18:
you mentioned missing eigen library in a private email. Please try the new ilcsoft installer 
   http://eutelescope.web.cern.ch/content/installation
and note the prerequisites. 

@tbisanz is the EUTelTestFitter.cc deprecated ?

tbisanz, 2017-08-18:
Side note: you will need a system wide installation of eigen3, see updated instructions here: http://eutelescope.web.cern.ch/content/installation and our docker files :)!

edorossi, 2019-01-21:
I close this issue since there has not been updates in quite some time.

closed 2019-01-21 by edorossi