eutelescope/eutelescope #362: EutelMilleGBL broken?

j1990grossmann, 2017-07-20:
EutelMilleGBL seems to be broken. Otherwise there should be a more detailed example how to use the current version. I tried the example given in jobsub/examples/datura-noDUT.

After performing the following steps:

    converter
    clustering
    filtering
    hitmaker
    hitlocal
    aligngbl

I get an assertion error from Eigen in line 1605 when calling
traj.getResults( ipos, aCorrection, aCovariance );

The error indicates that something is quite wrong with the data given to GBL. Unfortunately I am not the expert with this processor. Maybe it helps already to update the examples! What are the necessary switches and steps to be performed before applying eutelmillegbl to align the telescope. In the current master version all the Trackfinding processors disappeared or moved. Are those still relevant. Where are the non compatible hits removed for track candidates now?

Below there is some info and backtrace on the bug that I obtained using gdb.

    Warning in UnknownClass::SetDisplay: DISPLAY not set, setting it to hdpc195.local.hephy.at:0.0

    [ MESSAGE "Marlin"] ---- instantiated GEAR from file /afs/hephy.at/user/j/jgrossmann/ILCSOFT/v01-19-02/Eutelescope/master/jobsub/examples/datura-noDUT/gear_desy2012_150mm.xml
    [ MESSAGE "Marlin"] -----------------------------------------------------
    [ MESSAGE "Marlin"] GearMgr :
    [ MESSAGE "Marlin"] -----------------------------------------------------
    [ MESSAGE "Marlin"] ---- DetectorName ----
    [ MESSAGE "Marlin"]
    [ MESSAGE "Marlin"] EUTelescope
    [ MESSAGE "Marlin"]
    [ MESSAGE "Marlin"]
    [ MESSAGE "Marlin"] ----------- BField -------
    [ MESSAGE "Marlin"]
    [ MESSAGE "Marlin"] field vector at origin : Bx = 0 , By = 0 , Bz = 0 [Tesla]
    [ MESSAGE "Marlin"]
    [ MESSAGE "Marlin"] ---- SiPlanes ----
    [ MESSAGE "Marlin"]
    [ MESSAGE "Marlin"] ----------- SiPlanesParameters -------
    [ MESSAGE "Marlin"]
    [ MESSAGE "Marlin"] [StringVec] Geometry Mimosa26.so , Mimosa26.so , Mimosa26.so , Mimosa26.so , Mimosa26.so , Mimosa26.so ,
    [ MESSAGE "Marlin"]
    [ MESSAGE "Marlin"] Setup ID : 0
    [ MESSAGE "Marlin"] Telescope type : TelescopeWithoutDUT
    [ MESSAGE "Marlin"] Number of telescope planes : 6
    [ MESSAGE "Marlin"] Number of layers : 6
    [ MESSAGE "Marlin"]
    [ MESSAGE "Marlin"] layer parameters :
    [ MESSAGE "Marlin"] |----------------------------------------------------------------------------------------------------------------------------------------------------------------------|
    [ MESSAGE "Marlin"] | ladder: | sensitive part: |
    [ MESSAGE "Marlin"] | ID | pozX| pozY| pozZ |rotXY|rotZX|rotZY| sizeX| sizeY| Thick| ID | pozX| pozY| pozZ |sizeX|sizeY| Thick|NpixX|NpixY|PitchX|PitchY| Resol| Rot1| Rot2| Rot3| Rot4|
    [ MESSAGE "Marlin"] |----------------------------------------------------------------------------------------------------------------------------------------------------------------------|
    [ MESSAGE "Marlin"] | 0| 0.00| 0.00| 0.00| 0.00| 0.00| 0.00| 21.20| 10.60| 0.036| 0| 0.00| 0.00| 0.00|21.20|10.60| 0.014| 1152| 576| 0.02| 0.02|0.0045|-1.00| 0.00| 0.00|-1.00|
    [ MESSAGE "Marlin"] | 1| 0.00| 0.00| 150.00| 0.00| 0.00| 0.00| 21.20| 10.60| 0.036| 1| 0.00| 0.00| 150.00|21.20|10.60| 0.014| 1152| 576| 0.02| 0.02|0.0045|-1.00| 0.00| 0.00|-1.00|
    [ MESSAGE "Marlin"] | 2| 0.00| 0.00| 300.00| 0.00| 0.00| 0.00| 21.20| 10.60| 0.036| 2| 0.00| 0.00| 300.00|21.20|10.60| 0.014| 1152| 576| 0.02| 0.02|0.0045|-1.00| 0.00| 0.00|-1.00|
    [ MESSAGE "Marlin"] | 3| 0.00| 0.00| 450.00| 0.00| 0.00| 0.00| 21.20| 10.60| 0.036| 3| 0.00| 0.00| 450.00|21.20|10.60| 0.014| 1152| 576| 0.02| 0.02|0.0045|-1.00| 0.00| 0.00|-1.00|
    [ MESSAGE "Marlin"] | 4| 0.00| 0.00| 600.00| 0.00| 0.00| 0.00| 21.20| 10.60| 0.036| 4| 0.00| 0.00| 600.00|21.20|10.60| 0.014| 1152| 576| 0.02| 0.02|0.0045|-1.00| 0.00| 0.00|-1.00|
    [ MESSAGE "Marlin"] | 5| 0.00| 0.00| 750.00| 0.00| 0.00| 0.00| 21.20| 10.60| 0.036| 5| 0.00| 0.00| 750.00|21.20|10.60| 0.014| 1152| 576| 0.02| 0.02|0.0045|-1.00| 0.00| 0.00|-1.00|
    [ MESSAGE "Marlin"] |----------------------------------------------------------------------------------------------------------------------------------------------------------------------|
    [ MESSAGE "Marlin"]
    [ MESSAGE "Marlin"]
    [ MESSAGE "Marlin"] parameter sections (detectors) :
    [ MESSAGE "Marlin"] ------------------------------------------------------------------
    [ MESSAGE "Marlin"] materials :
    [ MESSAGE "Marlin"] ------------------------------------------------------------------
    [ MESSAGE "Marlin"]
    [ MESSAGE "MyAIDAProcessor"]
    [ MESSAGE "MyAIDAProcessor"] ---- MyAIDAProcessor - parameters:
    [ MESSAGE "MyAIDAProcessor"] Compress: 1
    [ MESSAGE "MyAIDAProcessor"] FileName: ./output/histograms/run000097-alignmentGBL_newGBL
    [ MESSAGE "MyAIDAProcessor"] FileType: root
    [ MESSAGE "MyAIDAProcessor"] -------------------------------------------------
    [ VERBOSE "MyAIDAProcessor"] ---------------------------------- R A I D A ----------------------------------
[... 228 lines cut ...]
    [ VERBOSE "MyEUTelMilleGBL"] deriv 3: 0.031746 -0.0432403 0.0114943
    [ VERBOSE "MyEUTelMilleGBL"] measurement at label 16 of type 2 from row 0: 0, 6.4446e+08
    [ VERBOSE "MyEUTelMilleGBL"] param 3: 27 29 31
    [ VERBOSE "MyEUTelMilleGBL"] deriv 3: 0.0114943 -0.0432403 0.031746
    [ VERBOSE "MyEUTelMilleGBL"] measurement at label 16 of type 2 from row 1: 0, 6.4446e+08
    [ VERBOSE "MyEUTelMilleGBL"] param 3: 28 30 32
    [ VERBOSE "MyEUTelMilleGBL"] deriv 3: 0.0114943 -0.0432403 0.031746
    Marlin: /cvmfs/ilc.desy.de/sw/Eigen/3.2.9/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:337: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::DenseCoeffsBase<Derived, 1>::Index, Eigen::DenseCoeffsBase<Derived, 1>::Index) [with Derived = Eigen::Matrix<double, -1, -1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = double; Eigen::DenseCoeffsBase<Derived, 1>::Index = long int]: Assertion `row >= 0 && row < rows() && col >= 0 && col < cols()' failed.

    Catchpoint 1 (signal SIGABRT), 0x0000003233832495 in raise () from /lib64/libc.so.6
    (gdb) bt
    #0 0x0000003233832495 in raise () from /lib64/libc.so.6
    #1 0x0000003233833c75 in abort () from /lib64/libc.so.6
    #2 0x000000323382b60e in __assert_fail_base () from /lib64/libc.so.6
    #3 0x000000323382b6d0 in __assert_fail () from /lib64/libc.so.6
    #4 0x00007ffff0a78793 in Eigen::DenseCoeffsBase<Eigen::Matrix<double, -1, -1, 0, -1, -1>, 1>::operator() (this=0x7fffffff7c70, row=0, col=4) at /cvmfs/ilc.desy.de/sw/Eigen/3.2.9/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:336
    #5 0x00007ffff0a9e757 in gbl::GblTrajectory::getJacobian (this=0x7fffffff8480, aSignedLabel=4) at /afs/hephy.at/user/j/jgrossmann/ILCSOFT/v01-19-02/GBL/V02-01-00/src/GblTrajectory.cpp:566
    #6 0x00007ffff0a9f63e in gbl::GblTrajectory::getResults (this=0x7fffffff8480, aSignedLabel=4, localPar=..., localCov=...) at /afs/hephy.at/user/j/jgrossmann/ILCSOFT/v01-19-02/GBL/V02-01-00/src/GblTrajectory.cpp:734
    #7 0x00007fffe6657fd2 in eutelescope::EUTelMilleGBL::processEvent (this=0x1df84c0, event=) at /afs/hephy.at/user/j/jgrossmann/ILCSOFT/v01-19-02/Eutelescope/master/processors/src/EUTelMilleGBL.cc:1605
    #8 0x00007ffff7dcd59a in marlin::ProcessorMgr::processEvent(EVENT::LCEvent*) () from /afs/hephy.at/user/j/jgrossmann/ILCSOFT/v01-19-02/Marlin/v01-11/lib/libMarlin.so.1.12
    #9 0x00007ffff79048fd in SIO::SIOReader::readStream(int) () from /afs/hephy.at/user/j/jgrossmann/ILCSOFT/v01-19-02/lcio/master/lib/liblcio.so.2.8
    #10 0x0000000000412db0 in main ()
    (gdb) f 7
    #7 0x00007fffe6657fd2 in eutelescope::EUTelMilleGBL::processEvent (this=0x1df84c0, event=) at /afs/hephy.at/user/j/jgrossmann/ILCSOFT/v01-19-02/Eutelescope/master/processors/src/EUTelMilleGBL.cc:1605
    1605 traj.getResults( ipos, aCorrection, aCovariance );

j1990grossmann, 2017-07-20:
This was a comment by Andre and I am fully aware about the problem. Unfortunately I could not find the reason why the matrix is not initialized correctly:
> 
> Besides:
> Your backtrace clearly shows the problem:
> 
> Assertion `row >= 0 && row < rows() && col >= 0 && col < cols()' failed.
> 
> The reason for this is, that you call the Eigen::Matrix constructor
> 
>     template<typename _Scalar, int _Rows, int _Cols, int _Options, int _MaxRows, int _MaxCols>
>

j1990grossmann, 2017-07-25:
Some further notes: 
I fear that this is maybe not the right repository to put this issue. The reason for the error is because the number of local derivatives for all the gbl::GblPoint 's is 0 . The Method GblTrajectory::getResults does not check against it. I think a check should may be somewhere in GblTrajectory::construct() . If GblTrajectory::getResults is called and the number of local derivatives  equals 0 it leads to error in assertion.

j1990grossmann, 2017-08-04:
Apparently some work was done to fix the glb_local example. I merged and tested the example. With the current master branch and the gbl version bundled to it. The current version has to have any local derivatives in order to call GblTrajectory::geResults . A hotfix is to comment out the lines where histograms are filled. For the future and better readability maybe some of the functions and structures from the kink estimator processor could be merged. Millepede shows no errors.

tbisanz, 2017-08-04:
Sorry for replying so little - I'm rather busy with other stuff as well. So the changes were that GBL didn't properly initialise varibales in a std::array - which is fixed in the new version of GBL!  As for the future of GBL, @TCThenna and Paul Schuetze may be interested, as they are the maintainers of this code!

j1990grossmann, 2017-08-04:
The error is fixed with GBL Version V02-01-02 . Example  examples/gbl_local works !

closed 2017-08-04 by j1990grossmann