eutelescope/eutelescope #374: Increased number of hits in Mimosa after update

agisen, 2017-10-23:
I updated my Eutelescope from v1.0 to master (installed with ilcsoft v01-17-10 and v01-19-02), it worked out-of-the-box with gcc 5.
Now I reconstructed data from Oct.16 and Oct.17 with both versions. With the new one I get ~2x the number of hits in the Mimosa planes compared to the number of hits I get in the old version.
This can clearly be seen in clustering step and logically influences all subsequent steps as well.

No differences are observed for the APIX sensors.

Maybe this is a feature and not a bug, but at least I would like to understand this changed behaviour.

tbisanz, 2017-10-23:
Please check with `dumpevent ` what exactly the difference is!

agisen, 2017-10-23:
Please find attached the output of dumpevent 1 on the particular converter.slcio file from the old and the new Eutelescope version.
One can clearly see more hits in the Mimosas.

[dumpevent_1_old.txt](https://github.com/eutelescope/eutelescope/files/1408016/dumpevent_1_old.txt)
[dumpevent_1_new.txt](https://github.com/eutelescope/eutelescope/files/1408017/dumpevent_1_new.txt)

tbisanz, 2017-10-24:
What do you mean by "old" version?

agisen, 2017-10-24:
I updated my Eutelescope from v1.0 tag (old) to master branch (new, from 2017-08-17), both installed with ilcsoft installer, either v01-17-10 or v01-19-02.
I used exactly the same .raw, config, gear and steering files, but getting different results.

tbisanz, 2017-10-24:
Can you attach you steering template and config file?

tbisanz, 2017-10-24:
Ahhh - I think I understand what happens here. Plese see here: https://github.com/eudaq/eudaq/commit/b8e5896b2f95c32478c25c7b67f883ac3218e32c#diff-62ef851bcf4aa5af70a6a7a97e17d62d - if this doesn't fix the issue, please reopen! Cheers

closed 2017-10-24 by tbisanz

agisen, 2017-10-24:
Indeed, this is the fix.
I changed it in the old eudaq version (1.6), and now I get the same results as with the new version.