eutelescope/eutelescope #379: Replace your GLB-align processors with the "common" EUTelAlignGBL

dreyling, 2018-05-07:
The goal is to merge different GLB implementations. 

Thus, we ask to test this processor for alignment 
https://github.com/eutelescope/eutelescope/blob/master/processors/src/EUTelAlignGBL.cc
and to post here the problems and outcomes.

If you determine a missing functionality or a missing parameter, please post it here as well.

For updating the documentation on this processor (e.g. possible parameters), please use this README.md file: processors/README.md

dreyling, 2018-05-16:
Question from meeting at DESY: How can you handle a 7th plane e.g. DUT for the alignement?

tbisanz, 2018-05-16:
First **important** step: update to the most current version.

tbisanz, 2018-05-17:
Please tell me once yu tested this :)

edorossi, 2019-01-14:
EUTelAlignGBL and EUTelFitterGBL now they are merged in the single processor EUTelGBL which is part of merge request #447

edorossi, 2019-01-25:
The new common GBL will come soon and there will be examples on how to use the GBL. We had quite a few discussion on how to improve it, which was the point of this issue.