eutelescope/eutelescope #383: Fatal errors while installing Eutelescope

clairedavid, 2018-05-15:
I just want to report the error messages I got while finishing a fresh install of eutelescope.
(just the install, will run examples later and report as well).

----- Setup ----- 
On DESY NAF (nafhh-atlas01)
Branch: dev-desynaf
Location: /nfs/dust/atlas/user/cldavid/eutelescope/
Versions: in file $ILCSOFT/ilcinstall/releases/v01-19/release-versions.py 
``
Eutelescope_version = "master"
Eudaq_version = "branches/v1.7-dev"
Millepede2_version = `'tags/V04-03-03'
``
Did at start:
``scl enable devtoolset-4 bash``

----- Errors -----
Four occurrences of "fatal error"
``$ILCSOFT/v01-19-02/Eutelescope/master/external/eudaq/v1.7-dev/nreader/include/EUTelBaseDetector.h:13:29: fatal error: eudaq/Platform.hh: No such file or directory
$ILCSOFT/v01-19-02/Eutelescope/master/external/eudaq/v1.7-dev/nreader/include/EUTelBaseDetector.h:13:29: fatal error: eudaq/Platform.hh: No such file or directory
$ILCSOFT/v01-19-02/Eutelescope/master/external/eudaq/v1.7-dev/nreader/include/EUTelBaseDetector.h:13:29: fatal error: eudaq/Platform.hh: No such file or directory
$ILCSOFT/v01-19-02/Eutelescope/master/external/eudaq/v1.7-dev/nreader/include/EUTelBaseDetector.h:13:29: fatal error: eudaq/Platform.hh: No such file or directory``

Log file (terminal dump of install, 6 MB) can be found here:
/nfs/dust/atlas/user/cldavid/eutelescope/ilcinstall/eutel_desynaf_install_201805151206.log

tbisanz, 2018-05-16:
This is an EUDAQ issue. They continued developing on the 1.7-branch. I modified the install scripts so that we install tag v1.8.0.

The most awesome thing you could do, is remove the $ILCSOFT directory and test the entire installation again? I know that's a bit annoying, but would verify that it runs on the NAF again!

clairedavid, 2018-05-16:
With a fresh login to the same machine, I did:
``scl enable devtoolset-4 bash
export ILCSOFT=/nfs/dust/atlas/user/cldavid/eutelescope_eudaq_v1-8
mkdir -p $ILCSOFT
cd $ILCSOFT
git clone -b dev-desynaf https://github.com/eutelescope/ilcinstall $ILCSOFT/ilcinstall``

The versions are now:
``Eutelescope_version = "master"
Eudaq_version = "tags/v1.8.0"
Millepede2_version = 'tags/V04-03-03'``

Installing (used nohup, but shouldn't be an issue)
``cd ilcinstall
nohup ./ilcsoft-install -i examples/eutelescope/release-standalone.cfg >& eutelescope_eudaq_v1-8_install_201805161430.log &``

As you guess the terminal content was dumped to:
/nfs/dust/atlas/user/cldavid/eutelescope_eudaq_v1-8/ilcinstall/eutelescope_eudaq_v1-8_install_201805161430.log

Seems like an ILCUTIL compilation problem (line 39055 from log file above)

Logfile for failed module: /nfs/dust/atlas/user/cldavid/eutelescope_eudaq_v1-8/v01-19-02/.ilcinstall-logs/release-standalone-2018-5-16_12-31.log

Shows some errors with
``/opt/rh/devtoolset-4/root/usr/libexec/gcc/x86_64-redhat-linux/5.3.1/ld: CMakeFiles/streamlog.dir/source/src/prefix.cc.o: relocation R_X86_64_32 against `_ZTVN9streamlog11prefix_baseE' can not be used when making a shared object; recompile with -fPIC``

Let me know what I can do to help debugging.

tbisanz, 2018-05-16:
Can you copy the logfiles to NAF? Also the old one.

clairedavid, 2018-05-16:
Log files are here:
/afs/desy.de/user/c/cldavid/public/Eutelescope/logfiles

Description:
eutel_desynaf_install_201805151206.log => first finished install with eudaq v1.7
eutelescope_eudaq_v1-8_install_201805161430.log => unfinished install with eudaq  v1.8 (after Tobias updated the release-version config file)
release-standalone-2018-5-16_12-31.log => more details on the failed install with eudaq v1.8

clairedavid, 2018-05-16:
Retrying with the -fPIC option

Note: I see it in the file 
$ILCSOFT/ilcinstall/examples/eutelescope/release-standalone.cfg
Before the fix: 
``101:ilcsoft.env["CXXFLAGS"]="-std=c++11"
105:   ilcsoft.env["CXXFLAGS"]="-m32 -std=c++11"``
After the fix: 
``101:ilcsoft.env["CXXFLAGS"]="-std=c++11 -fPIC"
105:   ilcsoft.env["CXXFLAGS"]="-m32 -std=c++11"``
The fix is not propagated to the if statement if the platform is 32 bit. Normal?

Installing using same commands as previous post. Error at compiling Eutelescope.

Logfiles:
/afs/desy.de/user/c/cldavid/public/Eutelescope/logfiles/eutelescope_eudaq_v1-8_install_201805162315.log

For convenience copy of error below:
``################################################################################
##### Compiling Eutelescope version master...
################################################################################

   ----- adding additional commands to build_env.sh :

+ Generated cmake build command:
  $  cmake -C /nfs/dust/atlas/user/cldavid/eutelescope_eudaq_v1-8/v01-19-02/ILCSoft.cmake -DCMAKE_BUILD_TYPE="Release" -DINSTALL_DOC="OFF" /nfs/dust/atlas/user/cldavid/eutelescope_eudaq_v1-8/v01-19-02/Eutelescope/master

Traceback (most recent call last):
  File "./ilcsoft-install", line 106, in <module>
    ilcsoft.makeinstall()
  File "/nfs/dust/atlas/user/cldavid/eutelescope_eudaq_v1-8/ilcinstall/ilcsoft/ilcsoft.py", line 479, in makeinstall
    mod.install([])
  File "/nfs/dust/atlas/user/cldavid/eutelescope_eudaq_v1-8/ilcinstall/ilcsoft/baseilc.py", line 874, in install
    self.compile()
  File "/nfs/dust/atlas/user/cldavid/eutelescope_eudaq_v1-8/ilcinstall/ilcsoft/eutelescope.py", line 82, in compile
    os.chdir( self.env[ "EUDAQ" ] + "/build" ) # needs to be defined in preCheckDeps (so it is written to build_env.sh)
OSError: [Errno 2] No such file or directory: '/nfs/dust/atlas/user/cldavid/eutelescope_eudaq_v1-8/v01-19-02/Eutelescope/master/external/eudaq/v1.8.0/build'``
in 
/nfs/dust/atlas/user/cldavid/eutelescope_eudaq_v1-8/v01-19-02/Eutelescope/master/external/eudaq/v1.8.0/

I only have files:
appveyor  appveyor.yml	AUTHORS  cmake	CMakeLists.txt	CONTRIBUTING.md  CTestConfig.cmake  doc  extern  extra	gui  legacy  LICENSE.md  main  monitors  README.md  README.release_notes  user

Sorry for the failed install again, hope the infos above help, happy to give more details if needed.

tbisanz, 2018-05-17:
Also added `-fPIC` to 32 bit version

jha92, 2018-05-17:
Hi, just checked on NAF.
`ls /opt/rh/` spits out `devtoolset-1.1  devtoolset-3  devtoolset-4  rh-java-common`
And sourcing as you described is also not possible.
Maybe asking NAF-admins for adding newer toolset?

tbisanz, 2018-05-17:
No, devtoolset-4 is fine - what you need are the parts of the softwarecollections: https://www.softwarecollections.org/en/

In particular git27: `yum -y install rh-git29` (I know that you can't install this)

tbisanz, 2018-05-17:
@clairedavid Can you just 
```
cd /nfs/dust/atlas/user/cldavid/eutelescope_eudaq_v1-8/v01-19-02/Eutelescope/master/external/eudaq/v1.8.0/
git checkout tags/1.8.0
cd build
cmake -BUILD_nreader=ON ..
make install -j2
```

clairedavid, 2018-05-17:
I did:
``scl enable devtoolset-4 bash``
``git checkout tags/v1.8.0``
For info I get:
``HEAD is now at 052d26b... Added LCIOsubevent to FEI4 Gen2 converter plugin (was made aware by Andreas thta it was still missing...strange...I thought it was already used).``

``cd build``
``cmake -DBUILD_nreader=ON ..``
And get the following error:
``CMake Error at CMakeLists.txt:2 (CMAKE_MINIMUM_REQUIRED):
  CMake 3.1 or higher is required.  You are running version 2.8.12.2

-- Configuring incomplete, errors occurred!``

I can change CMake version from lsetup but it sets some environment variables apparently conflicting with Eutelescope. So without lsetup nor root-permission, I can't upgrade CMake here.
That's why I think Jan-Hendrik has a good suggestion: we could ask for a newer toolset.

Hope it helps!

clairedavid, 2018-05-17:
Seems I forgot sourcing the build_env.sh, my bad. Sorry for that.

Seems to work. Couples of warnings. Will report on this thread if I bump into errors once I do the BTTB6 tutorial.
[install_eutelescope_dev-desynaf_eudaq_v1-8-0_20180517.log](https://github.com/eutelescope/eutelescope/files/2014073/install_eutelescope_dev-desynaf_eudaq_v1-8-0_20180517.log)

tbisanz, 2018-05-22:
Completed!

closed 2018-05-22 by tbisanz