eutelescope/eutelescope #386: Example anemone-2FEI4 - Unknown EUTelApplyAlignmentProcessor

ggiannin, 2018-05-17:
I am running the anemone example and I am stuck at the alignment. It seems like the processor EUTelApplyAlignmentProcessor is missing from the eutelescope version I downloaded a few weeks ago. 

16:50:54 jobsub.align(INFO): Now running Marlin on align-000017.xml
16:50:55 jobsub.align(INFO): 
16:50:55 jobsub.align(INFO): 
16:50:55 jobsub.align(INFO): 
16:50:55 jobsub.align(INFO): 
16:50:55 jobsub.align(INFO): 
16:50:55 jobsub.align(ERROR): ***********************************************
16:50:55 jobsub.align(INFO): XMLParser::parse : no <constants/> section found in align-000017.xml
16:50:55 jobsub.align(ERROR): A runtime error occured - (uncaught exception):
16:50:55 jobsub.align(ERROR): lcio::Exception:  ProcessorMgr::registerProcessor: unknown processor with type EUTelApplyAlignmentProcessor !
16:50:55 jobsub.align(ERROR): 
16:50:55 jobsub.align(ERROR): Marlin will have to be terminated, sorry.
16:50:55 jobsub.align(ERROR): ***********************************************
16:50:55 jobsub.align(ERROR): 
16:50:55 jobsub(ERROR): Marlin returned with error code 1

Thank you.

tbisanz, 2018-05-17:
Can you try `EUTelProcessorApplyAlignment` instead? I.e. change this line: https://github.com/eutelescope/eutelescope/blob/master/jobsub/examples/anemone-2FEI4/steering-templates/align-tmp.xml#L59 to:

`<processor name="ApplyPreAlignment" type="EUTelProcessorApplyAlignment">`

tbisanz, 2018-05-17:
For a more recent example - look at the `jobsub/examples/aconite-4chipLocal` one. If you want to understand the modern way of using EUTelscope!

ggiannin, 2018-05-17:
> Thank you Tobias,
>
> Now it’s giving me other errors, I guess the type of AlignMode were changed from ‘int’ to a string or something.
> (https://github.com/eutelescope/eutelescope/blob/master/jobsub/examples/anemone-2FEI4/steering-templates/align-tmp.xml#L155 <https://github.com/eutelescope/eutelescope/blob/master/jobsub/examples/anemone-2FEI4/steering-templates/align-tmp.xml#L155>)
> I’ll try with the newer example!
>
> Thanks again.
>
>
> =========
> Giulia Giannini
> PhD Student - ATLAS pixels group
> Institut de Física d’Altes Energies (IFAE)
> Edifici CM7 UAB Campus,
> Carrer de Can Magrans
> 08193 Bellaterra (Barcelona), Spain
> =========

I tried with aconite-4chipLocal but I am getting the same error as I mentioned before about AlignMode, while doing the DafFitter.

18:02:56 jobsub.align(ERROR): [ ERROR "PedeToGEAR"] The chosen AlignMode: '3' is invalid. Please correct your steering template and retry!
18:02:56 jobsub(ERROR): Marlin returned with error code -6

=========
Giulia Giannini
PhD Student - ATLAS pixels group
Institut de Física d’Altes Energies (IFAE)
Edifici CM7 UAB Campus, 
Carrer de Can Magrans
08193 Bellaterra (Barcelona), Spain
=========

tbisanz, 2018-05-17:
This is really embarrassing - yes, the align mode is now a string: instead of 3, please put "XYShiftsAllRot". There are two places where this needs to be changed. I'll change that in the repo tomorrow!

ggiannin, 2018-05-18:
Hi Tobias, 

No worries and thanks a lot. 

While running on the march test beam data I have noticed one thing on the fitter steering file and on the example you pointed meta it seems to be the same.
https://github.com/eutelescope/eutelescope/blob/master/jobsub/examples/aconite-4chipLocal/steering-templates/fitter-tmp.xml#L128 <https://github.com/eutelescope/eutelescope/blob/master/jobsub/examples/aconite-4chipLocal/steering-templates/fitter-tmp.xml#L128>
<parameter name="Undo Alignment (boolean)" type="bool" value="true”/>

What is this exactly doing? I tried to put false but i only get an error.
Also I was wondering how the fitter does to take into account the alignment from the previous step, 
The slcio file in the align-tmp.xml 
https://github.com/eutelescope/eutelescope/blob/master/jobsub/examples/aconite-4chipLocal/steering-templates/align-tmp.xml#L153 <https://github.com/eutelescope/eutelescope/blob/master/jobsub/examples/aconite-4chipLocal/steering-templates/align-tmp.xml#L153>
is not created and there is apparently nowhere in the fitter where the results from align step is considered. 
Sorry I am just trying to understand since the residuals from the fitter are quite off, like the align step was ignored. 

Thank you.

Cheers, 
Giulia

=========
Giulia Giannini
PhD Student - ATLAS pixels group
Institut de Física d’Altes Energies (IFAE)
Edifici CM7 UAB Campus, 
Carrer de Can Magrans
08193 Bellaterra (Barcelona), Spain
=========

tbisanz, 2018-05-18:
With #387 the AlignMode is set correct, also the anemone example should work again :)

`<parameter name="Undo Alignment (boolean)" type="bool" value="true”/>`  will undo the alignment, i.e. it will transform a global hit into the local sensor frame.

`EUTelProcessorCoordinateTransformHits` can transform hits from local to global - when the `Undo Alignment` is set to `false` or transform from global to local. A hit is flagged as local/global, if you try to transform a local hit into a local you will get an error (as you did).

The alignment it not done via alignment-databases anymore, this you don't find a alignment file, but instead the gear file - containing all the alignment corrections - is incremented. The transformation is done with this processor execution: https://github.com/eutelescope/eutelescope/blob/master/jobsub/examples/aconite-4chipLocal/steering-templates/align-tmp.xml#L20

closed 2018-05-22 by tbisanz

ggiannin, 2018-05-24:
Hi Tobias, 

Thank you so much. I tried with both anemone and aconite examples, the ones you pointed me to, and it worked in both cases. 
Now I am moved back to the march-tb data and I am getting good residuals after the fitter. But this sounded strange to me
since during the align step I was getting an error from millepede: 

18:02:54 jobsub.align(ERROR): [ ERROR5 "PedeToGEAR"] STOP ... no variable global parameters
18:02:54 jobsub.align(ERROR): [ ERROR5 "PedeToGEAR"] Problem during Pede execution, exit status: 0, error messages (repeated here):
18:02:54 jobsub.align(ERROR): [ ERROR5 "PedeToGEAR"] STOP ... no variable global parameters
18:02:54 jobsub.align(ERROR): [ ERROR5 "PedeToGEAR"]
18:02:54 jobsub.align(ERROR): [ ERROR5 "PedeToGEAR"] Will exit now

Is it possible that it I get good residuals after the fitter just by chance? 
Thanks.

Cheers, 
Giulia

=========
Giulia Giannini
PhD Student - ATLAS pixels group
Institut de Física d’Altes Energies (IFAE)
Edifici CM7 UAB Campus, 
Carrer de Can Magrans
08193 Bellaterra (Barcelona), Spain
=========