eutelescope/eutelescope #390: Hardcoded calculations in EUTelAlignGBL for iterations after the first

edorossi, 2018-05-22:
I have some concerns regarding the flag IsFirstAlignStep and the operation it implies:

https://github.com/eutelescope/eutelescope/blob/master/processors/src/EUTelAlignGBL.cc#L250
https://github.com/eutelescope/eutelescope/blob/master/processors/src/EUTelAlignGBL.cc#L286
https://github.com/eutelescope/eutelescope/blob/master/processors/src/EUTelAlignGBL.cc#L351

Basically, if that quantity is set to 1, the resolution that is used is not the one set in the config/steering, but it is just an hardcoded guess. In the same way, if it is set to 0, the triplet that are used are actually the value set in the config/steering, multiplied by six and divided by the energy of the beam.

To recap, however the flag IsFirstAlignStep is used, some of the parameters that the user have set will be overridden in the source code.

Can we just delete the flag and its calculations or there is some good reason to keep them?

tbisanz, 2018-05-22:
@TCThenna might want to comment on that. But yes, I think it might be a good idea to get rid of that flag, and instead use a well documented example!

tbisanz, 2018-05-22:
The heuristic values which were taken from DESY experience are listed here: https://github.com/eutelescope/eutelescope/wiki/Alignment-Residuals-at-DESY I'd propose we remove them from the processor and provide examples with a reference to this table

edorossi, 2018-05-22:
Hi @tbisanz, I will delete everything related to that, and place a comment addressing that page. The main problem is that these parameters are fine for DESY, but not for SPS (the triplet cuts in the first iteration would be smaller and not larger in that case)!

tbisanz, 2018-05-22:
Please commit and issue a PR for these changes :)

tbisanz, 2018-05-22:
Do we have SPS data we could provide as an example?

edorossi, 2018-05-22:
I have something from ITk strips (+FEI4) of a couple of years ago. My plan would be to create an example for the strips with a DESY and a SPS run, since it is nice to see how the cuts should be set differently depending on the beam energy. Now the alignment it's crashing after the first event is processed, though. After everything is fixed I will add the example.

edorossi, 2018-05-22:
Hi @TCThenna, I see the point of having an dirty formula, but I think it can get more confusing than helpful. I really think that the best solution would be to have a couple of working and well documented example where it is explained what these cuts are and how to set them (we can set them up quickly once we have the code running). In the end, everyone takes the best example to set it up its own reconstruction. 

I have always set the values for the cuts by looking at the output plots. That's not the fastest option, but it is simple and it helps understanding the pattern recognition process. In the end, the concept of the triplets is very easy to undestand.

I also think that how the resolution of the first alignment step is heavily dependent on the triplet cuts that are used, on the divergence of the beam and on budget of the DUT. I used to use 10um in the first alignment step (combined with tight triplet cuts), while with the formula that is suggested I would use 60um. If I used 60um as suggested, in the next alignment steps I would still need to use a resolution larger than the nominal 3.24um, since some more alignment work is needed. Basically, there would be the situation in which neither the hardcoded value or the real resolution is right.
Moreover, with the current hardcoded calculation, the resolution is set for all the planes, DUT included. It can be fixed easily, but I can think only about gross solutions :)

Regarding your points:

a) you mean to unify triCut and driCut, or to include also sixCut?

b) I do not understand why the energy scaling should be used. There is a sort of optimum for the cuts, and if they are decreased below that value, a lot of good tracks are removed. If you insert cuts that are scaled by the energy, you also must pay attention to use a starting cut that, when scaled, it doesn't become lower than this optimum value. Basically, you must already know the cut to be used in the final iteration, in order to set the starting cuts. I don't know if it is clear what I mean, but I think that it is not user-friendly, and actually one must be quite expert to use it without messing it up.

Ok, to recap, my main point is that I think that hardcoded values work until they don't work. When they stop working, it becomes a real pain for the user to debug it, and there are too many setups and beams to have accurate guesses.

TCThenna, 2018-05-22:
> Regarding your points:
> 

> a) you mean to unify triCut and driCut, or to include also sixCut?
I think we should keep the driCut option, but we could default the driCut to triCut, if driCut is not set.

> b) I do not understand why the energy scaling should be used. There is a sort of optimum for the cuts, and if they are decreased below that value, a lot of good tracks are removed. If you insert [...]

The thought behind this is, that if one would like to compare runs from various energies, one would like to have an appropriate set of cuts, that produce a comparable subset of tracks. 

Now, with less energy, you have more MS, so one needs to relax the triCut.
> 
> Ok, to recap, my main point is that I think that hardcoded values work until they don't work. When they stop working, it becomes a real pain for the user to debug it, and there are too many setups and beams to have accurate guesses.

So you opt for not defaulting the Cuts and throwing error when not specified? I'd be fine with that, so people are forced to read the manual :D 
Still, please let me know what you think about have this structure:
- the user specifies triCut, driCut and six-cut
- the user specifies with a boolean, if cuts should be scaled with energy.

edorossi, 2018-05-22:
Hi @TCThenna, I was thinking to have some default for the cuts (but just when the parameter is read, without any further calculation) and have always an output on the terminal pointing to the manual. When we have everything running, we should also have an example with the cut explicitly set, and a comment in capital letter pointing to the documentation :) 

I see what you mean regarding a single triplet cut. I totally agree with the possibility of setting driCut=triCut if there are not further settings.

On the other hand, I still not like the scaling with energy. I have some inner voice that is telling me that if you scale by energy, to set the initial cuts you need to know the final cuts :)
In any case, the scaling as it is can not left in place, since it would not work properly with both the SPS (~120GeV) and PS (~7GeV) beams. Can we delete it for now and postpone the discussion until we have some sort of validation for the CERN beams?

TCThenna, 2018-05-22:
> I was thinking to have some default for the cuts
This is again a magic number, that also does not work for everyone. Can we somehow do better?

Of course we can add this functionality later again, but I don't agree that one has to know more in advance when one scales a cut value, in comparison to not scaling.

edorossi, 2018-05-22:
> This is again a magic number, that also does not work for everyone. Can we somehow do better?

I agree. One possibility is to throw an error if the cuts are not explicitly defined, as you suggested (I didn't get it in your previous message, I am sorry). Another possibility would be to have some very large standard cuts (let's say 1cm), so Millepede fails, but still the plots of the cuts are filled. At this point an output on the terminal says something like: "Too many rejects, please check the cuts and the plots".

TCThenna, 2018-05-23:
This would certainly be possible, but would it be 'easy-to-use' for newcomers?

What about using some (better than now) heuristic formula if no cuts are specified, and not using it, when at least one parameter is specified? So it would give OKish results for the inexperienced users, and full flexibility for pros? 

I guess it would be nice if the example processors do not only work on the exact example given, but also if one applies them to one's own data, that are different in plane spacing and energy used. (But maybe that is a bad idea to begin with?)

tbisanz, 2018-05-23:
I think this is somehow developing into a fundamental discussion. Just to add my thoughts:

I agree that a set of working values is very helpful. However, I'm against the case that these values only work for a DESY set-up and not a SPS one. So, we could have some energy cut which suddenly switches them to SPS default values. The problem I have with this is that people might look into the code and see a lot of arbitrary magic numbers.

I think that one of the major problems with EUTelescope is that we have way too many parameters which interact weirdly with each other. The case "if flag X is set, parameters Y are used, Z instead if parameter P is larger than 1.0, or Q if all else fails" is what is really confusing to the user, and which obscures code tremendously.

Having a user friendly framework doesn't mean that a user can take EUTelescope, plug in data and press 'go' and everything works. It means that we can point them to an updated website, where they can easily understand what the parameter does and how to set it. In particular, with some heuristic starting values!

It's a bit like Lego, the Lego pieces are the processors - they are kept simple, it's obvious how you can stick them together. Sometimes you can stick pieces together and it's a bit useless - but you quickly kind of get the idea. However, if you really want to build something nice, you'll take a look into the instructions. It's out job not to build processors (Lego blocks) which have weird shapes and nobody understands how they are supposed to be used ("Why the fuck is my Lego brick a donut?"), and their job to step through some (not too many!) instructions.

TCThenna, 2018-05-23:
> I agree that a set of working values is very helpful. However, I'm against the case that these values 
only work for a DESY set-up and not a SPS one.

I never said it should work only for DESY :D  But this already implies, that there should be some if/elseif depending on energy and plane spacing to be a working example. At least if it should work for more than exact one setting. 

Fact is, that each, also the first, round of alignment needs uncertainties of hits, and one does not  know these a priori. That is how it is. So in this case, there always needs to be an educated guess about it. 

We can let every user scratch their heads to find good values, or we help them. If this is in the manual, be it. I am fine with it. 

But you haven't proposed smth that actually gets rid of magic numbers, or I just don't understand it :) You only proposed to make the heuristic formula a single magic value and hence work in less cases. If this is what we want, ok, I have no hard feelings about removing it. I only wanted to discuss, if you think that this really improves the processor. If you are positive, then get rid of that chunk of code.

tbisanz, 2018-05-23:
No, I didn't say that it would work only for DESY - I didn't propose any solution at all, nor did I only want to put one magic number.

I think we should opt for as little processor parameters as possible - so I'd suggest to provide only the resolution to the processor. Obviously, this means that the heuristic values go into the manual - nicely explaining why you choose which value at what energy. But I'm also open to other suggestions!

tbisanz, 2018-06-05:
As we removed them, we should redirect here: [link to wiki](https://github.com/eutelescope/eutelescope/wiki/Alignment-Residuals-at-DESY) and close this

closed 2018-06-05 by tbisanz