eutelescope/eutelescope #391: Alignment modes in EUTelAlignGBL

edorossi, 2018-05-22:
Right now, the most generic alignment mode involves shifts in the 3 dimensions, and rotation around Z (XY plane). If I am not mistaken, in this way we can't align geometries where there is a rotation around X and Y, that is relatively common for the DUT.

Another problem with this approach is that the alignment method is set for all the planes together. However, we have test beams with 5cm strips where it is often advantageous to fix the Y shift for the DUT, since we lack that information from the sensor.

Wouldn't be advantageous to not set the alignment method as a string, but to set which planes and for which dimensions are fixed?
Basically set 6 variables on the line of (random numbers from a config file):

`FixedXshift       =  0    5`
`FixedYrot  =  0  1  5`

I think it would be quite easy to create the Millepede steering file from something like this and perhaps more elegant than how it looks now.

On the other hand, I don't know much about Millepede and I don't know the difference (if any) between setting -1.0 and 0.0 in the third column of the steering file. I am understanding that -1.0 means that that shift/rotation is fixed, but I don't know what 0.0 is. If there is no difference, an option would be to set to -1.0 if that dimension is fixed for that plane, and to 1.0 if not.

thomaseichhorn, 2018-05-22:
These features are available in EUTelMilleGBL. You could either use that processor directly or check the code and copy the relevant parts :)

tbisanz, 2018-05-22:
I'm happy with any method, but if you implement these changes, make sure that it defaults to something useful and get rid of the `AlignMode` - i.e. don't use both possibilities. The goal should be as little parameters as possible. Maybe you could write a custom processor which writes the pede steering file and decouples it from the processor which creates GBL tracks and passes them to mille, i.e.:

GBLMille (this) -> WritePedeFile (new) -> RunPEDEandWriteGEAR (existing but needs to be adapted)

tbisanz, 2018-05-22:
In principle, since the RunPEDEandWriteGEAR part also needs to know about the configuration, the PEDE steering file writing could also be move in there!

edorossi, 2018-05-22:
Hi @thomaseichhorn, thank you very much. We have something like that in our branch as well, I will move it to EUTelAlignGBL soon (unless someone speaks in favor of the current method).

@tbisanz, exactly, my idea is to delete the current string for the alignment mode and replace with the vectors of planes. I have never checked the RunPEDEandWriteGear function, I will take a look at it and try to move the pede writing there!

I am sorry for repeating myself, but do you know the difference between setting -1.0 and 0.0 in the third column of the Millepede steering file?
I have the feeling that -1.0 means that the direction is fixed, while 0.0 means that it is not considered at all (i.e. not used for chi2 calculation). The difference would be small but significant.

tbisanz, 2018-05-22:
> Pre-sigma. The pre-sigma $s_{\ell}$ defines the status of the global parameter:

>$\boldsymbol{s_{\ell} > 0}$: The parameter is variable with the given initial value. A term $1/s_{\ell}^2$ is added to the diagonal matrix element of the global parameter to stabilize a perhaps poorly defined parameter. This addition should not bias the fitted parameter value, if a sufficient number of iterations is performed; it may however bias the calculated error of the parameter (this is available only for the matrix inversion method).
>$\boldsymbol{s_{\ell} = 0}$: The pre-sigma is zero; the parameter is variable with the given initial value.
>$\boldsymbol{s_{\ell} < 0}$: The pre-sigma is negative. The parameter is defined as fixed; the initial value of the parameter is used in the fit

tbisanz, 2018-05-22:
I'll gladly help you, just keep me up to date with the changes you want to incorporate.

edorossi, 2018-05-22:
Hi @tbisanz, actually, before doing this I would like to fix a problem I have when it finds the triplets (so I can actually test that the code is working). I have some ideas on what that problem is and I will let you know as soon as I am sure!

edorossi, 2018-05-23:
@thomaseichhorn, I went through your code and I got a bit lost, mainly because I don't have much experience with Millepede. In particular, I don't fully understand the constraints and the interaction between fixed planes, local derivatives and such constraints. How crucial is it to have those constraints? And how often do you use alignment modes that don't involve all six the derivatives? 
These questions actually can be translated to: how gross would it be to just use all six derivatives, no constraints, and simply fix the shifts/rotations for the planes that we don't want to consider? I am asking since that would be quite easy to implement and quite flexible :)

Apart from this, the only idea I would have  set the labels as plane-ID * 10 + 1...6, as Claus did in eutelpy. This would simplify all the cycles used for writing the steering file, I guess. Is it a valid approach or am I missing something?

I am sorry for the long list of questions, but I have always been quite puzzled by Millepede, and before touching anything I would like to have at least a clue about what I am doing :)

tbisanz, 2018-05-25:
@edorossi Any update?

edorossi, 2018-05-25:
@tbisanz, I refactored a bit EUTelAlignGBL, because I had quite a few problems with the plane ordering and with the excluded planes. I am also writing a proper steering file. However, I am having some problems when reading the millepede results (again, because of problems with the exclusion of planes, not because of Millepede or GBL). As soon as I have running until the end, I will open a pull request to discuss about it, even if it is far from being in the best shape.

edorossi, 2019-01-14:
Added to merge request #447

closed 2019-01-14 by edorossi