eutelescope/eutelescope #395: Crashes in GeometricClustering when no collection is present

tbisanz, 2018-05-23:
Confer #392

tbisanz, 2018-05-23:
To investigate the cause of the crash:

Make a `Marlin` steering file for the step: `jobsub -c config.cfg -csv runlist.csv clustering <run no> --dry-run`

Then start it with `gdb` using following command `gdb --args Marlin clustering-<run no>.xml`

In the `gdb` terminal:

```
catch throw
commands
(a > terminal will appear)
bt
c
end
(the > terminal will be escaped)
set pagination off
r
```

Once it ended `bt` and copy&paste the output.

tbisanz, 2018-05-23:
Just to update why this should not fail:

The typical case is that one will have multiple collections, e.g. from DUT and telescope. Both collections wills be clustered with different processors or multiple calls of the same one. If only one of them has no data, this will skip the entire event, even though the other collection (e.g. the telescope) had perfectly valid data.

edorossi, 2018-05-23:
Here there are the last lines of the output:

[GeometricClusteringOutput.txt](https://github.com/eutelescope/eutelescope/files/2030377/GeometricClusteringOutput.txt)

Just for clarity, the FEI4 has plane number 7, and there are some fake DUT planes (but it doesn't matter). Everything crashes at the event 5497, which doesn't have a collection zsdata_apix at all.
To me it looks like it is complaining about the line:

https://github.com/eutelescope/eutelescope/blob/master/processors/src/EUTelProcessorGeometricClustering.cc#L173

But to tell the truth, I have no clue on what is going on :)

tbisanz, 2018-05-23:
Did you type `bt` at the end, before pasting the output? I don't see that :S

edorossi, 2018-05-23:
Ah, of course I forgot it :) I will rerun in a few minutes.

> I think the dumpevent and printed event numbers might be one off, so please check that and paste the output.

The missing collection is in event 5497 (that is counted as 5496 in the terminal). I am sorry if I don't put the dumpevent, but it is a mess

> This should be reported as a bug to your EUDAQ converter plugin. However, for this case, I'll write a fix.

You found the trick :) The data has been converted to LCIO externally by EUDAQ2 and the input to the clustering is the output of EUDAQ2 "unfiltered". I would have fixed it myself when I did the conversion if I had this problem before, but it didn't happen somehow (even though I was using the GeometricClustering).

As a side note, if in the steering file I execute the clustering of the FEI4 before the one of the mimosas, everything crashes after the first event (the mimosa is missing event 0 somehow). while if I put the execution of the mimosa clustering first, everything crashes at the event 5497, where there is the missing FEI4 event.

edorossi, 2018-05-23:
Here there is the proper output!

[GeometricClusteringOutput2.txt](https://github.com/eutelescope/eutelescope/files/2030551/GeometricClusteringOutput2.txt)

tbisanz, 2018-05-23:
Following line is the problem: https://github.com/eutelescope/eutelescope/blob/master/processors/src/EUTelProcessorGeometricClustering.cc#L264 - given that we rely on the existance of a collection, we could either actually skip the event or throw an error.

tbisanz, 2018-05-23:
I'll modify the code, but this is actually not a bug, but expected undefined behaviour. EUDAQ output has to provide collections for all events! Please ensure that this happens in the future!

edorossi, 2018-05-23:
I totally agree with you! From my side, it would just be important that the two codes are consistent with each other: if I noticed this problem when I first run the data (immediately after the conversion), I would have fixed it immediately in EUDAQ!

closed 2018-05-24 by tbisanz