eutelescope/eutelescope #396: GBL Naming

tbisanz, 2018-05-23:
Dear all,

to decouple this a bit from the other threads. Naming. I know this somewhat developed historically, but we have driplets, triplets (I'm Austrian for god's sake, these are the same words!) Furthermorem, triCut, driCut (same again!) and six-cut and many more very unintuitive terms.

If we stick to these terms, we have to explain them somewhere. Maybe it would also be a good time to think about renaming some of them? I don't want to force them being renamed, I'd just like to hear your thoughts and at least get some documentation on them!

TCThenna, 2018-05-23:
triCut -> tripletResidualCut
driCut -> dripletResidualCut
sixCut -> triplet-driplet-matchingCut

tbisanz, 2018-05-23:
Is that a suggestion for renaming? If so, I'd go for _tripletDripletMatchingCut_

edorossi, 2018-05-24:
What about something like upstreamTripletCut, downstreamTripletCut and tripletsMatchingCut? Driplet looks like a typo :)

TCThenna, 2018-05-24:
upstreamTripletCut does not explain, upon which quantity the cut acts. The cut is on the residual of a straight line (from outer points) extrapolated to the middle plane, and their, the distance b/w the line and the hit.

maybe:
upstreamTripletResidualCut
downstreamTripletResidualCut
tripletsMatchingCut is fine for me

tbisanz, 2018-05-25:
Let's make a guideline: parameters should be of **theFollowingForm** - for any future processor.

For the naming, I'm perfectly fine with the last suggestions, personally I prefer "upstream" and "downstream" to "tri" and "driplet" as it makes talkign about them much easier. But I don't have any hard feelings towards any suggestions.

TCThenna, 2018-05-25:
I'm fine with the guideline.

and if Edo is fine with my suggestion, than we can close this issue :D

jha92, 2018-05-25:
I like the suggestion.
Only to be explicit: we should fix the parameter name to thFollowingForm - for any processor, we "renovate" and would like to maintain in our basic package, so also back-propagating to existing once.

edorossi, 2018-05-25:
I totally agree with @TCThenna suggestion!

TCThenna, 2018-05-25:
upstreamTripletSlopeCut (that would pair well to upstreamTripletResidualCut, same style)

tbisanz, 2018-05-25:
The cut is applied to the up- & downstream triplets: https://github.com/eutelescope/eutelescope/blob/master/processors/src/EUTelAlignGBL.cc#L410-L411, but modified with a hard-coded value. What shall we do there?

TCThenna, 2018-05-25:
For scattering studies, we need two cuts:
- the triplet (sry, upstreamTriplet) has a small divergence, 
- the downstreamTriplet can have large scattering angles.

So I'd opt for two separate ones. Maybe the `downstreamTripletSlopeCut` is defaulted to the upstream one, and overwritten when specified.

tbisanz, 2018-06-05:
Decided!

closed 2018-06-05 by tbisanz