eutelescope/eutelescope #406: Need new job submission scripts for DESY NAF (migration to HTCondor)

TCThenna, 2018-05-29:
@schuetzepaul @thomaseichhorn @edorossi @dreyling (what are the user names of michaela, Jan-Hendrik ...)

Has anyone already made new job submission scripts for HTCondor?

dreyling, 2018-05-31:
Talked quickly to my office colleague, who used it for a different software.

- `qsub` is now `condor_qsub`
- major changes are using absolute paths instead of relative path

References: https://confluence.desy.de/pages/viewpage.action?pageId=67639562

TCThenna, 2018-06-27:
> please migrate to HTCondor BIRD, as SGE bird will be shut down on 1.7.2018
Documentation: http://bird.desy.de/htc/
Contact: bird.service@desy.de

this becomes kind of urgent. 

Has anyone tried so far?

jha92, 2018-06-27:
Yes, Jan and I did this and will soon push our version. Need a bit cleanup, but otherwise it should be ready.
So far we implemented it as a new option (--htc). If SGE is gone soon, should we keep the current submission (--naf) or replace it with the new?

TCThenna, 2018-06-27:
I'd say replace it. The code is still on GitHub if we would ever need it again.

dreyling, 2018-06-27:
Yes, before we replace/optimize it, it would be nice if you @TCThenna and @schuetzepaul (as discussed yesterday at eating cake) could test this to collect some experience which improvements would be useful. Please find here the working example by @jha92 
https://github.com/beam-telescopes/eutelescope/tree/example_GBLnoDUT/jobsub/examples/htc-submission

jha92, 2018-07-02:
Perfect. To make a pull request (hopefully this afternoon), I created a new branch with only the submission part as new feature (see https://github.com/beam-telescopes/eutelescope/tree/submitCondor).
So far done:
 - new flag ```--condor``` for HTCondor submission
 - logs are stored in output/logs
 - steering files are stored in output -> how about a new subfolder there?
 - zipLogs function disabled (not usable with submission) -> do we want to preserve it?
 - removed ```--naf``` option
 - added new example ```condor_submission``` (removed ```naf_submission```)
 - let LXPlus submission untouched, but I think we should remove ```bsub``` to, as the general movement goes to condor; still need to validate running on LXplus with condor (installation ongoing)

TCThenna, 2018-07-02:
>  steering files are stored in output -> how about a new subfolder there?

maybe in output/runnumber, we can store mille files, steering files, job submission files, ... that belong to a certain run?

jha92, 2018-07-02:
Yes, possible. Then jobsub.py has to make sure that the sub-folder exists, otherwise make a new one.
Do you have opinions on this, @tbisanz?

dreyling, 2018-07-02:
I think @tbisanz works locally, but I think to have subfolders cold be nice.

TCThenna, 2018-07-03:
>  Yes, I know, but I think it is nicer to have one structure, nonetheless if the job was executed locally or via HTCondor.

Agreed.

> If creating a subfolders like output/runnumber one can ask why not sorting all types in there then (so no logs/histograms/database/lcio folders).

I see your point and discussed with Paul.
In contrast to my earlier comment, we actually think we can drop the runnumber folders and add `output/steering` and `output/mille` to the structure (and maybe one two more if needed. All should be created automatically if not present yet, so you don't have to set up the structure by hand.

jha92, 2018-07-03:
> In contrast to my earlier comment, we actually think we can drop the runnumber folders and add output/steering and output/mille to the structure (and maybe one two more if needed. All should be created automatically if not present yet, so you don't have to set up the structure by hand.

Yes, that sounds good. I would implement the automatic creation of the full structure (so ```output``` plus all its subfolder) then into ```jobsub.py```, so the user can't miss it.
Or was there some inherent reason, why one has to create it manually so far?

TCThenna, 2018-07-03:
> Yes, that sounds good. I would implement the automatic creation of the full structure (so output plus all its subfolder) then into jobsub.py, so the user can't miss it.

Great.

> Or was there some inherent reason, why one has to create it manually so far?
None that I know of.

jha92, 2018-07-03:
Implementation now done. So far I didn't create a folder ```output/mille``` as in general I think to store the results in ```output/database``` is good. Only question will be to store the extra outputs from mille there too.
I opened now a pull request #437.

jha92, 2019-01-25:
Merged. Updated version will come with planned release.

closed 2019-01-25 by jha92