eutelescope/eutelescope #409: EUDAQ2 connection

dreyling, 2018-05-29:
The workflow for data from EUDAQ2 is at the moment:
- data taking
- converting to slcio with euCliConverter (EUDAQ2 program)
- use these files directly for EUTelescope by skipping the Conversion step

If we don't need the conversion step in EUTelescope, we don't need the EUDAQ dependency. 
How complicated is to have a EUTelescope builtding without EUDAQ? Is it only adjusting the (c)makefiles by for example a flag or something similar?

tbisanz, 2018-05-29:
We can easily provide this. However, building EUDAQ is typically our least concern and all the test data needs EUDAQ 1 conversion. Quickly running through an example is the best way to check that all the external packages work fine - so I'd advise against it!

edorossi, 2019-01-14:
Closing this issue since in the meanwhile several EUDAQ 2 runs (taken using the standard mode) have been successfully reconstructed and analyzed.

closed 2019-01-14 by edorossi