eutelescope/eutelescope #412: Is the docker expected to work?

bnachman, 2018-06-02:
Hello,
I'm very happy to see the dockerfiles in the master: https://github.com/eutelescope/eutelescope/tree/master/docker.  Are they expected to work?  I've tried running 

docker build .

and it doesn't work for any of the dockerfiles, failing on the ROOT install with a cryptic message.  I'm running on Max OSX but I suppose it should not matter (?)  Thank you!

Sincerely,
Ben

tbisanz, 2018-06-03:
Hey Ben,
they should work - I built them just last week. I even made files (and built the images) for Ubuntu18 and Fedora 28 from Friday to Saturday and it built.

The entry points for the images are not really fit for running typical reconstruction jobs, but currently run a build on a Pull Request in Travis and run the test cases. It would be great to see someone use them in a different way though :) so I'm very interested in fixing them for you. I built them on Fedora26 (or 27, I'm on mobile right now and can't check)

tbisanz, 2018-06-04:
Added a openSUSE (tumbleweed, their pure rolling release version) docker file and built the image over night!

closed 2018-06-07 by tbisanz