eutelescope/eutelescope #416: NAF installation

dreyling, 2018-06-05:
Ongoing from #383 

The standard procedure 
```
export ILCSOFT=/nfs/dust/USER-PATH/NAME
...
$ git clone https://github.com/eutelescope/ilcinstall
...
$ git checkout -b dev-desynaf origin/dev-desynaf
$ scl enable devtoolset-4 bash
$ ./ilcsoft-install -i examples/eutelescope/release-standalone.cfg
```
still fails, since EUDAQ is checked out in the master branch:
```
...
OSError: [Errno 2] No such file or directory: '/nfs/dust/atlas/user/dreyling/v01-19-02/Eutelescope/master/external/eudaq/v1.8.0/build'
$ cd /nfs/dust/atlas/user/dreyling/v01-19-02/Eutelescope/master/external/eudaq/v1.8.0/
$ git status
# On branch master
nothing to commit (working directory clean)
```

There is a git version 1.9 on the naf:
```
$ /opt/git/1.9/bin/git --version
git version 1.9.0
```

What is the easiest way in the installer to use this version?

dreyling, 2018-06-05:
How can I quickly redo the installation after changing something or do I have to restart everything?

tbisanz, 2018-06-05:
Restart again, your git is too old.

dreyling, 2018-06-05:
Ok and 

> There is a git version 1.9 on the naf:
> ```
> $ /opt/git/1.9/bin/git --version
> git version 1.9.0
> ```
> What is the easiest way in the installer to use this version?

dreyling, 2018-06-05:
Adding simply the path is not a solution.. I am trying now by changing line 
https://github.com/eutelescope/iLCInstall/blob/71121a61e493d1eb9e6272112de653eb7ece3df8/ilcsoft/eutelescope.py#L79
into and doing the 
```
if( os.system( "git clone https://github.com/eudaq/eudaq eudaq/%s && cd eudaq/%s && git checkout %s" % (os.path.basename(self.env["EUDAQ_VERSION"]), os.path.basename(self.env["EUDAQ_VERSION"]), os.path.basename(self.env["EUDAQ_VERSION"])) + " 2>&1 | tee -a " + self.logfile ) != 0 ):
...
```

tbisanz, 2018-06-05:
The problem is that this is not only used for eudaq, but many ilcsoft packages. They compile right now but you might have the master branches there as well! So what you get is a not well defined installation ;)

Is there any way of obtaining a more recent version of git?

tbisanz, 2018-06-05:
Can you email the NAF admins and ask them to install `rh-git29`? That would make life so much easier!

dreyling, 2018-06-05:
What do you mean by this?
> ... that this is not...
Changing that line, I have linked, is only affecting EUDAQ, or am I am wrong?

I will mail the admins!

dreyling, 2018-06-05:
I asked the admins.

Concerning the py-installer: I checked this line in minimal example with my local git version (2.7) and it is not working due to the used syntax:
[test_eutel.py.txt](https://github.com/eutelescope/eutelescope/files/2071848/test_eutel.py.txt)

tbisanz, 2018-06-05:
Well, the syntax used in the installer will call:

`git clone https://github.com/eudaq/eudaq eudaq/v1.8.0 --branch v1.8.0`

Trying this locally:

```
lab14:tbisanz tbisanz $ git clone https://github.com/eudaq/eudaq eudaq/v1.8.0 --branch v1.8.0
Cloning into 'eudaq/v1.8.0'...
remote: Counting objects: 31160, done.
remote: Total 31160 (delta 0), reused 0 (delta 0), pack-reused 31160
Receiving objects: 100% (31160/31160), 23.31 MiB | 5.38 MiB/s, done.
Resolving deltas: 100% (18927/18927), done.
Note: checking out '052d26b4516a11d8d222aa0318633f50032b9581'.
```

with `git version 2.14.3`

tbisanz, 2018-06-05:
Yes, the line you touched only changes eudaq - that's the problem! The remaining packages will try to use the same syntax and fail, silently falling back to the master branches :)

dreyling, 2018-06-05:
Yes, you are right, but in the file `releases/v01-19/release-versions.py` 
this is used `Eudaq_version = tags/v1.8.0` which is taken here as %s
``` git clone https://github.com/eudaq/eudaq eudaq/%s --branch %s```
or am I wrong

tbisanz, 2018-06-05:
Can you link the file?

dreyling, 2018-06-05:
I thought here
https://github.com/eutelescope/iLCInstall/blob/28419eab6e6d8f1265284a8c82a98560b46b334f/releases/v01-19/release-versions.py#L238
the `Eudaq_version` variable is set, which is used here:
https://github.com/eutelescope/iLCInstall/blob/71121a61e493d1eb9e6272112de653eb7ece3df8/ilcsoft/eutelescope.py#L79
since it is set here
https://github.com/eutelescope/iLCInstall/blob/28419eab6e6d8f1265284a8c82a98560b46b334f/examples/eutelescope/release-standalone.cfg#L154
right?

tbisanz, 2018-06-05:
But that's not what you have up there ;-):

`git clone https://github.com/eudaq/eudaq eudaq/%s --branch %s" % (os.path.basename(self.env["EUDAQ_VERSION"]), os.path.basename(self.env["EUDAQ_VERSION"]))`

The `**os.path.basename**`

dreyling, 2018-06-05:
> Can you link the file? Typically we should do something like os.path.basename(Eudaq_version) as %s, i.e. take the v1.8.0 from the provided tag\branch/version.
Yes!

tbisanz, 2018-06-05:
```
fooBar = os.path.basename("tags/v1.8.0")
print(fooBar) #will print "v1.8.0"
```

tbisanz, 2018-06-05:
The package Yves Kemp suggested is the one you want, by the way!

dreyling, 2018-06-05:
Of course, you are completely right with the parsing stuff. Two things:
- I asked for the git29 installation
- I don't completely understand your argument above, if I adjusting the line in 
>  The problem is that this is not only used for eudaq, but many ilcsoft packages. They compile right now but you might have the master branches there as well! So what you get is a not well defined installation ;)

I would assume this only affects EUDAQ.

tbisanz, 2018-06-05:
Yes! What was the buggy behaviour? You tried to get EUDAQ 1.8.0 but instead checked out the master branch which didn't compile, correct?

The entire EUTelescope installation will install several packages, let's just say three for this example:

```
install Marlin("v1.2.3")
install LCIO("v3.1")
install EUTelescope("master")
install EUDAQ("v1.8.0")
```

What happens is:

```
trying to clone Marlin ...
can't get Marlin/v1.2.3, checked out Marlin/master
built Marlin/master

trying to clone LCIO ...
can't get LCIO/v3.1, checked out LCIO/master
built LCIO/master

trying to clone EUTelescope ...
built EUTelescope /master

trying to clone EUDAQ...
can't get EUDAQ/v1.8.0, checked out EUDAQ/master
ERROR: can't build EUDAQ!
```

See the problem?

dreyling, 2018-06-05:
Thanks :) So, modifying only this line would not solve the git syntax for other packages!

Okay, DESY IT is asking
> eine Nachfrage seitens der Kollegen: Wäre Centos/RHEL 7.x hinreichend für Ihre Zwecke oder muss es zwingend auch für SL6.x sein?

dreyling, 2018-06-05:
Just log here an intermediate workaround, if atlas resources are available (@edorossi):
```
$ export ATLAS_LOCAL_ROOT_BASE=/cvmfs/atlas.cern.ch/repo/ATLASLocalRootBase
$ export DQ2_LOCAL_SITE_ID=DESY-HH_SCRATCHDISK                             
$ source /cvmfs/atlas.cern.ch/repo/ATLASLocalRootBase/user/atlasLocalSetup.sh
$ lsetup git
$ git --version 
git version 2.11.1
$ lsetup gcc -h
$ lsetup "gcc gcc620_x86_64_slc6"
$ ./ilcsoft-install -i examples/eutelescope/release-standalone.cfg
...
-- Set runtime path of "/nfs/dust/atlas/user/dreyling/eutelescope/v01-19-02/Eutelescope/master/external/eudaq/v1.8.0/bin/EURunsplitter.exe" to "/nfs/dust/atlas/user/dreyling/eutelescope/v01-19-02/Eutelescope/master/external/eudaq/v1.8.0/lib:/nfs/dust/atlas/user/dreyling/eutelescope/v01-19-02/lcio/master/lib:/nfs/dust/atlas/user/dreyling/eutelescope/v01-19-02/Eutelescope/master/lib"

****************************** Finished installation ******************************

&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&
&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&

```

tbisanz, 2018-06-06:
Regarding the DESY IT question: on Centos/RHEL 7.x you need `devtoolset-6-gcc-c++` and `rh-git29` (`devtoolset-4` would also be okay).

jha92, 2018-06-07:
For me the same, running test currently ongoing with gbl_local example (looks good).

dreyling, 2018-06-07:
> @dreyling Once you get the rh-git29 package, will you test and we can close this?

@tbisanz: yes!

tbisanz, 2018-06-07:
@jha92 go into the `build` directory and type `ctest -V -R gbl*`

jha92, 2018-06-07:
@tbisanz Here it is: [ctest_output.log](https://github.com/eutelescope/eutelescope/files/2079957/ctest_output.log)
Two tests failed, but the interpretation I let to you.

tbisanz, 2018-06-07:
Hmh, that's a bit weird - but okay!

tbisanz, 2018-06-07:
Can we close this?

jha92, 2018-06-07:
Maybe waiting for the fix with a new git version on NAF and then re-install?

jha92, 2018-06-12:
NAF admins have now pushed rh-git29 on the machines.
So, new installation routine is as following:
```
//Enable devtoolset (gives you gcc 5.3)
scl enable devtoolset-4 bash

//Enable new git (gives you git 2.9.3)
source /opt/rh/rh-git29/enable

//make environment
export ILCSOFT=.../ilcsoft
mkdir -p $ILCSOFT
cd $ILCSOFT

//clone ilcinstall with desynaf branch
git clone -b dev-desynaf https://github.com/eutelescope/ilcinstall $ILCSOFT/ilcinstall 

//check versions (currently, eudaq:tags/v1.8.0 and eutelescope:master)
emacs $ILCSOFT/ilcinstall/releases/v01-19/release-versions.py 

//start installation
cd ilcinstall 
./ilcsoft-install -i examples/eutelescope/release-standalone.cfg 
```
I tested it already with running `gbl_local` example.

dreyling, 2018-06-12:
Now we have a 2.9 version of git on the cluster. Below you find the log of the out-of the box installation. I have updated this Wiki content and close this issue.

```
export ILCSOFT=/nfs/dust/USER-PATH/NAME
cd $ILCSOFT
git clone https://github.com/eutelescope/ilcinstall
cd ilcinstall
git checkout -b dev-desynaf origin/dev-desynaf
scl enable devtoolset-4 bash
scl enable rh-git29 bash
./ilcsoft-install -i examples/eutelescope/release-standalone.cfg

[...]

.../eutel-naf/v01-19-02/Eutelescope/master/lib"

****************************** Finished installation ******************************

&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&
&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&
```

closed 2018-06-12 by dreyling