eutelescope/eutelescope #423: attach DUT

XiaocongAi, 2018-06-06:
The EUTelTripletGBLUtility::AttachDUT  method assumes the DUT to be two dimensional and takes the two dimensional measurement to calculate the distance between the track and DUT. It should be modified to take into account the possibility of 1 dimensional measurement  of the DUT, e.g. the Strip sensor:
https://github.com/eutelescope/eutelescope/blob/master/eutelescope/libraries/src/EUTelTripletGBLUtility.cc#L305

tbisanz, 2018-06-06:
Hey! How do you know if you're dealing with a 1- or 2-D sensor? Did you include some mechanism for that into EUTelescope?

XiaocongAi, 2018-06-07:
In the version of Eutelescope we used for the ATLAS ITk Strip, we have the mechanism to tell Eutelescope whether it's a 1D or 2D sensor. I think we should have that mechanism here as well. I'll do some work to  include the mechanism into the master branch.

tbisanz, 2018-06-07:
I'll help with the implementation in the `EUTelTripletGBLUtility` once you merged the basic functionality!

tbisanz, 2018-06-13:
@XiaocongAi Any news on that?

XiaocongAi, 2018-06-18:
@tbisanz I have pushed some new commits to my branch: XiaocongAi:master_fixAttachDUT, but the CI test still failed somehow. Do you have any idea of the reason for the failures shown at https://api.travis-ci.org/v3/job/393578730/log.txt 
My modification should not have any impact on the no-DUT example.

edorossi, 2019-01-14:
Addressed with an alternative approach in merge request #447

closed 2019-01-14 by edorossi