eutelescope/eutelescope #426: Removing global histogram pointers in GBLAlign

tbisanz, 2018-06-07:
We need someone to refactor the histograms in the `EUTelAlignGBL` processor. I really don't have much time but would definitely help with any questions.

https://github.com/eutelescope/eutelescope/blob/master/processors/src/EUTelAlignGBL.cc#L80-L145

This is a great task to get involved ;-) @dreyling maybe you could advertise this a bit?

TCThenna, 2018-06-07:
and there is a good example on how to do this in the other gbl example :)

dreyling, 2018-06-07:
@tbisanz I am happy to advertise, if I understand 100% what to do :)

@tbisanz What do you mean exactly be refactoring?
@TCThenna Can you please point to the good example?

I just quickly looked with @jha92 on that, and we found by coincidence some histograms which are not filled, for example https://github.com/eutelescope/eutelescope/blob/2a2807aad5ca36bd299b7bce7678ea19d5a0e04d/processors/src/EUTelAlignGBL.cc#L127

And we need an explanation for the different histograms, that new people understand the cryptic abbreviations ax, dx, kx, triplet-driplet etc... @TCThenna I propose a meeting with Jan-Hendrik to go through all plots and then to discuss where and how to explain.

tbisanz, 2018-06-08:
All the histograms are currently stored in global pointers to histogram objects - there are [several](http://wiki.c2.com/?GlobalVariablesAreBad) reasons why this is not the best choice. Thus, someone should sit down and maintain them, i.e. make them not global - in this process it is also a very good idea to check if they are filled, properly names, have correct axis labels, are needed, etc.

TCThenna, 2018-06-08:
Is it done OK for the GBL example, where I put them in a different class?

tbisanz, 2018-06-08:
They are in the same class `EUTelTripletGBLKinkEstimator`, you just moved the `EUTelTripletGBLKinkEstimator::bookHistos()` into a seperate implementation file. But they are not global objects, but members of the `EUTelTripletGBLKinkEstimator`, so that's perfectly fine!

I would try to implement them in a more dynamic way, with some containers so that they scale with the amount of sensors in the set-up. E.g. some matrix-like structure:

https://github.com/eutelescope/eutelescope/blob/master/processors/include/EUTelGBLFitter.h#L363-L365

For example for cluster size-dependent histograms: `clusterSizeHistogram[sensorID][clusterSize]->fill(...)` - this allows dynamic scaling with more sensors (DUTs) etc.

TCThenna, 2018-06-08:
the fact that they are pointers is fine?

tbisanz, 2018-06-08:
Yes, they have to be, `AIDAProcessor::histogramFactory(this)` is the corresponding factory which will create the objects!

TCThenna, 2018-06-08:
BTW, the fact that they are in a separate implementation file is nice during compilation. If you change smth in the code that fills the histogramms, the histogramms themselves do not have to be recompiled! (I am mentioning this in case someone wondered why we did this in the first place).

I'll be on holidays soon, so no time to do this ...

jha92, 2018-06-11:
@dreyling and I will tackle this tomorrow.
In the meantime I compared a bit the different implementations and I am still not clear, what is the desired way:
Version 1 (as in GBLFitter): histogram definitions in header, creation with AIDAProcessor, `bookHistos()` in source file
Version 2 (as in TripletGBLKinkEstimator): histogram definitions in header, creation with AIDAProcessor, `bookHistos()` in `eutelescope/libraries/src/EUTelTripletGBLKinkEstimatorHistos.cc`

P.S: BTW, how is version 2 working cannot figure out the connection to the main source file?

TCThenna, 2018-06-11:
Is it maybe just the name of the class, and the compiler is smart enough to recognize that?

https://github.com/eutelescope/eutelescope/blob/e4a4ea8215eeeebda9c616b07553f93ec19cf4ee/eutelescope/libraries/src/EUTelTripletGBLKinkEstimatorHistos.cc#L104

tbisanz, 2018-06-12:
It's not the compiler who is smart enough, but the linker. In the header file we have a _class declaration_ this will allow us to use the name of the class as a keyword in C++. The class will have _member declarations_ for data members, for example for pointers to the histograms.

The class declaration itself is enough for us to work with the class, we know all the signatures of functions we can call, data members we can access, etc. When we build the framework, what we basically tell the compiler is "make object code out of all the .cc/.cxx/etc. files" and then we link them together.

So what happens is:

Here we build the object file:
```
[  7%] Building CXX object CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelTripletGBLKinkEstimatorHistos.cc.o
```
We can actually look into it:
```
nm CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelTripletGBLHistos.cc.o
...
00000000000000a0 T _ZN11eutelescope15EUTelTripletGBL10bookHistosEv
...
```
If you head over here: https://demangler.com/ we can demangle this:
```
eutelescope::EUTelTripletGBL::bookHistos()
```

So that's what we have in this object file! Then we'll link it:

```
[ 74%] Linking CXX shared library lib/libEutelescope.so
```

If you really want to know what's happening, let's turn on verbose output:

```
cmake -DCMAKE_VERBOSE_MAKEFILE=ON ..
make install
...
[ 74%] Linking CXX shared library lib/libEutelescope.so
/usr/bin/cmake -E cmake_link_script CMakeFiles/Eutelescope.dir/link.txt --verbose=1
/usr/bin/g++ -fPIC -std=c++11 -O2 -g -fdiagnostics-show-option -Weffc++ -Wcast-align -Wcast-qual -Wdisabled-optimization -Winit-self -Wmissing-include-dirs -Wredundant-decls -Wsign-promo -Wstrict-null-sentinel -Wswitch-default -Wundef  -shared -Wl,-soname,libEutelescope.so.1.0 -o lib/libEutelescope.so.1.0.0 CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTELESCOPE.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelAlignmentConstant.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelEtaFunctionImpl.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelEventImpl.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelGenericPixGeoDescr.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelGenericPixGeoMgr.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelGeoSupportClasses.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelGeometryTelescopeGeoDescription.cpp.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelHistogramManager.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelMatrixDecoder.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelPseudo1DHistogram.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelROI.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelRunHeaderImpl.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelTrackFinder.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelTrackFitter.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelTripletGBLHistos.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelTripletGBLKinkEstimatorHistos.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelTripletGBLUtility.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelUtility.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/GEARPixGeoDescr.cc.o CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/anyoption.cc.o CMakeFiles/Eutelescope.dir/eutelescope/interfaces/pixel/src/EUTelBaseSparsePixel.cc.o CMakeFiles/Eutelescope.dir/eutelescope/interfaces/pixel/src/EUTelGenericSparsePixel.cc.o CMakeFiles/Eutelescope.dir/eutelescope/interfaces/pixel/src/EUTelGeometricPixel.cc.o CMakeFiles/Eutelescope.dir/eutelescope/interfaces/pixel/src/EUTelMuPixel.cc.o CMakeFiles/Eutelescope.dir/eutelescope/interfaces/pixel/src/EUTelSimpleSparsePixel.cc.o CMakeFiles/Eutelescope.dir/eutelescope/interfaces/cluster/src/EUTelBrickedClusterImpl.cc.o CMakeFiles/Eutelescope.dir/eutelescope/interfaces/cluster/src/EUTelDFFClusterImpl.cc.o CMakeFiles/Eutelescope.dir/eutelescope/interfaces/cluster/src/EUTelFFClusterImpl.cc.o CMakeFiles/Eutelescope.dir/eutelescope/interfaces/cluster/src/EUTelGeometricClusterImpl.cc.o CMakeFiles/Eutelescope.dir/eutelescope/interfaces/cluster/src/EUTelSparseClusterImpl.cc.o CMakeFiles/Eutelescope.dir/external/millepede2/tags/V04-03-03/Mille.cc.o -Wl,-rpath,/work2/tbisanz/ilcsoft_test_may_18/v01-19-02/Marlin/v01-11/lib:/work2/tbisanz/ilcsoft_test_may_18/v01-19-02/lcio/master/lib:/work2/tbisanz/ilcsoft_test_may_18/v01-19-02/gear/v01-06-eutel-pre/lib:/work2/tbisanz/ilcsoft_test_may_18/v01-19-02/CLHEP/2.3.1.1/lib:/work2/tbisanz/ilcsoft_test_may_18/v01-19-02/ilcutil/v01-02-01/lib:/work2/tbisanz/ilcsoft_test_may_18/v01-19-02/RAIDA/v01-08/lib:/work2/tbisanz/ilcsoft_test_jan_18/v01-19-02/root/6.12.04/lib:/work2/tbisanz/ilcsoft_test_may_18/v01-19-02/lccd/v01-04/lib:/work2/tbisanz/ilcsoft_test_may_18/v01-19-02/GBL/V02-01-02/lib:::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::: /work2/tbisanz/ilcsoft_test_may_18/v01-19-02/Marlin/v01-11/lib/libMarlin.so /work2/tbisanz/ilcsoft_test_may_18/v01-19-02/Marlin/v01-11/lib/libMarlinXML.so /work2/tbisanz/ilcsoft_test_may_18/v01-19-02/lcio/master/lib/liblcio.so /work2/tbisanz/ilcsoft_test_may_18/v01-19-02/lcio/master/lib/libsio.so -lz /work2/tbisanz/ilcsoft_test_may_18/v01-19-02/gear/v01-06-eutel-pre/lib/libgearsurf.so /work2/tbisanz/ilcsoft_test_may_18/v01-19-02/gear/v01-06-eutel-pre/lib/ [... cut ...]
/usr/bin/cmake -E cmake_symlink_library lib/libEutelescope.so.1.0.0 lib/libEutelescope.so.1.0 lib/libEutelescope.so
...
```

Or to take the important parts:

```
[ 74%] Linking CXX shared library lib/libEutelescope.so
/usr/bin/g++ -o lib/libEutelescope.so.1.0.0  CMakeFiles/Eutelescope.dir/eutelescope/libraries/src/EUTelTripletGBLHistos.cc.o
```

We're building `lib/libEutelescope.so.1.0.0` (later we'll create a symlink to lib/libEutelescope.so) with (amongst many others) the `EUTelTripletGBLHistos.cc.o`

tbisanz, 2018-06-12:
Where you have your implementation code doesn't matter (not entirely true when you start with templates) - as long as it's there!

tbisanz, 2018-06-13:
@jha92 Sorry I missed the meeting today - do you want to quickly Skype for that?

jha92, 2018-06-13:
Me too :) Jan and I had a look yesterday and I think we will just start, but certainly come back on you with some topics.

dreyling, 2018-06-14:
There was no meeting, I will start today...

jha92, 2018-10-02:
The version we proposed was merged already (#434). So one can close this issue here?!

closed 2019-01-25 by jha92