eutelescope/eutelescope #427: Processor naming rules

jha92, 2018-06-07:
I find some processor names a bit weird and maybe now with our current developments it would be good to start with some renaming / setting rules for naming.
Examples of in point of view not so good names are all processors starting with EUTelProcessor, as it should be clear that we are talking in the `processor` folder about such. Same goes a bit for EUTelUtilityPrintEventNumber, as it is the only one from a maybe general class of EUTelUtility ones.
Most often there is a noun of the action in the name, such as Maker, Finder, or Fitter.
What do you think about this?   When deciding to change it, I am not sure if one should do it directly on the master branch as it will be maybe mess up the setup of users.

tbisanz, 2018-06-07:
Well, there is no good way - just three hours ago I closed this issue: #28 which basically told users to do the exact opposite (and is the reason we have many `EUTelProcessor...`s). 

We won't change all the names as this will break many existing reconstructions jobs.

tbisanz, 2018-06-07:
As for new naming: I agree that `Processor` is not required and more confusing than helping. If one does a complete refactoring of a processor (including changing parameter names etc.) - then the processor may also be renamed.

jha92, 2018-06-07:
Yes, I think mostly of the last: when doing a deep refactoring implying in general changes, then one should also think about all details such as the name. Therefore this should maybe also not happen directly on the master, but on a special development branch which will most probably evolve to a new tag.
In the same way goes to get rid off unused/unmaintained processors...

tbisanz, 2018-06-07:
No, don't try to diverge off into branches/tags - this will cause a huge mess, people will have different reconstructions, the CI will break with the changes, etc.

If you really want to open this renaming you can do it - but then do it once and for all right now, or leave it like it is.

tbisanz, 2018-06-12:
#428

closed 2018-06-18 by tbisanz