eutelescope/eutelescope #429: Commit 6a0c47  breaks compilation in a ubuntu 16.04

duartej, 2018-06-07:
The -Werror flag making the compilation fails because of the spread usage of NULL as pointer here and there:  "zero as null pointer constant [-Werror=zero-as-null-pointer-constant]"
Please take a look at the log file in you need more details (compiler/linker/...): https://hub.docker.com/r/duartej/eudaqv1-ubuntu/builds/bsldsxutbqatbzuetrp8wky/

tbisanz, 2018-06-07:
This is weird as I built it on Ubuntu 17 just yesterday :o but I agree, the faulty lines are there and need to be fixed! Thanks for pointing this out!

tbisanz, 2018-06-07:
Using:
```
lab11:build tbisanz [master] $ /usr/bin/g++ --version
g++ (GCC) 7.3.1 20180303 (Red Hat 7.3.1-5)
```
this is confusing:
```
...
[ 36%] Building CXX object CMakeFiles/EutelProcessors.dir/processors/src/EUTelDafBase.cc.o
/usr/bin/g++  -DDO_TESTING -DEutelProcessors_EXPORTS -DMARLINDEBUG -DUSE_AIDA -DUSE_CED -DUSE_CLHEP -DUSE_GEAR -DUSE_GSL -DUSE_MARLIN -DUSE_MARLINUTIL -isystem /usr/include/eigen3 -isystem /work1/tbisanz/ILCSOFT/v01-19-02/Marlin/v01-11/include -isystem /work1/tbisanz/ILCSOFT/v01-19-02/lcio/master/include -isystem /work1/tbisanz/ILCSOFT/v01-19-02/gear/v01-06-eutel-pre/include -isystem /work1/tbisanz/ILCSOFT/v01-19-02/gear/v01-06-eutel-pre/include/gear -isystem /work1/tbisanz/ILCSOFT/v01-19-02/CLHEP/2.3.1.1/include -isystem /work1/tbisanz/ILCSOFT/v01-19-02/ilcutil/v01-02-01/include -isystem /work1/tbisanz/ILCSOFT/v01-19-02/RAIDA/v01-08/include -isystem /work1/tbisanz/ILCSOFT/v01-19-02/root/6.13.02/include -isystem /work1/tbisanz/ILCSOFT/v01-19-02/lccd/v01-04/include -isystem /work1/tbisanz/ILCSOFT/v01-19-02/GBL/V02-01-03 -isystem /work1/tbisanz/ILCSOFT/v01-19-02/Eutelescope/master/./external -I/work1/tbisanz/ILCSOFT/v01-19-02/Eutelescope/master/./eutelescope/libraries/include -I/work1/tbisanz/ILCSOFT/v01-19-02/Eutelescope/master/./eutelescope/interfaces/pixel/include -I/work1/tbisanz/ILCSOFT/v01-19-02/Eutelescope/master/./eutelescope/interfaces/cluster/include -I/work1/tbisanz/ILCSOFT/v01-19-02/Eutelescope/master/./processors/include -I/work1/tbisanz/ILCSOFT/v01-19-02/Eutelescope/master/./datasources/include -I/work1/tbisanz/ILCSOFT/v01-19-02/Eutelescope/master/./processors/include/alibava -I/work1/tbisanz/ILCSOFT/v01-19-02/Eutelescope/master/./processors/include/cms -I/work1/tbisanz/ILCSOFT/v01-19-02/Eutelescope/master/./geometries/include -I/work1/tbisanz/ILCSOFT/v01-19-02/Eutelescope/master/./external/millepede2/tags/V04-03-09  -std=c++11 -fPIC -O2 -g -fdiagnostics-show-option -Weffc++ -Wcast-align -Wcast-qual -Wdisabled-optimization -Winit-self -Wmissing-include-dirs -Wredundant-decls -Wsign-promo -Wstrict-null-sentinel -Wswitch-default -Wundef -fPIC   -DUSE_AIDA -DMARLIN_USE_AIDA -DUSE_RAIDA -DUSEROOT -DUSE_ROOT -DMARLIN_USE_ROOT -DUSE_GBL  -Wno-effc++ -Wall -std=c++11 -Wdisabled-optimization -Wold-style-cast -pedantic -Wextra -Wcast-align -Wcast-qual -Wuseless-cast -Wzero-as-null-pointer-constant -Wundef -Wctor-dtor-privacy -Wredundant-decls -Wdeprecated -Wformat-security -Wmissing-declarations -Wmissing-include-dirs -Wmissing-include-dirs -Wswitch-default -Wno-sign-promo -Werror -o CMakeFiles/EutelProcessors.dir/processors/src/EUTelDafBase.cc.o -c /work1/tbisanz/ILCSOFT/v01-19-02/Eutelescope/master/processors/src/EUTelDafBase.cc
[ 37%] Building CXX object CMakeFiles/EutelProcessors.dir/processors/src/EUTelDafFitter.cc.o
...

```

duartej, 2018-06-08:
It could be different things, you can take a look for instance here: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=77513 .  or here: https://stackoverflow.com/questions/13091804/whats-better-to-use-in-c11-zero-or-null

But in any case, the point is that those NULLs should disappear from the code as the use of NULL is not supported anymore (I think from c++11 on, gcc-5, but not sure about this last).

tbisanz, 2018-06-08:
Very interesting! I just tried to build my own docker image with Ubtunu:16:04 and I get the same error! [Ubuntu17](https://github.com/eutelescope/eutelescope/blob/master/docker/Ubuntu17/Dockerfile) and [Ubuntu18](https://github.com/eutelescope/eutelescope/blob/master/docker/Ubuntu18/Dockerfile) work fine!

Well, NULL is still supported, it's typically some preprocessor macro defined in some headers [(see cppreference)](https://en.cppreference.com/w/cpp/types/NULL) but I fully agree that we should get rid of it, but I need some system on which I can test if I fixed all occurrences ;)

tbisanz, 2018-06-08:
Ahh, that might also explain it - maybe the standard library in Ubuntu 16.04 will map `NULL` to `0` (so an `integer`-type) while the other versions will map it to `nullptr` (so a `nullptr_t`-type).

duartej, 2018-06-08:
Sorry, yes, it is not deprecated, but  in the page you linked: 
`The macro NULL is an implementation-defined null pointer constant, which may be
.... 
an integer literal with value zero, or a prvalue of type std::nullptr_t | (since C++11)
`
This ambiguity could lead to problems, and recommendation is to move from the NULL to std::nullptr (or the explicit null you want)

duartej, 2018-06-08:
Anyway, I see you take care of that in the #430. Thanks!

tbisanz, 2018-06-08:
You don't need to convince me ;) I'm perfectly aware that we should fix it. It's just that I compile on gcc 8.1.1 and I don't see the error there, so I'd like to understand the differences!

tbisanz, 2018-06-08:
I think I merged (hopefully) all fixes - I'm building on Ubuntu 16.04 right now, but won't have time today to check (need to build root before)! If you want you can try to build, but I might still need to fix something that slipped!

duartej, 2018-06-08:
I just built my image using your last commit and everything compiles smoothly. Thank you very much for taking care of this so quickly!!

closed 2018-06-08 by duartej