eutelescope/eutelescope #439: Update default value of FixedPlane in EUTelPreAlignment.cc from 0 to 1

mbomben, 2018-07-19:
Hello @tbisanz, could you please update the default value of FixedPlane in EUTelPreAlignment.cc from 0 to 1? It is at lines 59-60. 
Otherwise condition at line 237 is never fulfilled: 
``      // identify fixed plane
      if (sensorID != _fixedID)
        continue;''

Now it reads:

``  registerOptionalParameter("FixedPlane", "SensorID of fixed plane", _fixedID,
                            0);``

while it should read:

``  registerOptionalParameter("FixedPlane", "SensorID of fixed plane", _fixedID,
                            1);``

Many thanks,
Marco Bomben

tbisanz, 2018-07-20:
@mbomben You can always change this value in the steering-files of your reconstruction. This would break all the examples and user's existing alignment templates!

What makes you want 1 instead of 0 as a default?

mbomben, 2018-07-22:
Hi @tbisanz, now that telescope planes are numbered from to 1 to 6 you are obliged to declare FixedPlane in hitmaker otherwise PreAligner will not work as it waits for planeID=0. Please have a look at the code lines I quoted above

TCThenna, 2018-07-23:
> now that telescope planes are numbered from to 1 to 6

I usually make sure to name the MIMOSA planes ID from 0 .. 5 in the eudaq config file / NI producer part. It is not a necessity to name them 1 .. 6.

tbisanz, 2018-07-25:
If there is no change of the EUDAQ default, please use the steering files then!

closed 2018-07-25 by tbisanz