eutelescope/eutelescope #442: Problem installing EUTelescope on lxplus

Aheggelund, 2018-09-05:
Hello,

I've tried a couple of times to install EUtelescope on lxplus. 
After a lot of debugging and solved problems I end up with this error: 

```
CMake Error at main/lib/CMakeLists.txt:23 (FIND_PACKAGE):
  By not providing "FindILCUTIL.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ILCUTIL", but
  CMake did not find one.

  Could not find a package configuration file provided by "ILCUTIL" with any
  of the following names:

    ILCUTILConfig.cmake
    ilcutil-config.cmake

  Add the installation prefix of "ILCUTIL" to CMAKE_PREFIX_PATH or set
  "ILCUTIL_DIR" to a directory containing one of the above files.  If
  "ILCUTIL" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/afs/cern.ch/user/a/aheggelu/TestbeamReco_RD53/v01-19-02/Eutelescope/master/external/eudaq/v1.x-dev/build/CMakeFiles/CMakeOutput.log".
See also "/afs/cern.ch/user/a/aheggelu/TestbeamReco_RD53/v01-19-02/Eutelescope/master/external/eudaq/v1.x-dev/build/CMakeFiles/CMakeError.log".
make: *** No rule to make target `install'.  Stop.

****************************** Finished installation ******************************

```
I followed the installation guide and have set the versions as follow: 

EUDAQ : branch/1.x-dev
cmake version 3.11
and cloned the 'dev-findpython' branch of eutelescope

the log can be found here: 
/afs/cern.ch/user/a/aheggelu/EUTelescope/v01-19-02/Eutelescope/master/external/eudaq/v1.x-dev/build/CMakeFiles/CMakeError.log

kwraight, 2018-09-05:
I have the same issue. I thought it was a git problem but I've checked my version is >1.9

kwraight, 2018-09-11:
I think the problem lies in the ILCutil environment variable being exported as "ilcutil" rather than "ILCUTIL_DIR". To work around, source build_env.sh in $EUTELESCOPE then export ILCUTIL_DIR=$ilcutil. EUdaq should now build with cmake and EUtelescope will follow.

Aheggelund, 2018-09-12:
You're right, the problem was indeed the "ILCUTIL_DIR". 
I set it to $ILCSOFT/Eutelescope/v01-19-02/master/ilcutil/v01-02-01/build/
where it could locate the required cmake files. 

Another error popped up regarding "Protobuf_INCLUDE_DIR" not being found, so it was set as follows:

export Protobuf_INCLUDE_DIR=/usr/lib/x86_64-redhat5E

and then everything worked !!

closed 2018-09-12 by Aheggelund