eutelescope/eutelescope #448: ERROR loading shared library

ZhangGang00, 2019-01-22:
Dear eutelescope developers,

I used the following dockerfile to install the package: https://github.com/eutelescope/eutelescope/blob/master/docker/Fedora28/Dockerfile     It was very convenient and compiled successfully. But when I run the processor "jobsub -c config.cfg -csv runlist.csv converter 499 -g", following error arose:

**22:10:37 jobsub.converter(ERROR):  ERROR loading shared library : /ilcsoft/v01-19-02/Eutelescope/master//lib/libEutelescope.so
22:10:37 jobsub.converter(INFO): ->    Trying to load DUPLICATE library 
22:10:37 jobsub.converter(INFO): 
22:10:37 jobsub.converter(INFO):  Loading shared library : /ilcsoft/v01-19-02/Eutelescope/master/lib/libEutelProcessors.so.1.0.0 (libEutelProcessors.so)
22:10:37 jobsub.converter(INFO): 
22:10:37 jobsub.converter(ERROR):  ERROR loading shared library : /ilcsoft/v01-19-02/Eutelescope/master//lib/libEutelProcessors.so
22:10:37 jobsub.converter(INFO): ->    Trying to load DUPLICATE library 
22:10:37 jobsub.converter(INFO): 
22:10:37 jobsub.converter(INFO):  Loading shared library : /ilcsoft/v01-19-02/Eutelescope/master/lib/libEutelReaders.so.1.0.0 (libEutelReaders.so)**

I checked the above library indeed exist. Cloud you please help me figure it out?

Best,
Gang

jha92, 2019-01-23:
Hello Gang,
unfortunately I could not test it myself (have to first set up docker again...)
But normally the shared library error is a sign for sourcing the environment not only once. So, may I ask you if you did before running an extra `source build_env.sh` in the `Eutelescope` folder?
It can be, that everything is already setted in the docker.
Best,
Jan-Hendrik

edorossi, 2019-01-23:
Hi Gang,
As mentioned by Jan-Hendrik, the problem should be that /ilcsoft/v01-19-02/Eutelescope/masterbuild_env.sh is sourced twice, or it is sourced after /ilcsoft/v01-19-02/init_install.sh. 
To fix it for good, open those files, and remove ":$MARLIN_DLL" from the line that starts with "MARLIN_DLL" (it should be line 90 in build_env.sh and line 90 in init_ilcsoft.sh). For example, in my case it looks like:
`export MARLIN_DLL="/nfs/dust/atlas/user/rossi/master/v01-19-02/Eutelescope/master/lib/libEutelescope.so:$EUTELESCOPE/lib/libEutelProcessors.so:$EUTELESCOPE/lib/libEutelReaders.so"`
Then, source again build_env.sh, or restart the docker.
Let me know if this helps!
Cheers,

Edo

ZhangGang00, 2019-01-23:
Dear eutelescope developers,

Thanks for your replies. It works! But I encountered following errors immediately. Do you have any idea about this error? I am using the latest Eutelescope and latest Eudaq in v1.x-dev branch.

**18:18:02 jobsub.converter(INFO): [ MESSAGE9 "NoisyPixelMaskerM26"] SensorID: 2 Rotation/Reflection matrix for this object.
18:18:02 jobsub.converter(INFO): [ MESSAGE9 "NoisyPixelMaskerM26"]          1  0   0
18:18:02 jobsub.converter(INFO): [ MESSAGE9 "NoisyPixelMaskerM26"]          0  1   0
18:18:02 jobsub.converter(INFO): [ MESSAGE9 "NoisyPixelMaskerM26"]          0  0   1
18:18:02 jobsub.converter(INFO): [ MESSAGE9 "NoisyPixelMaskerM26"] SensorID: 2 Translation vector for this object.
18:18:02 jobsub.converter(ERROR): A runtime error occured - (uncaught exception):
18:18:02 jobsub.converter(INFO): [ MESSAGE9 "NoisyPixelMaskerM26"]          0         0       305
18:18:02 jobsub.converter(ERROR): PluginManager::GetPlugin(): Unkown event type _RAW:Yarr
18:18:02 jobsub.converter(INFO): [ VERBOSE "NoisyPixelMaskerM26"] Box for sensor: 2 is: 10.6|5.3|0.01
18:18:02 jobsub.converter(ERROR): From /ilcsoft/v01-19-02/Eutelescope/master/external/eudaq/main/lib/src/PluginManager.cc:46
18:18:02 jobsub.converter(INFO): [ VERBOSE "NoisyPixelMaskerM26"] Sensor 2 has pitch: 0.0184028|0.0184028
18:18:02 jobsub.converter(ERROR): In eudaq::DataConverterPlugin& eudaq::PluginManager::GetPlugin(eudaq::PluginManager::t_eventid)
18:18:02 jobsub.converter(INFO): [ VERBOSE "NoisyPixelMaskerM26"] Rotating plane 3 to gamma: 0
18:18:02 jobsub.converter(ERROR): Marlin will have to be terminated, sorry.
18:18:02 jobsub.converter(INFO): [ MESSAGE9 "NoisyPixelMaskerM26"] THESE MATRICES ARE USED TO TAKE A POINT IN THE LOCAL FRAME AND MOVE IT TO THE GLOBAL FRAME.**

Cheers,
Gang

edorossi, 2019-01-23:
Dear Gang,

It looks like the error is related to the conversion with EUDAQ and I do not have much experience with that. Just to get an overview, for data taking did you use the same version of EUDAQ as the one installed in EUTelescope? Moreover, did you use Yarr as DAQ for your DUT (since it looks like there is something wrong with Yarr in the error message)?

Cheers, 

Edo

ZhangGang00, 2019-01-24:
Dear eutelescope developers,

After downloading  YarrConverterPlugin.cc from https://gitlab.cern.ch/YARR/YarrEudaqProducer/blob/master/YarrConverterPlugin.cc and recompiling, the whole package eventually works!  However, I am confused about the meaning of rotation1~rotation4 in gear.xml file. I supposed they were matrix element of SO(2) :
cos(\phi)       -sin(\phi)
sin(\phi)        cos(\phi) 
But they seem not. Could you remind me what the meaning of them?

Cheers,
Gang

edorossi, 2019-01-24:
Dear Gang,

That's almost right! Those parameters represent actually represent a 2x2 matrix, but its determinant can be negative and I am not sure it needs to be orthogonal. This matrix represent the transformation of the XY plane from the local coordinate system of the device, to the XY plane of the global (telescope) coordinate system, but it is not just a simple rotation.

For example, usually this is a identity matrix, thus with rotation1=rotation4=1 and rotation2=rotation3=0. And up to here this is exactly as your case.

However, it is possible to invert X and Y (X(local)->Y(global) and Y(local)->X(global)) by using rotation1=rotation4=0 and rotation2=rotation3=1. Note that this can't be done using your matrix, since it represent a rotation in XY plane followed by an inversion of the ZY plane.

It is possible also to invert single planes. For example, by setting rotation1=-1 and rotation4=1, X would be inverted (X(local)->-X(global)) while Y would maintain its direction. Again, this could not be done using a single rotation in the XY plane.

I hope it is clear now what these parameters are! In any case, it is totally equivalent to use rotationXY, rotationZX and rotationZY to perform such operation. 
We usually just use these parameters just when X and Y are inverted (a 90 degree rotation in XY plane, basically) and/or when one direction is inverted between the local and global coordinate systems. In these cases there are only 2 entries in rotation[1-4] and they are either 1 or -1.

Cheers,

Edo

jha92, 2019-01-25:
I assume the original cause for this issue is fixed. The idea is to prevent a double loading by improving the `build_env.sh.` If any further questions, @ZhangGang00, we are happy to help.

closed 2019-01-25 by jha92

ZhangGang00, 2019-01-25:
Hi Edo,

Thanks for the detailed explanation. Could you remaind me what the matrix for Y(local)->-Y(global) is? In my case, the X and Y of DUT plans are inverted, then X and Y are both reflected, i.e. X(local)-> -Y(global) and Y(local)-> -X(global). I think the matrix should be ratation1=-1, ratation2=1, ratation3=1, ratation1=-1. Is it right?

Cheers,
Gang

edorossi, 2019-01-25:
Dear Gang,

Basically, the matrix is:

rotation1 rotation2
rotation3 rotation4

So, the relationship between local and global is:

X(global) = rotation1 * X(local) + rotation2 * Y(local)
Y(global) = rotation3 * X(local) + rotation4 * Y(local)

So in your case, you should use rotation2 = rotation3 = -1 and rotation1 = rotation4 = 0.

The matrix you see in the output is the one obtained after rotationXY etc are applied. Are you sure you don't have a 180 degrees rotation there?
In any case, I am not sure of what happens when a non-orthogonal matrix is used, so this may be the problem as well.

As a side note, effectively, the rotations (and shifts) are not used before the hitmaker. To find out the right rotations, my suggestion is to run up to the hitmaker and then check the correlations. If all the correlations are right,  then the rotations are right as well. If you see anti-correlations, it means that there is a reflection involved, if you don't see any kind of correlations, normally there is an invertion involved (X->Y and Y->X).
I hope this helps!

Cheers, 

E.