eutelescope/eutelescope #451: Error when runing jobsub.converter

ZhangGang00, 2019-01-31:
Dear eutelescope developers,

When I run jobsub.converter processor, following error happens:

**23:17:31 jobsub.converter(WARNING): [ WARNING3 "NoisyPixelMaskerAPIX"] EUTelGeometryTelescopeGeoDescription: Geometry already initialized, using old initialization
23:17:31 jobsub.converter(ERROR): [ ERROR5 "NoisyPixelMaskerAPIX"] Pixel: 165|188 on plane: 13 fired.
23:17:31 jobsub.converter(ERROR): [ ERROR5 "NoisyPixelMaskerAPIX"] This pixel is out of the range defined by the geometry. Either your data is corrupted or your pixel geometry not specified correctly!
23:17:31 jobsub.converter(ERROR): [ ERROR5 "NoisyPixelMaskerAPIX"] Pixel: 167|188 on plane: 13 fired.
23:17:31 jobsub.converter(ERROR): [ ERROR5 "NoisyPixelMaskerAPIX"] This pixel is out of the range defined by the geometry. Either your data is corrupted or your pixel geometry not specified correctly!
23:17:31 jobsub.converter(ERROR): [ ERROR5 "NoisyPixelMaskerAPIX"] Pixel: 168|188 on plane: 13 fired.
23:17:31 jobsub.converter(ERROR): [ ERROR5 "NoisyPixelMaskerAPIX"] This pixel is out of the range defined by the geometry. Either your data is corrupted or your pixel geometry not specified correctly!**

It seems that it cannot read the DUT arguments we set in gear file. I didn't define plane 13 at all. Do you have any idea?  I have attached the gear, steering and config files.

Best,
Gang
[Archive.zip](https://github.com/eutelescope/eutelescope/files/2818733/Archive.zip)

jha92, 2019-01-31:
Dear Gang,
thanks for giving us these information. Nonetheless, it would be good to know your input data. For me it seems, that there was during data-taking in EUDAQ a plane number 13 defined. Can you maybe link the input raw file to your afs-public, so that we can have a look?
Best,
Jan-Hendrik

ZhangGang00, 2019-02-01:
Hi Jan-Hendrik,

You are right. Somebody updated YarrConverterPlugin.cc in EUDAQ which label DUTs plane with number 13. We have fix this problem.  But there is still some error. I have attached one raw file and YarrConverterPlugin.cc.

**00:27:48 jobsub.converter(INFO): [ MESSAGE9 "NoisyPixelMaskerM26"] SensorID: 2 Rotation/Reflection matrix for this object.
00:27:48 jobsub.converter(INFO): [ MESSAGE9 "NoisyPixelMaskerM26"]          1  0   0
00:27:48 jobsub.converter(INFO): [ MESSAGE9 "NoisyPixelMaskerM26"]          0  1   0
00:27:48 jobsub.converter(ERROR): ***********************************************
00:27:48 jobsub.converter(INFO): [ MESSAGE9 "NoisyPixelMaskerM26"]          0  0   1
00:27:48 jobsub.converter(ERROR): A runtime error occured - (uncaught exception):
00:27:48 jobsub.converter(INFO): [ MESSAGE9 "NoisyPixelMaskerM26"] SensorID: 2 Translation vector for this object.
00:27:48 jobsub.converter(ERROR): lcio::Exception:  BitFieldValue 'sensorID': out of range : 200 for width 7
00:27:48 jobsub.converter(INFO): [ MESSAGE9 "NoisyPixelMaskerM26"]          0         0       305
00:27:48 jobsub.converter(ERROR): Marlin will have to be terminated, sorry.
00:27:48 jobsub.converter(INFO): [ VERBOSE "NoisyPixelMaskerM26"] Box for sensor: 2 is: 10.6|5.3|0.01
00:27:48 jobsub.converter(ERROR): ***********************************************
00:27:48 jobsub.converter(INFO): [ VERBOSE "NoisyPixelMaskerM26"] Sensor 2 has pitch: 0.0184028|0.0184028
00:27:48 jobsub.converter(ERROR): 
00:27:48 jobsub.converter(INFO): [ VERBOSE "NoisyPixelMaskerM26"] Rotating plane 200 to gamma: 0
00:27:48 jobsub.converter(INFO): [ MESSAGE9 "NoisyPixelMaskerM26"] THESE MATRICES ARE USED TO TAKE A POINT IN THE LOCAL FRAME AND MOVE IT TO THE GLOBAL FRAME.**

Best,
Gang
[Archive2.zip](https://github.com/eutelescope/eutelescope/files/2819620/Archive2.zip)

edorossi, 2019-02-02:
Hi Gang,

It looks like LCIO doesn't like IDs >128. Can you modify the converter plugin to assign smaller numbers as ID?

Cheers,

Edo

ZhangGang00, 2019-02-02:
Dear eutelescope developers,

Thank you very much! I have got meaningful results using Eutelescope. I have a last question. In gear.xml, what is the difference between element "ladder" and "sensitive"? I noticed they are quite similar.

Best,
Gang