eutelescope/eutelescope #455: Millepede cannot find binary file

ZhangGang00, 2019-03-14:
Dear developers,

I am using version 2.0-alpha. In align step, following error happened(details in align_output.txt):

**20:24:00 jobsub.align(ERROR): [ ERROR5 "PedeToGEAR"] STOP FILETC: no binary files
20:24:00 jobsub.align(ERROR): [ ERROR5 "PedeToGEAR"] Problem during Pede execution, exit status: 0, error messages (repeated here):
20:24:00 jobsub.align(ERROR): [ ERROR5 "PedeToGEAR"] STOP FILETC: no binary files
20:24:00 jobsub.align(ERROR): [ ERROR5 "PedeToGEAR"]
20:24:00 jobsub.align(ERROR): [ ERROR5 "PedeToGEAR"] Will exit now**

But I checked run001568-align-mille.bin indeed exist. Could you please help me figure it out?

Thanks,
Gang
[align_output.txt](https://github.com/eutelescope/eutelescope/files/2964611/align_output.txt)

edorossi, 2019-03-14:
Dear Gang,

Can you try run again in the exact same way?

I think the problem may be the order of execution of the processors. In the align steering file you need to execute EUTelPedeGear before the Align processor. If you don't, the binary file is created AFTER you have tried to process it with EUTelPedeGear.

Cheers,

Edo

ZhangGang00, 2019-03-14:
Dear developers,

Let me tell you the story. I am working on SLAC testbeam data using 11 GeV electron beam. However, I got much worse spatial resolution for all planes than what we expected using EUTelescope v2.0-alpha. I asked help from some testbeam experts, and they said the bad spatial resolution caused by DAF fitter. But I knew someone got reasonable resolution using DAF fitter for CERN testbeam data in June where the beam particles are 120 GeV protons. So I requested the raw data of CERN testbeam to check if I can get good resolution using DAF fitter for this testbeam data. **DAF cannot deal with multiple scattering well, right?** But I encounter following error in converter step:

**18:52:17 jobsub.converter(INFO): [ VERBOSE "UNKOWN"] GeoID : 43
18:52:17 jobsub.converter(INFO): [ VERBOSE "UNKOWN"] Name : C:/opt/eudaq_v17dev/conf/AIDA2020_WP7_June2018/AIDA_RD53A_BDAQ_FEI4.conf
18:52:17 jobsub.converter(INFO): [ VERBOSE "UNKOWN"] timeDelay : 0
18:52:17 jobsub.converter(INFO): [ VERBOSE "UNKOWN"] Test confi param: 192.168.4.2
18:52:17 jobsub.converter(INFO): [ VERBOSE "UNKOWN"] Added USBPix Board: -1 to list!
18:52:17 jobsub.converter(INFO): [ MESSAGE3 "Marlin"] Deleting Mimosa26.so
18:52:17 jobsub(ERROR): Marlin returned with error code 1
18:52:17 jobsub(WARNING): There were 1 error messages reported**

So I also requested converter-slcio file to skip converter step. Then the strange Millepede error happened. The same steering files works well for other data. The Millepede binary file will be reproduced every time in align step. I tried many times, so I am afraid the problem is not what you said. Anyway, now I moved to v2.0.0 and use GBL fitter. The resolution in alignGBL step already look good. But I encounter some problems in last fitGBL step. I will open another issue.

Best,
Gang

edorossi, 2019-03-14:
Dear Gang,

Independently from the resolution, I think passing to GBL is the best choice. The GBL implementation is rather new and clean and we can help you much more than with DAF. Let us just know the problem you have and I am pretty sure you can solve it rather quickly.
As you said, GBL has better resolution, in particular at low energies. We know that the difference is quite significant at DESY energies (<6GeV), and probably it is the same at 11GeV as your beam. At 120GeV basically all the fitting methods should give similar results, since the multiple scattering is "negligible".

I don't fully understand the problem with the converter. If you are starting from an lcio, the best strategy is to remove the execution of EUTelNativeReader processor from the converter steering file (in this way you basically obtain the noisypixel steering file of GBL_DUT example). Can this be somehow behind the problem you had with Millepede?

Cheers,

Edo

jha92, 2019-03-15:
Hi,
yesterday in the ITk pixel testbeam workshop, I had also one occasion of your above error, not finding the binary even though it was created. But unfortunately, it was one person out of 8 or so and they were using the same settings. So it is a bit hard to solve this issue, even though we will try further.
We will come back to this.

Cheers,
Jan-Hendrik

edorossi, 2019-03-15:
Hi Jan-Hendrik,

So you see this problem on lxplus or NAF? I noticed that on lxplus sometimes error like this happen, and you just need a fresh terminal to fix it. For this reason, I think it may just not be a EUTelescope problem (but we should surely investigate a bit more).

Cheers,

Edo

jha92, 2019-03-15:
I think it was indeed on LXplus happening. Anyway, there should be then a reason for this faulty behavior...

edorossi, 2019-03-15:
Hi Jan-Hendrik,

I think we should wait a couple of weeks before digging deeper. I have never heard of similar problems with EUTelescope on lxplus until recently. Moreover, I recalled some similar issue on NAF a while ago. and it disappeared after a few weeks. It may be just a temporary hiccup (related to the migration to eos perhaps?).

Cheers,

Edo

ZhangGang00, 2019-03-15:
Dear developers,

I just work on the standalone docker image, not on lxplus.

Cheers,
Gang

tofitsch, 2019-03-19:
Hi,
The problem was caused by some sloppy string comparison in millipede:
To find the binary file(s), millipede reads the steering file (for example `output/database/run000272-pede-steer.txt`). For me, the first two lines of this file look like this:
```
  Cfiles
  /afs/cern.ch/work/t/tofitsch/euTelescope/workshop/output/database/run000272-align-mille.bin
```
Now, millipede reads this file line by line. In this process, it skips the line containing 'Cfiles' because it is only interested in the file path. This is defined in the following lines of `Eutelescope/master/external/millepede2/tags/V04-03-09/pede.f90`:
```
7159         keystx='Cfiles'
7160         mat=matint(text(ia:ib),keystx,npat,ntext)
7161         IF(mat == ntext) THEN ! exact matching
7162             nuf=1
7163             !         WRITE(*,*) 'Cfiles'
7164             CYCLE
7165         END IF
```
Here, `text(ia:ib)` is the line that was just read from the steering file. So, matint(...) compares the string 'Cfiles' with the line and returns the number of matching characters. If this number is equal to the number of characters in 'Cfiles' (i.e. `6`), the line is skipped.
The problem here is, that matint(...) does not care if there are non-matching characters in the line in-between the matching ones (and it is also not case-sensitive). The only thing that matters is, that the characters are in the correct order. As a consequence of this, matint(...) returns the same value `6` for all of these examples: 
```
matint('Cfiles','Cfiles',npat,ntext)
matint('cFiLeS','Cfiles',npat,ntext)
matint('cXFiXLeXS','Cfiles',npat,ntext)
```
With that in mind, take a look at the path to my binary file again:
/afs/**c**ern.ch/work/t/to**fi**tsch/euTe**les**cope/workshop/output/database/run000272-align-mille.bin

and the path to Gang's binary file:
/il**c**so**f**t/v01-19-02/Eutelescope/master/jobsub/examples/Jul**i**en_examp**le**/output/databa**s**e/run001568-align-mille.bin

Until this is fixed in the code, you can prevent this error by taking care that the path to your binary files do not contain the characters `cfiles` in that order. In my example, the easiest way to do this was to move my working directory `workshop` out of the `euTelescope` folder. This also explains why I was the only one in the workshop who encountered this error, even though I followed the same instructions as the other participants. In the end, the problem was just that I have a `fi` in my name.

Cheers,
Tobias **Fi**tschen

jha92, 2019-03-19:
Hi Tobias,
that is kind of crazy, thanks a lot for digging that far in the millipede fortran code!
I am not sure, if we can change it in Millipede (maybe we give them your explanation, what is triggering errors), but in general it is good to know!
Cheers,
Jan-Hendrik

TCThenna, 2019-03-20:
We should tell this to Claus, I think he is responsible for MPII code.

ckleinw, 2019-04-30:
Problem fixed with Millepede-II V04-05-00:

190430: Update of (approximate) string matching for keyword detection. Matching is now symmetric in pattern and text. Previously e.g. a binary file with the letters from 'Cfiles' in the name in that order was treated as that keyword and not as a binary file.

 Greetings Claus

edorossi, 2019-05-10:
Thanks a lot @cleinw, I did a quick try and I can't replicate the error anymore!

I changed the dev-base of ilcinstall to use MP-II V04-05-00.

@tofitsch I hope this will solve your problem, and thanks for investigating that far for finding the problem!

I close the issue since the original problem seems to be solved.

closed 2019-05-10 by edorossi

ckleinw, 2019-05-10:
Hi Edo,

please change to V05-05-01, V04-05-00 has a problem is the modifies parser.

      Greetings Claus

> Thanks a lot @cleinw, I did a quick try and I can't replicate the error anymore!
>
> I changed the dev-base of ilcinstall to use MP-II V04-05-00.
>
> @tofitsch I hope this will solve your problem, and [...]

+-------------------------------------------------------------------+
|    Claus Kleinwort                                                |
|    H1, CMS, Belle-II Collaboration                                |
|    DESY / Belle                                                   |
|    Office 1a-O2.119               phone :   0049 +40 8998 3545    |
|    Notkestrasse 85                fax   :   0049 +40 8994 3545    |
|    22603 Hamburg                  email : Claus.Kleinwort@desy.de |
+-------------------------------------------------------------------+

edorossi, 2019-05-10:
Hi Claus,

Thanks for pointing that out. The tests I did were actually with V05-05-01, but today I got confused when I came back to this issue. I will change it immediately!

Cheers,

Edo

ckleinw, 2019-05-11:
Hi Edo,

> On 10. May 2019, at 22:04, Edo <notifications@github.com> wrote:
> 
> Thanks for pointing that out. The tests I did were actually with V05-05-01, but today I got confused when I came back to this issue. I will change it immediately!
> 
> 
sorry again for a type in the hurry: the latest version is V04-05-01 !

    Greetings Clas
 
+-------------------------------------------------------------------+
|    Claus Kleinwort                                                |
|    H1, CMS, Belle-II Collaboration                                |
|    DESY / Belle                                                   |
|    Office 1a-O2.119               phone :   0049 +40 8998 3545    |
|    Notkestrasse 85                fax   :   0049 +40 8994 3545    |
|    22603 Hamburg                  email : Claus.Kleinwort@desy.de |
+-------------------------------------------------------------------+