eutelescope/eutelescope #456: Noisy pixel finder in v2.0.0

ZhangGang00, 2019-03-14:
Dear developers,

When I use v2.0.0, overwhelming following warnings happened:

**[ WARNING2 "NoisyPixelMaskerM26"] Input collection not found in the current event. Skipping...lcio::DataNotAvailableException: LCEventImpl::getCollection: collection not in event:zsdata**

And I got 0 hot pixel for all Mimosa26. But hot pixels in Mimosa26 can be found with the same frequency cut using v2.0.0-alpha. I used the same noisypixel steering file as GBL_DUT example. So what is the problem here?

Raw file is here: https://drive.google.com/open?id=1MTbGKSGz8bkswgyhe1wGOoo-3sRWflWc

I also have a small question. Before I source build_env.sh, I noticed some environment variables have been set, such as $EUDAQ and $MARLIN_DLL. I wonder which file is called automatically to set the environment?

Cheers,
Gang

jha92, 2019-03-14:
Dear Gang,

to start answering your question (without having a look on your raw file), I expect the error comes from the name of the collection of the Mimosa26. In the GBL_DUT example, data were taken with EUDAQ2 and as well converted with a non-up-to-date converter. Therefore, it can be that the names in the steering-file of the noisypixel is not matching what is in your data.

Following questions come later. Hope to start with this at least already a bit.

edorossi, 2019-03-14:
Dear Gang,

As pointed out by @jha92 , the problem is probably that the name of the mimosa collection changes with different eudaq versions. In the example it is set to "zsdata", but often times it is "zsdata_m26". You need to change your steering files accordingly.
Moreover, the GBL_DUT example starts from an LCIO file, since the conversion is done externally with EUDAQ2. If you run using EUDAQ1 you need to execute the converter in the noisypixel steering file (the code is already there but it is commented).

If none of these is the problem, can you send us a log file?

Cheers,

Edo

ZhangGang00, 2019-03-14:
Dear developers,

Thank you very much! I changed "zsdata" to "zsdata_m26", and it works! But a problem happened in fitGBL step. I will open another issue.

Chheers,
Gang

edorossi, 2019-03-14:
Dear Gang, 

I am happy it worked! The fitGBL is the last step of the reconstruction, so it is already a huge progress :)
I close the issue since the problem has been solved.

Cheers,

Edo

closed 2019-03-14 by edorossi