eutelescope/eutelescope #457: AconiteQuadFEI4_DAF need bug fixes

jha92, 2019-03-14:
The example "AconiteQuadFEI4_DAF" is currently not working as expected (it is an updated copy of the example "legacy/aconite4ChipLocal").

On the one hand, some minor fixes in the steering-files are there.
On the other hand, the results from the original example cannot be reproduced.

I am working on it.

edorossi, 2019-03-14:
Hi @jha92 , what error do you see? I want to try it very quickly to check again some changes I made a while ago to EUTelMille.

Cheers,

Edo

jha92, 2019-03-14:
It is mostly some copy-paste-stuff to make it nice and out-the-box. And then I wanted to compare in detail the outputs from both variants (first, I saw the effect of not activating the flag "UnitConversion").

tbisanz, 2019-05-21:
Won't work as you don't use the pixel layouts anymore ;)