eutelescope/eutelescope #459: No tracks is found in fitGBL step

ZhangGang00, 2019-03-15:
Dear developers,

I am using v2.0.0 EUTelescope and moving from DAF fitter to GBL fitter now. The example GBL_DUT is nice, and I can run it successfully. Then I modified this template for my testbeam analysis. I even got good results in 3rd iteration of alignGBL. But in fitGBL step, it always reported 0 track was found. It is strange as the steearing file and config are almost the same as example GBL_DUT.  Could you help me figure it out? Config, steering and log files are attached.

Cheers,
Gang
[files.zip](https://github.com/eutelescope/eutelescope/files/2968910/files.zip)

edorossi, 2019-03-15:
Dear Gang,

The reason is the flag "RequiredPlane". You are fitting only tracks with a hit on plane 6, but since there is no such plane, you are not fitting any track. If you don't have any reference plane, you can just leave this parameter empty and then no requirement will be set.

I see a few suspicious things in the alignment, though. Can you fix the x rotation and y rotation for all the planes in the alignment steps, if you don't need them explicitly? I think they are creating problems even though they shouldn't. I will investigate this, but fixing these rotations should improve things without any loss of resolution.
Can you run again the alignment with this fix and then put the logs here? 

Cheers,

Edo

ZhangGang00, 2019-03-15:
Dear Edo,

Thank you very much! When I left "RequiredPlane" empty, the problem disappeared. I tried to fix X and Y rotation, however, the final results didn't change a lot. But it indeed influnced the alignment. The final chi2/ndf is 3.4. Is it OK? And the resolution of DUTs are still worse than we expected.

Cheers,
Gang
[logs.zip](https://github.com/eutelescope/eutelescope/files/2969297/logs.zip)

edorossi, 2019-03-15:
Dear Gang,

A chi2 of 3.4 is not great: are you taking into account all the radiation length of the planes in the gear file?

Regarding the DUT resolution, how are you estimating that? The residuals are affected both by DUT resolution and tracking resolution, and to obtain the first, you need first to take into account the latter.

Cheers,

Edo

Cheers,

Edo

TCThenna, 2019-03-15:
>  The final chi2/ndf is 3.4
the chi2 of what?

ZhangGang00, 2019-03-15:
Dear developers,

The Sum(chi2)/Sum(ndf)=3.4 in Millepede. The avarage chi2/ndf equals to 2.3 for tracks. 

As for resolution of DUTs, for 50\*50 microns RD53A pixel module, the expected resolution should be pitch/sqrt(12)=14.4 microns, however, I got 26 microns. The track resolution should be just 2~3 microns, right? 

Usually, we set thickness of Mimosa26 as 30 microns. But I noticed in example GBL_DUT and GBL_noDUT, thickness of Mimosa26 are 71.3 microns. And in examples, you set the radiation length of DUTs as the same as Mimosa26 which is 93.66.

I have attached all the gear, config and steering files before in this issue. I also attached the output histograms this time. Could you give me some suggestion on which possible parameters in configuration I should change? 

Cheers,
Gang
[histograms.zip](https://github.com/eutelescope/eutelescope/files/2972376/histograms.zip)

edorossi, 2019-03-15:
Dear Gang,

The tracking resolution depends a lot on your setup, since it depends critically on material on the beam and distance between planes, for example. Since the closest mimosa planes are at >10cm from your DUT and this will degrade the tracking resolution wrt the values you mentioned. 
I think the best way to estimate the tracking resolution is to run again fitGBL while excluding also a plane close to the DUTs (plane 3, for example). In this way you obtain unbiased residuals for plane 3 and, under the approximation that everything is gaussian, the residuals will have a sigma of:

sigma_residuals^2 = sigma_tracking^2 + resolution_mimosa^2

You can resolve this equation to have an estimate of the tracking resolution, assuming a mimosa resolution of 4.5um. @TCThenna can comment if this is right. If you can let us know the tracking resolution obtained in this way it can really help understanding more your results.
As a side note, I suggest you to obtain the residuals standard deviation using a gaussian fit. If you just use the simple standard deviation from the plots, a few outliers will dominate it. For example, the standard deviation in X for the plane 50 obtained with a gaussian fit is about 19um, which is much better than what you mentioned. In any case, I see that the resolutions are worse than they should be, and I will let you know in a few days after I put some more thought on it.
 
Regarding the thickness of the Mimosas, we are setting the radiation length of silicon, which is the active area for mimosas and DUT. However, for mimosas, we are setting an effective thickness of 71.3um, which takes into account both the silicon and the kapton layers that surround it. 

Can you post the latest logs from the alignment? There may be something useful for understanding more the situation there.

Cheers,

Edo

ZhangGang00, 2019-03-15:
Hi Edo,

I attached the log files in this reply before. https://github.com/eutelescope/eutelescope/issues/459#issuecomment-473143620

Cheers,
Gang

edorossi, 2019-03-15:
Dear Gang,

I would need the new logs after you fixed the rotations in x and y.

Cheers,

Edo

ZhangGang00, 2019-03-15:
Hi Edo,

I indeed fix x and y rotation in config for those logs.
FixedPlanes = 0 5
FixedXShift = 
FixedYShift =
FixedZShift = 0 1 2 51 50 3 4 5 
FixedXRot = 0 1 2 51 50 3 4 5
FixedYRot = 0 1 2 51 50 3 4 5
FixedZRot = 

Cheers,
Gang

ZhangGang00, 2019-03-16:
Dear developers,

Taking all material(kapton around Mimosa26, copper and front-end on DUT sensor) into account, I have got super good resolution for Mimosa26 and reasonable resolution for DUTs(see the output.zip). Thanks for providing us so powerful EUTelescope v2.0.0! 

Following may be beyond this issue:

1) The title of x and y axis in correlation plots should be exchanged according to the correlation plots of our testbeam. plane0 should be ~20 cm in x direction. Could you check the relevant source codes?
[image]

2) Is it possible to deal with the raw files of different runs at the same time to get more statistics in EUTelescope? I mean combining several raw files into one.

Cheers,
Gang
[output.zip](https://github.com/eutelescope/eutelescope/files/2973458/output.zip)

edorossi, 2019-03-16:
Dear Gang,

I am glad it works! The material is very critical for the tracking and it is crucial to have good results. Regarding your points:

1) I am not sure what you mean here. The orientation of the correlation plots in EUTelescope and EUDAQ may be different, since in EUTelescope we apply the rotations from the gear, which doesn't happens in EUDAQ. Regarding the shift, this plot is in global coordinate system, where, with an align geometry, you don't expect to have any shift in x and y between the devices. In EUDAQ, the plot is basically made by comparing local coordinate system and that's why the shift appears. If you check the EUTelescope correlation using the geometry before the alignment, you should see something rather similar to EUDAQ.

2) I know there is a way to do it, but I have never used it. @TCThenna may know more about this.

Cheers,

Edo

ZhangGang00, 2019-03-16:
Hi Edo,

> I am not sure what you mean here. The orientation of the correlation plots in EUTelescope and EUDAQ may be different, since in EUTelescope we apply the rotations from the gear, which doesn't happens [...]

I mean the correlation plots in output/histograms/correlator.

Cheers,
Gang

edorossi, 2019-03-16:
Dear Gang, 

Then I did not understand your question. Why do you say that the x and y axis should be exchanged in the correlation plot? And why there should be a 20cm shift?

Cheers,

Edo

ZhangGang00, 2019-03-16:
Hi Edo,

I just mean the size of plane0 in x direction is ~20 mm, and the size of plane50 in x direction is ~10 mm. So the title of x and y axis of above plot should be exchanged.

Cheers,
Gang

jha92, 2019-03-18:
Hi Gang,
I think I see your point, this is always a bit confusing, but you are right about the switch of axis relative to how the data is filled. I wrote it down on my list.
In general: if your DUT sensor is only 10mm wide, why not specify it in the gear file. As you can see in the plot, this parameter is taken and in your gear file the sizeX is 20mm.

Cheers,
Jan-Hendrik

TCThenna, 2019-03-18:
> sigma_residuals^2 = sigma_tracking^2 + resolution_mimosa^2
>
> You can resolve this equation to have an estimate of the tracking resolution, assuming a mimosa resolution of 4.5um. @TCThenna can comment if this is right.

the 4.5 um is the expected single-point resolution of a sensor. The track resolution however is the accuracy with which you can point onto a sensor (which is not the same).

You can read our paper for more insight:
https://epjti.epj.org/articles/epjti/abs/2016/01/40485_2016_Article_33/40485_2016_Article_33.html

Or just simulate your track resolution using
https://github.com/simonspa/resolution-simulator/tree/master

TCThenna, 2019-03-18:
> I know there is a way to do it, but I have never used it. @TCThenna may know more about this.

You can either use a "hadd.C" method (which I think is available from ROOT tutorials) and just adds all histograms from input files. 

The old jobsub.py had an option for this
http://eutelescope.web.cern.ch/content/job-submission#sec-4-4
Have a look at "4.4 Concatenation"

@jha92 Do you know if this also works now?

ZhangGang00, 2019-03-18:
> Hi Gang,
> I think I see your point, this is always a bit confusing, but you are right about the switch of axis relative to how the data is filled. I wrote it down on my list.
> In general: if your DUT [...]

DUTs were rotated by 90 degree in our testbeam for some reason.

jha92, 2019-04-09:
Only slight update to this with commits https://github.com/eutelescope/eutelescope/commit/4e36a23412d0fb656bb858ecc858e429ae50d7c4 and https://github.com/eutelescope/eutelescope/commit/46fec110b7f7570f9e50e6b042f2f6956db8e992