eutelescope/eutelescope #460: Marlin dryrun not working

maweer, 2019-03-20:
Hello all,

I wanted to see which processors are there with the default values (I'm on the master-branch), so I started the "Marlin -x -dryrun > test.log" which worked fine in older versions. But now I get a strange error message from Marlin (attached below) and the test.log is not created.

Cheers,
Mareike
[error_marlin.txt](https://github.com/eutelescope/eutelescope/files/2989231/error_marlin.txt)

edorossi, 2019-03-21:
Hi Mareike,

Unfortunately I can't be of much help now. I tried to run the same command in my installation and I see an error very similar to yours, but then in my case the file gets created (I attach it [here](https://github.com/eutelescope/eutelescope/files/2990510/test.log) just in case). I will investigate this though.

Cheers,

Edo

jha92, 2019-03-21:
Hi Mareike,
are you sure, that this syntax worked before?
What I know (and which I refer now to the web page documentation) is the following:

> Creating a base steering file using Marlin
>
>By running the command
>
>`Marlin -x > steering-base.xml`
>
>you can produce the steering file steering-base.xml which features the basic structure of a steering file and contains the options of every processor known to Marlin. While all optional settings are commented out, each processor's mandatory options are active and set to their default values as given in the processor source code.

There exists also an option `--dry-run`, but this belongs to the `jobsub` command and will produce the filled steering-files in the first place before execution with Marlin in an extra step.

Hope this helps to clarify,
Jan-Hendrik

maweer, 2019-03-21:
Hello Jan-Hendrik,
I'm sure it worked before without any error message (with the version I installed on 07.09.18 I do not get an error). 
I tried your suggestion with the latest version(master): Marlin -x > steering-base.xml, but I get the same error as described in my first post. 

Thanks for the help,
Mareike

edorossi, 2019-04-09:
Hi Mareike,

After some investigation, I found out that the problem is related to the library "libEutelReaders.so". @jha92 may comment on this, but the main point is that we have basically removed these Readers, so the library should not be needed anymore. If you open the build_env.sh and check the variable MARLIN_DLL, you can remove the path to that library and then Marlin -x should work fine again (after resourcing that file). 
We will try to fix the problem and let you know more!

Cheers,

Edo

jha92, 2019-04-09:
Hi Edo,

yes, it is again related to the libEutelReaders even though I cannot understand why. The occurring error before was solved by adding the dummy shared library but it seems not for this.

So, for you Mareike, you can do like Edo say. I wrote back at the time the following to do: https://github.com/beam-telescopes/eutelescope/issues/11

I would plan now for the next release to break the libEutelReaders thing completely, but stating it in the release message to do this clean-up of the MARLIN_DLL. In parallel, we were anyway planning to also make a release for ilCinstall to make fresh installations also work with it from scratch.

Cheers,
J-H

maweer, 2019-04-23:
Hi Edo and J-H,
I commented the MARLIN_DLL out in my build_end.sh and tried Marlin -x -dryrun > test.log again. You are right: the error is gone. But I only get a very view processors instead of all processors which are available...  So just commenting it out won't solve the problem.
I added both files here, so you can see the deferences.

Cheers,
Mareike
[test_withoutMARLIN_DLL.log](https://github.com/eutelescope/eutelescope/files/3106849/test_withoutMARLIN_DLL.log)
[test_withMARLIN_DLL.log](https://github.com/eutelescope/eutelescope/files/3106851/test_withMARLIN_DLL.log)

edorossi, 2019-04-25:
Hi Mareike,

Perhaps I misunderstood your message, but you should not comment out the whole MARLIN_DLL variable, but just remove the path to the libEutelReaders from it. MARLIN_DLL contains the paths of the processor libraries but the libEutelReaders library is currently not working (and will soon be removed). 

For example, in my local installation the MARLIN_DLL variable in build_env.sh is defined like this:

export MARLIN_DLL="/nfs/dust/atlas/user/rossi/master/v01-19-02/Eutelescope/master/lib/libEutelescope.so:$EUTELESCOPE/lib/libEutelProcessors.so:$EUDAQ/lib/libNativeReader.so"

So I am loading the EUTelescope and EUDAQ libraries, but I don't load the libEutelReaders anymore. (also note that you may have an additional :$MARLIN_DLL in the end. That's fine, but it may be advantageous to remove it in some cases)
 
Cheers,

Edo

closed 2019-09-18 by jha92