eutelescope/eutelescope #461: EUTelescope installation finished with error

yotarid, 2019-03-26:
Dear EUTelescope experts, 

I am trying to install EUTelescope v01-19-02 software on CentOS 7.6, x86_64 GNU/Linux. 
I followed the installation instructions and steps on the EUTelescope website : 
./ilcsoft-install -i examples/eutelescope/release-standalone.cfg
but the installation finished with an error : 

CMake Warning at CMakeLists.txt:37 (MESSAGE):
  Your do not seem to have fetched the latest tags in your git repository --
  please consider running 'git fetch upstream'

-- Main EUDAQ library directory is clean (all changes have been commited).
-- Determined EUDAQ version 1.9.1
-- Determined EUDAQ stripped version 1.9.1
-- Setting build type to 'RelWithDebInfo' as none was specified.
-- Found EUTelescope: /home/tbuser/testbeam/v01-19-02/Eutelescope/master/eutelescope/libraries/include;/home/tbuser/testbeam/v01-19-02/Eutelescope/master/eutelescope/interfaces/pixel/include
-- Found EUTelescope: /home/tbuser/testbeam/v01-19-02/Eutelescope/master/lib/libEutelescope.so
-- Looking for LCIO...
-- Found LCIO: /home/tbuser/testbeam/v01-19-02/lcio/master/lib/liblcio.so
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Could NOT find Protobuf (missing: Protobuf_INCLUDE_DIR)
CMake Error at main/lib/CMakeLists.txt:23 (FIND_PACKAGE):
  By not providing "FindILCUTIL.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ILCUTIL", but
  CMake did not find one.

  Could not find a package configuration file provided by "ILCUTIL" with any
  of the following names:

    ILCUTILConfig.cmake
    ilcutil-config.cmake

  Add the installation prefix of "ILCUTIL" to CMAKE_PREFIX_PATH or set
  "ILCUTIL_DIR" to a directory containing one of the above files.  If
  "ILCUTIL" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/tbuser/testbeam/v01-19-02/Eutelescope/master/external/eudaq/v1.9.1/build/CMakeFiles/CMakeOutput.log".
See also "/home/tbuser/testbeam/v01-19-02/Eutelescope/master/external/eudaq/v1.9.1/build/CMakeFiles/CMakeError.log".
make: *** No rule to make target `install'.  Stop.

****************************** Finished installation ******************************

Can anyone tell me how to fix it ? 

Thanks in advance,
Younes

kwraight, 2019-03-26:
Hey Younes,

Probably the ILCUTIL has the wrong case in the build_env.sh file.
try:
In $EUTELESCOPE$/build_env.sh
Add at end of file:
export ILCUTIL=$ilcutil
source build_env.sh (new terminal if already sourced)

Hope this helps,
Kenny

yotarid, 2019-03-26:
Hi Kenny, 

Thanks a lot for your answer, seems to be working. 

Are there any included scripts that I can run in order to be sure that my EUTelescope software has been installed correctly ? (I don't have any data for the moment)

Thanks in advance,
Younes

kwraight, 2019-03-26:
Hey,

I suggest this: 

Fix eudaq build
	· Fix for ILCUTIL export case sensitivity:
	In $ILCSOFT/v01-19-02/Eutelescope/master/build_env.sh
	Add at end of file:
	export ILCUTIL=$ilcutil
	Ø source build_env.sh
	· In $ILCSOFT/v01-19-02/Eutelescope/master/external/eudaq/v1.x-dev/build/CMakeCache.txt
	Check BUILD_nreader:BOOL=ON
	Check ILCUTIL_DIR:PATH=/PATH_TO_ILCSOFT/v01-19-02/ilcutil/v01-02-01
	Ø cd $ILCSOFT/v01-19-02/Eutelescope/master/external/eudaq/v1.x-dev/build
	Ø cmake ..
	Ø make install

Build EUtelescope
	Ø cd $ILCSOFT/v01-19-02/Eutelescope/master/build
	Ø cmake ..
	Ø make install

Test run (requires afs rights to data)
	Ø cd $EUTELESCOPE/jobsub/examples/datura-noDUT
	Ø mkdir output
	Ø mkdir output/logs
	Ø mkdir output/histograms
	Ø mkdir output/lcio
	Ø mkdir output/database
	Ø jobsub -c config.cfg -csv runlist.csv converter 97

yotarid, 2019-03-26:
Hey, 

Building EUtelescope works fine. However I don't find any datura-noDUT directory. But there is one called GBL_noDUT. And there is no jobusb either.

yotarid, 2019-03-26:
Hey, 

Building EUTelescope works fine. There is no datura-noDUT directory, but there is GBL_noDUT. In this directory the config file is name config_DESY2018.cfg (or 2012). I tried running the jobsub command and I got this error : 

15:43:34 jobsub(INFO): Loaded config file config_DESY2018.cfg
15:43:34 jobsub(WARNING): Parameter 'databasepath' was not found in template file converter-tmp.xml
15:43:34 jobsub(WARNING): Parameter 'basepath' was not found in template file converter-tmp.xml
15:43:34 jobsub(WARNING): Parameter 'hotpixelrunnumber' was not found in template file converter-tmp.xml
15:43:34 jobsub(WARNING): Parameter 'steeringpath' was not found in template file converter-tmp.xml
15:43:34 jobsub(INFO): Successfully loaded csv file'runlist.csv'.
15:43:34 jobsub(INFO): Will now start processing the following runs: 117
15:43:34 jobsub(INFO): Now generating steering file for run number 000117..
15:43:34 jobsub(INFO): Parameter 'beamenergy' from 'runlist.csv' was not found in the cfg/template file. Parameter is not used in this step.
15:43:34 jobsub.converter(INFO): Now running Marlin on /home/tbuser/testbeam/v01-19-02/Eutelescope/master/jobsub/examples/GBL_noDUT/output/steering/run000117-converter.xml
15:43:34 jobsub.converter(INFO): 
15:43:34 jobsub.converter(INFO): 
15:43:34 jobsub.converter(INFO): 
15:43:34 jobsub.converter(INFO): 
15:43:34 jobsub.converter(INFO):
15:43:34 jobsub.converter(ERROR): 
15:43:34 jobsub.converter(INFO):
15:43:34 jobsub(ERROR): Marlin returned with error code -11
15:43:34 jobsub(WARNING): There were 1 error messages reported

kwraight, 2019-03-26:
Sorry the instructions I sent are a little out of date.
Make sure you have the output/steering file in the run directory.
From the runlist.csv file, it looks like the run number for 2018 is 701.
I don't know what this error code means sorry.

dreyling, 2019-03-26:
Hi @yotarid 

you are right: "GBL_noDUT" is the updated example which should run out of the box, please follow the 
README: https://github.com/eutelescope/eutelescope/blob/master/jobsub/examples/GBL_noDUT/README.md
or these slides: https://indico.cern.ch/event/731649/contributions/3237171/attachments/1779014/2893880/BTTB7_EUTelescopeTutorial_Arling.pdf

Thanks and yes @kwraight, 2018p-cfg is for run701 without conversion step from EUDAQ2 data, and 2012-cfg is for run117 from EUDAQ1 data as indicated in the README.

Good luck!

yotarid, 2019-03-26:
Hi again, 

I tried to follow the README steps, and also look at the slides, but I keep getting the same error : 

16:36:47 jobsub(INFO): Loaded config file config_DESY2012.cfg
16:36:47 jobsub(WARNING): Parameter 'databasepath' was not found in template file converter-tmp.xml
16:36:47 jobsub(WARNING): Parameter 'basepath' was not found in template file converter-tmp.xml
16:36:47 jobsub(WARNING): Parameter 'hotpixelrunnumber' was not found in template file converter-tmp.xml
16:36:47 jobsub(WARNING): Parameter 'steeringpath' was not found in template file converter-tmp.xml
16:36:47 jobsub(INFO): Successfully loaded csv file'runlist.csv'.
16:36:47 jobsub(INFO): Will now start processing the following runs: 117
16:36:47 jobsub(INFO): Now generating steering file for run number 000117..
16:36:47 jobsub(INFO): Parameter 'beamenergy' from 'runlist.csv' was not found in the cfg/template file. Parameter is not used in this step.
16:36:47 jobsub.converter(INFO): Now running Marlin on /home/tbuser/testbeam/v01-19-02/Eutelescope/master/jobsub/examples/GBL_noDUT/output/steering/run000117-converter.xml
16:36:47 jobsub.converter(INFO): 
16:36:48 jobsub.converter(INFO): 
16:36:48 jobsub.converter(INFO): 
16:36:48 jobsub.converter(INFO): 
16:36:48 jobsub.converter(INFO):
16:36:48 jobsub.converter(ERROR): 
16:36:48 jobsub.converter(INFO):
16:36:48 jobsub(ERROR): Marlin returned with error code -11
16:36:48 jobsub(WARNING): There were 1 error messages reported

Seems like it doesn't find libNativeReader.so

dreyling, 2019-03-26:
Is it the same for the 2018-701 combination? 
But you have afs access in order to access the data files?

yotarid, 2019-03-27:
Hey again, 

Same for 2018-701. 
No I don't have afs access ? Can I somehow request it ?

jha92, 2019-03-27:
Hi,
okay, so if you don't have afs-access, you cannot get the files as given in the paths in the examples.
For this, please refer to the wiki page https://github.com/eutelescope/eutelescope/wiki/Installation-and-preparation-for-BTTB7 where we showed alternatives to get the files (please do not forget to change in this case the the nativepath in the config-file).
I assume you have freshly started with EUTel? May I ask you, for which project you want to use the software?
Cheers,
Jan-Hendrik

yotarid, 2019-03-27:
Hi, 

Thanks a lot for the hint about the files, I'll have a look at the link you provided. 

Indeed I have freshly started with EUTel. I am working for the CMS-BRIL group at CERN. We will be using EUTel for test-beam at DESY in few months and I want to have EUTel locally to do some tests. 

I have an additional question. At some point I tried to install EUTel without installing CMake and ROOT using the nocmakenoroot config file. But I got this error immediately after launching the command : 

failed to find ROOT in /home/tbuser/testbeam/ILCsoft/v01-19-02/root/None

*** ERROR in module [ ROOT ]: DEBUG INFO: {'PLATFORM': 'Linux - CentOS Linux release 7.6.1810 (Core)', 'GCC_VERSION': 'gcc (GCC) 6.3.1 20170216 (Red Hat 6.3.1-3)', 'LSB_RELEASE': 'LSB Version:\t:core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch\nDistributor ID:\tCentOS\nDescription:\tCentOS Linux release 7.6.1810 (Core) \nRelease:\t7.6.1810\nCodename:\tCore', 'UNAME': 'Linux sunilcernpc 3.10.0-957.10.1.el7.x86_64 #1 SMP Mon Mar 18 15:06:45 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux'}

*** ERROR in module [ ROOT ]: At least one of these files: ['None/lib/libCore.so', 'None/lib64/libCore.so', 'None/lib/libCore.dylib']

How can I install EUTel without re-installing ROOT and CMake? And is it somehow useful to do it or do you think I should just rely on the standalone installation ? 

Thanks in advance. 

Cheers, 
Younes

edorossi, 2019-04-03:
Hi Younes,

I am sorry for the late reply. I would not rely on the nocmakenoroot config file, since we haven't tried it recently. Do you have problems using the standard configuration for the installation?
If you want I can try to install without cmake and root myself and try to fix the config, but it will take a few days at least.
Cheers,

Edo

edorossi, 2019-05-10:
Dear Younes,

I close this issue just to move the conversation to the similar issue #464 . 

Cheers,

Edo

closed 2019-05-10 by edorossi