eutelescope/eutelescope #462: Output of EUTelescope for analysis

ZhangGang00, 2019-05-02:
Dear EUTelescope experts,

I am using EUTelescope v2.0.0.  In our testbeam, there are 2 RD53A modules, both with size 20.0mm×9.6mm(X*Y) 
50𝝻m×50𝝻m sensor:   400 pixels in x direction, 192 pixels in y direction
100𝝻m×25𝝻m sensor: 200 pixels in x direction, 384 pixels in y direction
I got normal residuals for each plane with GBL fitter. I have several concerns about the output of this latest EUTelescope.

(1) In directory "Tracks", the xPos and yPos are local coordinates. But many yPos are actually outside the module. I do not quite understand why this happen. I believe you have considered the alignment when transfering global coordinates to local coordinates, right? And what does "omega" and "phi" mean in "Tracks" directory?
[image]

(2) I tried to use TBmon2 to analyse the output of EUTelescope v2.0.0. But some variables TBmon2 needs are missing in this output. Is it possible to add these variables in the output?
They are "dxdz" and "dydz" in directory "Tracks", "tot" and "lv1" in directory "ZeroSuppressed".

The output is in attachment. Looking forward to your reply!

Cheers,
Gang
[tbtrack000446.root.zip](https://github.com/eutelescope/eutelescope/files/3140001/tbtrack000446.root.zip)

edorossi, 2019-05-02:
Dear Gang,

Can you post your config file and the steering template for fitGBL?

1) The positions in the plots are from the tracks and some of them may be outside the active area of your module.
There is another problem I can see, and it is related to the flag `applyCenterShift` which was recently added. If this flag is set to false, which is the default, the origin of the axes is the geometrical center of the sensor. For TBmon2 you want the origin to be in the bottom left corner of the sensor, so that flag should be set to true.
Regarding "omega" and "phi", these quantities are supposed to be the incidence angle of the track, but right now they are giving the proper numbers. Do you need the incidence angles?

2) TBmon2 needs a few changes to be compatible with the GBL output, since some branches in the final TTree have different names. For example "lv1" is called "Time" and "tot" is now "Signal". I know that @maweer managed to make TBmon2 compatible with the GBL output, so she surely has more info on this!

Cheers,

Edo

ZhangGang00, 2019-05-02:
Hi Edo,

Thanks for the quick reply!

I keep "applyCenterShift" as false before, so I will set it as true. Do you mean "omega" and "phi" are equal to "dx/dz" and "dy/dz"? TBmon2 needs "dx/dz" and "dy/dz". The files you may need are all in attachments. 

Cheers,
Gang
[config.zip](https://github.com/eutelescope/eutelescope/files/3140257/config.zip)

edorossi, 2019-05-02:
Dear Gang,

I am not sure about the nomenclature in TBmon2, since I have never used it, but yes, I think omega and phi corresponds to dx/dz and dy/dz. However, in GBL the calculation is not done totally properly right now, so the values may be quite off (but if you need it we can try to fix it quite quickly). 

Cheers,

Edo

ZhangGang00, 2019-05-02:
Hi Edo,

There are 2 variables called "dxdz" and "dydz" in directory "tracks" in the output produced by old version EUTelescope.

Cheers,
Gang

edorossi, 2019-05-02:
Dear Gang,

I just checked how `dxdz` and `dydz` were calculated and dumped with DAF, and I can confirm that the first corresponds to `omega` and the second to `phi`!

Cheers,

Edo

edorossi, 2019-05-03:
Dear Gang, 

By mistake I closed the issue. Let me know if you have further problems related to the track dumping, then we can reopen it!

Cheers,

Edo

closed 2019-05-03 by edorossi

ZhangGang00, 2019-05-03:
Dear Edo,

Thank you very much! As you said, if you could fix calculation of omega and phi quickly,  it would be great!

Cheers,
Gang

edorossi, 2019-05-03:
Dear Gang,

I implemented a fix for the incidence angle. Unfortunately, this can be considered only temporary, since there are a few things I am not 100% sure about. Moreover, this is the incidence angle in the global coordinate system not in the local coordinate system.

Cheers,

Edo

maweer, 2019-05-03:
Dear Gang,

you can fork my repository in TBmon2 and checkout the dortmund.dev branch (https://gitlab.cern.ch/mweers/tbmon2.git) . This is capable to analyse the output of EUTelescope and GBL. I wanted to merge it in the main repository, but the merge request is still open....
Regarding to those angle parameters: I think TBMon2 never really used them ;-)

Cheers,
Mareike