eutelescope/eutelescope #463: Some further questions about track reconstruction in GBL fitter

ZhangGang00, 2019-05-07:
Dear developers,

I am using EUTelescope v2.0.0. I have some further questions about track reconstruction in GBL fitter.

(1) Why are the pulls distribution of DUTs all empty? Doesn't the codes fill such histograms in EUTelescope?

(2) For [fitGBL] in config, we need to set estimated resolutions for each plane to calculate the chi2 in track reconstruction. I wonder if these resolution will update in the fitting then to calculate chi2 more precisely?

(3) Usually, we exclude the DUTs in fitting. So the hits on DUTs are extrapolated. Is there any way to get the error of such extrapolation on DUTs?

Cheers,
Gang

edorossi, 2019-05-07:
Dear Gang,

I am sorry for my brevity, but I am writing from my phone now :)

1) At a very first approximation, the pull residuals are given by residuals/resolutions. However, if you exclude a plane, you are not passing its hit positions and resolution to the GBL fit. Since no resolution is passed, it is not possible to calculate the pull residuals, thus we leave the histogram empty.

2) No, the resolutions used for track fitting are the ones set in the config and they are not changed by the fitting procedure.

3) The extrapolation error is basically the tracking resolution, which we discussed in a recent issue.

Cheers,

Edo

edorossi, 2019-05-10:
Dear Gang,

Was my answer explicative or you have further question?
Just asking to know whether I can close the issue!

Cheers,

Edo

ZhangGang00, 2019-05-10:
Dear Edo,

Thanks for the reply. I have no further questions for the moment. Please close this issue. 

Cheers,
Gang

closed 2019-05-10 by edorossi