eutelescope/eutelescope #464: ROOT conflict when running example analysis

yotarid, 2019-05-07:
Dear experts, 

As I try to run the example analysis described here https://github.com/eutelescope/eutelescope/blob/master/jobsub/examples/GBL_noDUT/README.md , the conversion step crashes with this error saved in logs folder : 

---=== Analysis started on Tuesday, 07. May 2019 02:18PM ===---

Warning in <TClassTable::Add>: class TAtomicCount already in TClassTable
Warning in <TClassTable::Add>: class TConditionImp already in TClassTable
Warning in <TClassTable::Add>: class TCondition already in TClassTable
Warning in <TClassTable::Add>: class TMutexImp already in TClassTable
Warning in <TClassTable::Add>: class TMutex already in TClassTable
Warning in <TClassTable::Add>: class TRWLock already in TClassTable
Warning in <TClassTable::Add>: class ROOT::TSpinMutex already in TClassTable
Warning in <TClassTable::Add>: class TSemaphore already in TClassTable
Warning in <TClassTable::Add>: class TThread already in TClassTable
Warning in <TClassTable::Add>: class TThreadFactory already in TClassTable
Warning in <TClassTable::Add>: class TThreadImp already in TClassTable
Warning in <TClassTable::Add>: class TPosixCondition already in TClassTable
Warning in <TClassTable::Add>: class TPosixMutex already in TClassTable
Warning in <TClassTable::Add>: class TPosixThread already in TClassTable
Warning in <TClassTable::Add>: class TPosixThreadFactory already in TClassTable
Warning in <TClassTable::Add>: class TArchiveFile already in TClassTable
Warning in <TClassTable::Add>: class TArchiveMember already in TClassTable
Warning in <TClassTable::Add>: class TBufferFile already in TClassTable
Warning in <TClassTable::Add>: class TBufferText already in TClassTable
Warning in <TClassTable::Add>: class TBufferJSON already in TClassTable
Warning in <TClassTable::Add>: class TGenCollectionProxy already in TClassTable
Warning in <TClassTable::Add>: class TGenCollectionProxy::Value already in TClassTable
Warning in <TClassTable::Add>: class TGenCollectionProxy::Method already in TClassTable
Warning in <TClassTable::Add>: class TCollectionProxyFactory already in TClassTable
Warning in <TClassTable::Add>: class TCollectionStreamer already in TClassTable
Warning in <TClassTable::Add>: class TCollectionClassStreamer already in TClassTable
Warning in <TClassTable::Add>: class TCollectionMemberStreamer already in TClassTable
Warning in <TClassTable::Add>: class TDirectoryFile already in TClassTable
Warning in <TClassTable::Add>: class TEmulatedCollectionProxy already in TClassTable
Warning in <TClassTable::Add>: class TEmulatedMapProxy already in TClassTable
Warning in <TClassTable::Add>: class TFPBlock already in TClassTable
Warning in <TClassTable::Add>: class TFile already in TClassTable
Warning in <TClassTable::Add>: class TFileCacheRead already in TClassTable
Warning in <TClassTable::Add>: class TFileCacheWrite already in TClassTable
Warning in <TClassTable::Add>: class TFileMerger already in TClassTable
Warning in <TClassTable::Add>: class TFilePrefetch already in TClassTable
Warning in <TClassTable::Add>: class TFree already in TClassTable
Warning in <TClassTable::Add>: class TKey already in TClassTable
Warning in <TClassTable::Add>: class TKeyMapFile already in TClassTable
Warning in <TClassTable::Add>: class TLockFile already in TClassTable
Warning in <TClassTable::Add>: class TMapFile already in TClassTable
Warning in <TClassTable::Add>: class TMapRec already in TClassTable
Warning in <TClassTable::Add>: class TMemFile already in TClassTable
Warning in <TClassTable::Add>: class TStreamerInfo already in TClassTable
Warning in <TClassTable::Add>: class TStreamerInfoActions::TConfiguration already in TClassTable
Warning in <TClassTable::Add>: class TStreamerInfoActions::TConfiguredAction already in TClassTable
Warning in <TClassTable::Add>: class TStreamerInfoActions::TActionSequence already in TClassTable
Warning in <TClassTable::Add>: class TVirtualArray already in TClassTable
Warning in <TClassTable::Add>: class TVirtualObject already in TClassTable
Warning in <TClassTable::Add>: class TZIPFile already in TClassTable
Warning in <TClassTable::Add>: class TZIPMember already in TClassTable
Warning in <TClassTable::Add>: class ROOT::Experimental::TBufferMerger already in TClassTable
Warning in <TClassTable::Add>: class ROOT::Experimental::TBufferMergerFile already in TClassTable
<built-in>:11:9: warning: '__CLING__GNUC__' macro redefined [-Wmacro-redefined]
#define __CLING__GNUC__ 7
        ^
<built-in>:374:9: note: previous definition is here
#define __CLING__GNUC__ 4
        ^
<built-in>:12:9: warning: '__CLING__GNUC_MINOR__' macro redefined [-Wmacro-redefined]
#define __CLING__GNUC_MINOR__ 3
        ^
<built-in>:375:9: note: previous definition is here
#define __CLING__GNUC_MINOR__ 8
        ^
Warning in cling::IncrementalParser::CheckABICompatibility():
  Possible C++ standard library mismatch, compiled with __GLIBCXX__ '20180303'

It seems like a ROOT conflict on my machine. Note that I already had a ROOT version before installing EUTEL. 
Do you have any idea how I can fix that ? 

Thanks in advance for the answer. 

Cheers,
Younes

edorossi, 2019-05-07:
Dear Younes,

Are you running locally or on lxplus (or similar)? And are the two versions of ROOT different? (the one installed with EUTelescope and the one installed locally)

I have never stumbled in this problem, but I think commenting out lines 42-45 from the build_env.sh may help. In this way you would not source the EUTelescope ROOT, and hopefully you use your local ROOT. If you do this, then you need a fresh terminal and sourcing the build_env.sh to effectively implement the change.

Cheers,

Edo

yotarid, 2019-05-07:
Dear Edo, 

I am running locally. ROOT versions are different. 

I have commented out ROOT lines in the build_env.sh so that EUTelescope ROOT is not sourced. But the conversion process still crashes. You can find the log file under 

/afs/cern.ch/user/y/yotarid/BCM1F/ILCsoft/v01-19-02/Eutelescope/master/jobsub/examples/GBL_noDUT/output/logs

Here is the error in the log file : 
15:09:20 jobsub.converter(ERROR): libCore dictionary payload:144:10: fatal error: 'ROOT/RArrayView.hxx' file not found
15:09:20 jobsub.converter(ERROR): #include "ROOT/RArrayView.hxx"
15:09:20 jobsub.converter(ERROR): ^~~~~~~~~~~~~~~~~~~~~
15:09:20 jobsub.converter(ERROR): Error in <TInterpreter::AutoParse>: Error parsing payload code for class TAttAxis with content:
15:09:20 jobsub.converter(ERROR):
15:09:20 jobsub.converter(ERROR): #line 1 "libCore dictionary payload"

dreyling, 2019-05-08:
Hi @yotarid ,

which Eutelescope installation have you performed?
```./ilcsoft-install -i examples/eutelescope/release-standalone.cfg``` (ROOT and all other dependencies will be installed)
or 
```./ilcsoft-install -i examples/eutelescope/release-nocmakenoroot.cfg``` for which cmake and ROOT has to be installed already.

If you choose the first option, I guess it is important to haven't set any ROOT environment aka thisroot.sh in your bash profile or similar. For example, it should look like the first empty output and not as the second:
```dreyling@atlas11:~$ env | grep -i ROOT
dreyling@atlas11:~$ source /opt/root/root_v6.16.00/bin/thisroot.sh
dreyling@atlas11:~$ env | grep -i ROOT
LD_LIBRARY_PATH=/opt/root/root_v6.16.00/lib
DYLD_LIBRARY_PATH=/opt/root/root_v6.16.00/lib
ROOTSYS=/opt/root/root_v6.16.00
CMAKE_PREFIX_PATH=/opt/root/root_v6.16.00
JUPYTER_PATH=/opt/root/root_v6.16.00/etc/notebook
SHLIB_PATH=/opt/root/root_v6.16.00/lib
LIBPATH=/opt/root/root_v6.16.00/lib
PYTHONPATH=/opt/root/root_v6.16.00/lib
MANPATH=/opt/root/root_v6.16.00/man::/opt/puppetlabs/puppet/share/man
PATH=/opt/root/root_v6.16.00/bin:/home/dreyling/.local/bin:/afs/desy.de/common/passwd:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/opt/puppetlabs/bin
```

closed 2019-05-13 by yotarid