eutelescope/eutelescope #465: Installation on Lxplus CC7

gauzinge, 2019-05-08:
Dear Eutelescope Developers,

we've been frantically trying to install the Eutelescope reconstruction framework on one of our lab PCs - it turns out there is a conflict with the original root version...

Are there any up-to-date instructions on how to install the minimal version required for Telescope data analysis on lxplus (minding that now the standard SLC version is CC7)?

Or is there a centralised installation maintained somewhere on cvmfs?

We'd be thrilled to get any help on this!

Best

dreyling, 2019-05-08:
Hi @gauzinge ,

have you tried this: https://github.com/eutelescope/eutelescope/wiki/Installation-and-preparation-for-BTTB7
?

Cheers,
Jan

edorossi, 2019-05-09:
Dear @gauzinge ,

I managed to fix the problem by setting manually the cmake flag ROOT_DIR to the path of the root version I want to use. I needed to do it for GEAR, RAIDA, GBL and Eutelescope. I did it by adding to the file examples/eutelescope/release-standalone.cfg lines like this:

ilcsoft.module("GEAR").envcmake['ROOT_DIR']='"/cvmfs/sft.cern.ch/lcg/releases/LCG_88/ROOT/6.08.06/x86_64-slc6-gcc62-opt/"'

The problem is present on CentOS7 but not on slc6. You can find the release-standalone.cfg file I used attached (and note that it is actually a .txt file, since Github doesn't like .cfg). In that config you can also see that I am skipping the ROOT installation and the cmake installation, since I am using already a suitable version for the latter (3.11.0).

Basically the steps I did are:

1. export ILCSOFT=/path/where/you/want/to/install
2. mkdir -p $ILCSOFT && cd $ILCSOFT
3. git clone -b dev-base https://github.com/eutelescope/ilcinstall $ILCSOFT/ilcinstall
4. Setting up gcc6.2 (this may not be needed if you source ROOT as in the next point, but I haven't tried it directly)
5. Setting up cmake 3.11.0 (not needed if you install cmake through ilcinstall in the release-standalone.cfg or if you already have a suitable version)
6. source /cvmfs/sft.cern.ch/lcg/releases/LCG_88/ROOT/6.08.06/x86_64-slc6-gcc62-opt/ROOT-env.sh
7. I modified the file examples/eutelescope/release-standalone.cfg as explained above
8. ./ilcsoft-install -i examples/eutelescope/release-standalone.cfg

With these steps the installation worked. Let me know if it works for you as well.
This is just a temporary solution, but we are working to fixing this in a stable way.

[release-standalone.txt](https://github.com/eutelescope/eutelescope/files/3164110/release-standalone.txt)

Cheers,

Edo

gauzinge, 2019-05-10:
Dear Edo,

thanks for your update! We have circumvented the problem for now by using a virtual CC7 machine without any local root - this covers our most urgent needs! I'll therefore close the issue and hope you can come up with a stable fix.

thanks
Georg

closed 2019-05-10 by gauzinge