eutelescope/eutelescope #468: Dump the event header parameters into the output TTree

cyrilbecot, 2019-05-20:
This should be done within processors/EUTelGBLOutput.cc 
The `_dumpHeader` flag is already present to switch it on/off
The proper structure to do the dump is probably a map<string, type>
LCIO provides the functions to run over all the parameters for each possible type it stores for the header - string, int and float

cyrilbecot, 2019-05-20:
Targets branch dumpHeaderParams

edorossi, 2019-05-21:
Now it is merge request #469

edorossi, 2019-05-21:
As an update, with ROOT6 everything seems to work fine. With ROOT5, however, I can compile the code and run, but the header TTree is not created and I have such errors:

18:29:13 jobsub.fitGBL(ERROR): Error in <TTree::Branch>: The pointer specified for IntHeaders is not of a class or type known to ROOT

Since the standard is ROOT6 now, this is not a big issue, but I will add a comment on this on the parameter description.

cyrilbecot, 2019-05-21:
Ok, there was a way to get the header for std::string compiled and make it readable by the TTree. Admittedly, I haven't done it in a long time (if ever... ROOT5 is quite far already :)), and I'm not sure I'd fine easily how to put it within the whole EUTelescope infrastructure. However, I could find probably just find a more plain-C like structure to hold this part and not have the crash. To be seen.

edorossi, 2019-05-21:
Just for clarity, with ROOT5 it doesn't crash, it just doesn't create the header TTree. For this reason, I wouldn't be too concerned. That's not very nice, but it makes sense to avoid to be too stubborn for the backward compatibility with ROOT.

jha92, 2019-09-04:
Any news here or can be closed, @cyrilbecot & @edorossi ?