eutelescope/eutelescope #471: Detailed study about clusters

ZhangGang00, 2019-05-29:
Dear EUTelescope experts,

As far as I know, EUTelescope is using charge-weighted algorithm(analog readout) to calculate position of clusters. I wonder if we can just use binary readout to calculate position of clusters? Is there such option in configure files? Furthermore, how can I just keep the events in the output where cluster size on DUTs is exactly 1 after reconstructing tracks successfully? 

Cheers,
Gang

TCThenna, 2019-05-31:
Hi Gang,

the MIMOSA26 is binary, so we are clustering binary data all the time :) Just have a look at the no_DUT example.

You could write a new collection and only write clusters into it that fulfil a certain requirement, e.g. having a cluster size of 1.

ZhangGang00, 2019-05-31:
Dear EUTelescope experts,

I am surprised EUTelecope is using binary readout as I read from many documentations that charge-weighted clustering is used, such as section 4.3 in paper https://iopscience.iop.org/article/10.1088/1748-0221/12/12/C12053/pdf .

Could you please point me which source codes I should modify to filter the events where size of clusters on DUTs is exact 1? It is quite hard to identify the relevant codes in EUTelescope by myself.

Cheers,
Gang

TCThenna, 2019-06-01:
The weights are just  0 or 1 in the binary case.

This is how you get the cluster associated to a hit, and other information:
    TrackerHit * meshit = dynamic_cast<TrackerHit*>( collection->getElementAt(ihit) );
    const double * pos = meshit->getPosition();
    const EVENT::FloatVec cov = meshit->getCovMatrix();
    float charge = 0.;
    int clusterdim[2] = {0,0};

    TrackerDataImpl* clusterVector = static_cast<TrackerDataImpl*>( meshit->getRawHits()[0]);
    EUTelSimpleVirtualCluster * cluster=0;

    float locx = -1;
    float locy = -1;
    if ( meshit->getType() == kEUTelSparseClusterImpl ) 
    {
      cluster = new EUTelSparseClusterImpl< EUTelGenericSparsePixel > ( clusterVector );
      charge = cluster->getTotalCharge();
      cluster->getClusterSize(clusterdim[0], clusterdim[1]); // x = [0], y = [1]
      cluster->getCenterOfGravity(locx, locy);
    }

ZhangGang00, 2019-06-03:
Dear EUTelescope experts,

Thank you very much for the above information!  But I still need your further help.

My purpose is to add selection `clusterdim[0]==1 && clusterdim[1]==1` for DUT in clustering step, so that only events with one-pixel cluster on DUT are kept and passed to hitmaker step. I find several relevant processors in EUTelescope, such as EUTelSparseClustering.cc, EUTelClusterFilter.cc, EUTelClusteringProcessor.cc and EUTelProcessorClusterAnalysis.cc. I am not sure where I should insert my selection. Could you please point me exactly where I should add my selection?

Thanks,
Gang

jha92, 2019-07-15:
Dear Gang,
sorry for the longer silence from our side. What is the status of your request?
Cheers,
Jan-Hendrik

ZhangGang00, 2019-07-16:
Dear Jan-Hendrik,

It seems I need to modify the source codes of EUTelescope to achieve the aforementioned study. But it is not realistic for me to spend considerable time on this. So I do not intend to continue this study. You can close this issue now.

Cheers,
Gang

closed 2019-09-04 by jha92